maakbaas/esp8266-iot-framework #3: Failed to build

mtoko, 2020-07-03:
Thanks for the great work
I have tried to build using PlatformIO but have the following errors
webServer.cpp
'updater' was not declared in this scope [91,9]
'updater' was not declared in this scope [100,32]
Also a warning in multiple places about SPIFFS is deprecated
Any assistance is appreciated
Darryl

mtoko, 2020-07-03:
I have found the solution, I have added an include at the top of webServer.cpp
#include "updater.h"

maakbaas, 2020-07-04:
Glad you resolved the issue. The include to `updater.h` is actually part of `main.c`. Were you trying with a customized `main.c`? Otherwise it is curious why it would be missing this dependency...

mtoko, 2020-07-05:
Hi
I don't think the issue is really resolved. The code will build if the updater.h is included in webServer.cpp, but when the firmware is flashed and you enter a new SSID and password nothing happens and it does not connect to the new WiFi. I am using the standard code, the only change was the board type which is a Wemos D1 Mini with 4MB.
The error is
src/webServer.cpp: In lambda function:
_**src/webServer.cpp:92:9: error: 'updater' was not declared in this scope
         updater.requestStart("/" + request->arg("filename"));
         ^
src/webServer.cpp: In lambda function:
src/webServer.cpp:101:32: error: 'updater' was not declared in this scope
         jsonBuffer["status"] = updater.getStatus();**_
                                ^
src/webServer.cpp: In static member function 'static void webServer::handleFileUpload(AsyncWebServerRequest*, String, size_t, uint8_t*, size_t, bool)':
src/webServer.cpp:164:24: warning: 'SPIFFS' is deprecated (declared at /home/darryl/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:269): SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
         fsUploadFile = SPIFFS.open(filename, "w");
                        ^
*** [.pio/build/d1_mini/src/webServer.cpp.o] Error 1

maakbaas, 2020-07-05:
I can't reproduce the issue you are having with updater.h. What does your main.c look like?

The other issue you describe with WiFi might be related to issue #6, which is resolved in the latest revision, so you could try that!

tssva, 2020-07-05:
I don't have the WiFi issues but a clean clone of the repository with no changes will not build for me without adding an include for updater.h to webServer.cpp.

maakbaas, 2020-07-06:
I have included `updater.h` into `webServer.cpp` in commit ed476fc8e0d57aa88370d20c76a82951bd3be493. It should have been there in the first place, thanks for flagging. 

@mtoko: I will close this issue, if you still have issues connecting to WiFi, please open a separate issue so I can investigate that further.

closed 2020-07-06 by maakbaas