maakbaas/esp8266-iot-framework #6: Difficulty starting up due to HTTPRequest::begin loop

factormystic, 2020-07-03:
Hi, I'm just getting started with this framework and have got it compiled, uploaded etc.

I noticed that while the captive portal loads if you navigate directly to the IP address, it wouldn't seem to connect to my wifi or run requests from the main `loop`. This lead me to adding some `Serial.println`'s around the code, and I realized that the main `setup` function wasn't even being completed, and therefore the main `loop` wasn't getting started (and therefore nothing was fully working).

Eventually I realized that it's getting stuck in the while loop in `fetch.cpp`, [here](https://github.com/maakbaas/esp8266-iot-framework/blob/master/src/fetch.cpp#L10-L14):

```cpp
    time_t now = time(nullptr);
    while (now < 8 * 3600 * 2)
    {
        delay(500);
        now = time(nullptr);
    }
```

The initial `now` value is `2` and increments up from there, so that means the loop will wait basically need to wait about 16 hours before completing...!

I commented out the `while` loop and everything seems to then be working normally (main `loop` runs, http requests succeed, captive portal automatically detected, etc).

I'm wondering what the while loop is for? And is it intentional that it should take this long? Is there any harm in removing it?

maakbaas, 2020-07-04:
The reasoning behind the `fetch.begin()` function is to read the time from a server, since this is required to validate HTTPS requests. 

This blocking loop is waiting until the time is obtained. As long as the ESP is running a captive portal and server connection can not be made, this will prevent the application from running. And this should indeed not be the case.

Doing some further testing I found to my surprise that even when I am not explicitly requesting the time from the server, the ESP8266 will still fetch the time when connected to WiFi. This is probably part of the ESP SDK which is used at the lowest level.  Even though I did not find explicit documentation on this, I found [this topic](https://forum.arduino.cc/index.php?topic=655222.0) where the same behavior is observed and attributed to the SDK. 

Because of this I have completely removed the `fetch.begin()` function in commit  a9b93f94fac866e15158d5ee87806829369e2cf1 since that means it is no longer necessary.

factormystic, 2020-07-04:
Good to know, thanks. It's pretty surprising that the SDK will make network activity without explicitly opting in, though.

maakbaas, 2020-07-04:
Agreed, that is also why I am still looking for a better source/confirmation on this.

factormystic, 2020-07-04:
I just pulled in that commit, and now https requests don't work. I think that the NTP sync may actually be required (I wonder if this is something that can differ between hardware). If I check `gmtime_r` without `configTime` first, I get `Thu Jan  1 00:00:07 1970`. It seems like the safest option might be to restore `configTime` in some form to guarantee the device time is up to date.

maakbaas, 2020-07-04:
Yes, I figured the same, which is why I re-opened the ticket. I have reverted to a basic non-blocking NTP request in commit e9cafb4ddb863a443febd6a586594cdf0cc2fc3a. 

I was a bit too quick committing (and should start using branches).

I found that if I connect to my home WiFi it pulls the time automatically, but not when I connect to a hotspot on my phone. Currently investigating if it could fetch the time from the access point itself somehow for some access points and not for others...

closed 2020-07-04 by maakbaas

factormystic, 2020-07-05:
Alright sweet. I think that your latest update here should work okay as long as an https request isn't attempted before the clock sync is completed, and as long as it's able to complete after connecting to wifi but before the first https request. I have an idea about how this could be done.

One more question on `configTime` itself, should the first parameter be `0` for UTC?