maakbaas/esp8266-iot-framework #8: `forget` (and probably other destructive requests) should be `POST`

factormystic, 2020-07-05:
I noticed this after I connected to the device dashboard, and noticed that the captive portal was started in the device serial monitor. I *think* what happened is that the browser either replayed a previous "forget" action, or prefected the "forget" request on the wifi settings page.

Either way, this could be solved with a `post` (or possibly `delete`) request instead. It's a common web design principle to only use `get` requests for safe, non-destructive operations ([source](https://restfulapi.net/http-methods/#get)).

I can open a PR for this unless you/someone else does first; I'm just making a note here right after I noticed it, to revisit later.

maakbaas, 2020-07-07:
Good suggestion. If you want me to contribute, let me know. Otherwise I will await your PR.

maakbaas, 2020-07-09:
Thanks to @tssva in #15 `DELETE`, `PUT` and `PATCH` requests are now also available.

closed 2020-07-19 by maakbaas