maakbaas/esp8266-iot-framework #9: Decide how to integrate the framework in new projects

maakbaas, 2020-07-05:
What would be the best way for you to start a new project based on this framework? 

As raised by @factormystic  in #7: 

> I think another part of it is, what is the expectation for how this project should be used? There's a few ways to think about it. For example, people who want to build apps on this framework might clone/fork and start editing/customizing whatever they want. Of course that's allowed, but it might not be helpful especially if people want to incorporate core changes into their projects later on. I'm still figuring that out myself.
> 
> One idea I had is to have people start a fresh repo for their own projects instead of fork, but then add this code as a git submodule. To reference an inspiration, this is how themes for the static blogging program Hugo typically work (explanation). The obvious problem is that's a pure web project whereas this is a compiled IoT project so I actually don't know if it can work, but conceptually I like the idea of pulling this project in as a dependency somehow, then in my own repo I have only my own configuration.json, main.cpp, etc, and can easily pull in updates.

Another suggestion could be to make it into a PlatformIO [library](https://docs.platformio.org/en/latest/librarymanager/creating.html) so that you can include it into your project in that way.

I guess for convenience for users, cloning or forking the project to get started is the easiest, but I also see the drawbacks in that. At the same time, when making the project into for instance a PlatformIO library, it is less straightforward to make modifications to the web interface or other parts of the framework to tailor towards your specific application, or is this mainly a good thing?

TyIsI, 2020-07-08:
Found this project while searching for certificate solutions on ESP(8266) and I'm thinking about adopting it for a project. Thanks for the great work!

Going to brain dump here for a moment.

I think there are roughly 3 options:
- Move the general framework into the lib directory (or a sub-directory) and call the respective "lib_setup" and "lib_loop" functions from the library
- Create a separate app directory within the `src` directory with a separate app.cpp or main.cpp "app_setup" and "app_loop" that get called from the main `setup` and `loop`
- Make this into a fully fledged library and provide config macro checks for overriding specific files with custom files

File generation/customization could easily be filtered through the PIO include directory, but I don't know how much complexity this would bring.

I think this could easily be used as a template project or with a project tool, much like `create-react-app`.

To me, it would be nice to have this as a library, which could just be invoked as a class instance.

factormystic, 2020-07-09:
The Platform IO library is interesting, I'm not very familiar with that option. From glancing at the documentation it looks like that'd mean adding a few files to this repo (and then registering with them I assume).

What would it look like for people who want to use this project in their own code? Guessing it'd be to add it via lib_deps?

maakbaas, 2020-07-09:
They would indeed simply add it via `lib_deps`. For me this also seems the best approach when using PlatformIO.

I will try this out in a branch to see what is needed and if all the pre-build functions of the framework could be implemented within a library as well. Expect a branch somewhere in the coming days.

maakbaas, 2020-07-18:
I changed the repository to work as a PlatformIO library, while maintaining the possibility to work directly from a clone in pull request #19.

The documentation is updated accordingly.

Only thing I still need to do is to add it to the library manager from PlatformIO.

Let me know if you have any feedback, ideas or issues!

closed 2020-07-18 by maakbaas

maakbaas, 2020-07-19:
See this location for the published package :).

https://platformio.org/lib/show/10789/ESP8266%20IoT%20Framework