maakbaas/esp8266-iot-framework #27: Tailor certificates.h generation to the URLs connected to by the application

gl1024, 2020-08-10:
You might want to consider a future enhancement to tailor the generation of certificates.h to only the certs needed for the specific application program to greatly reduce the amount of program memory used for cert storage.  For many applications it is likely not an issue, but for those that hit a memory constraint then it would be helpful if certificates.h could be generated based on a data file that specifies the set of URLs needed for application connections.

maakbaas, 2020-08-12:
Thanks for your suggestion. As discussed: 

Although it is nice to save memory where possible, I feel that for most use cases and projects the ESP8266 has an abundance of space, so the ~170k is probably not an issue for most. 

That being said, it is of course much more elegant to not pack anything in the binary that is not needed. Your comment got me thinking: one mechanism could be to add the URLs to generate the certificates for as preprocessor defines during the build process.

I have no timeframe when I would work on this. More people who would like this?

maakbaas, 2020-08-30:
A new build flag has been included:

`DOMAIN_LIST=google.com,maakbaas.com`: This option defines a comma separated list of domains. When this option is included the `REBUILD_CERTS` option will only include the root certificates in `certificates.h` that are part of this list. 

The default certificate store when this option is not active contains ~150 certificates, and is roughly 170kB in size. Use this option to reduce the size of the certificate store.

closed 2020-08-30 by maakbaas