maakbaas/esp8266-iot-framework #32: does not compile after changes

pioioTwo, 2020-09-02:
does not execute changes to html build files, I am translating the web server for my language and ends up not copying

Log 

0 info it worked if it ends with ok
1 verbose cli [
1 verbose cli   'C:\\Program Files\\nodejs\\node.exe',
1 verbose cli   'C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js',
1 verbose cli   'run',
1 verbose cli   'build'
1 verbose cli ]
2 info using npm@6.14.6
3 info using node@v12.18.3
4 verbose run-script [ 'prebuild', 'build', 'postbuild' ]
5 info lifecycle esp@1.0.0~prebuild: esp@1.0.0
6 info lifecycle esp@1.0.0~build: esp@1.0.0
7 verbose lifecycle esp@1.0.0~build: unsafe-perm in lifecycle true
8 verbose lifecycle esp@1.0.0~build: PATH: C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin;C:\Users\pcps2\OneDrive\Documentos\PlatformIO\Projects\Modulo_Central_V2.0.0\.pio\libdeps\esp07\ESP8266 IoT Framework\node_modules\.bin;C:\Users\pcps2\.platformio\packages\toolchain-xtensa\bin;C:\Users\pcps2\.platformio\packages\tool-esptool;C:\Users\pcps2\.platformio\packages\tool-mkspiffs;C:\Users\pcps2\.platformio\packages\tool-esptoolpy;C:\Users\pcps2\.platformio\penv\Scripts;C:\Users\pcps2\.platformio\penv;C:\Python27\;C:\Python27\Scripts;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Git\cmd;C:\Program Files\nodejs\;C:\Users\pcps2\AppData\Local\Microsoft\WindowsApps;C:\Users\pcps2\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\pcps2\AppData\Roaming\npm
9 verbose lifecycle esp@1.0.0~build: CWD: C:\Users\pcps2\OneDrive\Documentos\PlatformIO\Projects\Modulo_Central_V2.0.0\.pio\libdeps\esp07\ESP8266 IoT Framework
10 silly lifecycle esp@1.0.0~build: Args: [ '/d /s /c', 'webpack --mode=production' ]
11 silly lifecycle esp@1.0.0~build: Returned: code: 1  signal: null
12 info lifecycle esp@1.0.0~build: Failed to exec build script
13 verbose stack Error: esp@1.0.0 build: `webpack --mode=production`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:315:20)
13 verbose stack     at ChildProcess.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\lib\spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:315:20)
13 verbose stack     at maybeClose (internal/child_process.js:1021:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
14 verbose pkgid esp@1.0.0
15 verbose cwd C:\Users\pcps2\OneDrive\Documentos\PlatformIO\Projects\Modulo_Central_V2.0.0\.pio\libdeps\esp07\ESP8266 IoT Framework\build
16 verbose Windows_NT 10.0.18362
17 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "run" "build"
18 verbose node v12.18.3
19 verbose npm  v6.14.6
20 error code ELIFECYCLE
21 error errno 1
22 error esp@1.0.0 build: `webpack --mode=production`
22 error Exit status 1
23 error Failed at the esp@1.0.0 build script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

maakbaas, 2020-09-02:
Can you try running once the command `npm ci` from the command line? And then try to do the build again? This will install the correct packages from NPM. 

If this indeed resolves your issue I will add this instruction to the documentation.

pioioTwo, 2020-09-05:
hello, this did revolve, but the problem arose in python, for rebuild.

I tried to reinstall python, and add the environment variables, but there was no effect, I'm already crazy.
now I will not go back, I will be persistent until I learn to use your framwork

Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = 'C:\Users\pcps2\.platformio\penv\Scripts\platformio.exe;C:\Windows\SYSTEM32\python27.zip;c:\users\pcps2\.platformio\penv\DLLs;c:\users\pcps2\.platformio\penv\lib;c:\users\pcps2\.platformio\penv\lib\plat-win;c:\users\pcps2\.platformio\penv\lib\lib-tk;c:\users\pcps2\.platformio\penv\scripts;C:\Python27\Lib;C:\Python27\DLLs;C:\Python27\Lib\lib-tk;c:\users\pcps2\.platformio\penv;c:\users\pcps2\.platformio\penv\lib\site-packages'
  program name = 'C:\Python38\python.exe'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = 'C:\\Python38\\python.exe'
  sys.base_prefix = 'C:\\Python38'
  sys.base_exec_prefix = 'C:\\Python38'
  sys.executable = 'C:\\Python38\\python.exe'
  sys.prefix = 'C:\\Python38'
  sys.exec_prefix = 'C:\\Python38'
  sys.path = [
    'C:\\Users\\pcps2\\.platformio\\penv\\Scripts\\platformio.exe',
    'C:\\Windows\\SYSTEM32\\python27.zip',
    'c:\\users\\pcps2\\.platformio\\penv\\DLLs',
    'c:\\users\\pcps2\\.platformio\\penv\\lib',
    'c:\\users\\pcps2\\.platformio\\penv\\lib\\plat-win',
    'c:\\users\\pcps2\\.platformio\\penv\\lib\\lib-tk',
    'c:\\users\\pcps2\\.platformio\\penv\\scripts',
    'C:\\Python27\\Lib',
    'C:\\Python27\\DLLs',
    'C:\\Python27\\Lib\\lib-tk',
    'c:\\users\\pcps2\\.platformio\\penv',
    'c:\\users\\pcps2\\.platformio\\penv\\lib\\site-packages',
    'C:\\Python38\\python38.zip',
    'C:\\Python38\\DLLs',
    'C:\\Python38\\lib',
    'C:\\Python38',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
  File "c:\users\pcps2\.platformio\penv\lib\encodings\__init__.py", line 123
    raise CodecRegistryError,\
                            ^
SyntaxError: invalid syntax

Current thread 0x00002798 (most recent call first):
<no Python frame>

maakbaas, 2020-09-05:
This seems to be a local issue with your Python installation. See here:

https://stackoverflow.com/questions/5694706/py-initialize-fails-unable-to-load-the-file-system-codec

Since your original issue with the framework seems to be resolved, I am closing this issue.

closed 2020-09-05 by maakbaas