maakbaas/esp8266-iot-framework #35: Prebuild html error

chezzandyto, 2020-09-08:
I Have this error, trying to enable -DREBUILD_HTML flag.

import os
^^^^^^

SyntaxError: Cannot use import statement outside a module

maakbaas, 2020-09-11:
Are you using PlatformIO to import the framework with `lib_deps = ESP8266 IoT Framework`, or do you use a clone of this repository directly?

Did you follow the steps here to install NPM and the required packages?
https://github.com/maakbaas/esp8266-iot-framework/blob/master/docs/getting-started.md#editing-the-web-interface

In case you use the framework through `lib_deps` the comment 'Go to the command line in your project folder, and run the command npm ci to install all required JS packages for the web GUI' should be changed into: 'Go to the command line in the ESP8266 IoT Framework folder that is located within you .pio folder, and run the command npm ci to install all required JS packages for the web GUI'

If this is indeed the issue I will make the documentation more clear.

maakbaas, 2020-09-16:
Any update on this? If not I will close this issue next week.

maakbaas, 2020-09-25:
I will close this issue due to inactivity.

closed 2020-09-25 by maakbaas