maakbaas/esp8266-iot-framework #36: Error with -DREBUILD_CERTS

gl1024, 2020-09-09:
See error message below.  Note that I have openssl installed and in the path.  I installed the asn1crypto python library in order for preBuildCertificates.py to run .  I'm using v1.3.0 of the ESP8266 IOT Framework.  File ca_000.pem is being created in the build directory prior to the error.

Start building certificate store
For all domains
Added: AC Camerfirma, S.A.:AC Camerfirma SA CIF A82743287:http://www.chambersign.org
Traceback (most recent call last):
  File "C:\Users\xxx\Documents\PlatformIO\Projects\yyy\.pio\libdeps\d1_mini_pro\ESP8266 IoT Framework\build\preBuildCertificates.py", line 120, in <module>
    ssl = Popen([openssl,'x509','-inform','PEM','-outform','DER','-out', certName + '.der'], shell = False, stdin = PIPE)
  File "C:\Users\xxx\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\xxx\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified

maakbaas, 2020-09-11:
Thanks for your feedback. You say that you have openSSL in the path. 

Did you also change the path location in `preBuildCertificates.py`?

https://github.com/maakbaas/esp8266-iot-framework/blob/3fbce2dcbb4b3139a5a3324685f75213e1485bb0/build/preBuildCertificates.py#L44

If this is the issue, I can see how that can be overlooked. I can change this to a preprocessor variable in the future so that it can be set from `platformio.ini`.

gl1024, 2020-09-11:
OK, I updated the openssl.
Apparently, I'm seeing a couple problems.  With the V1.3.0 library and no build flags set I'm getting the following errors.

Traceback (most recent call last):
  File "C:\Users\xxx\Documents\PlatformIO\Projects\yyy\.pio\libdeps\d1_mini_pro\ESP8266 IoT Framework\build\preBuildConfig.py", line 4, in <module>
    import inspect, os.path
  File "C:\Users\xxx\.platformio\python3\lib\inspect.py", line 38, in <module>
    import importlib.machinery
  File "C:\Users\xxx\.platformio\python3\lib\importlib\__init__.py", line 51, in <module>
    _w_long = _bootstrap_external._w_long
AttributeError: module 'importlib._bootstrap_external' has no attribute '_w_long'
Found 33 compatible libraries...
**However, it all links run correctly.**

When I add the -DREBUILD_CERTS with openssl updated to my path, I'm now seeing the following.

<SCons.Script.SConscript.SConsEnvironment object at 0x00000133930F0C48>
Traceback (most recent call last):
  File "C:\Users\xxx\Documents\PlatformIO\Projects\yyy\.pio\libdeps\d1_mini_pro\ESP8266 IoT Framework\build\preBuildConfig.py", line 4, in <module>
    import inspect, os.path
  File "C:\Users\xxx\.platformio\python3\lib\inspect.py", line 38, in <module>
    import importlib.machinery
  File "C:\Users\xxx\.platformio\python3\lib\importlib\__init__.py", line 51, in <module>
    _w_long = _bootstrap_external._w_long
AttributeError: module 'importlib._bootstrap_external' has no attribute '_w_long'
Traceback (most recent call last):
  File "C:\Users\xxx\Documents\PlatformIO\Projects\ifttt-d1_mini_pro-sleep_v2\.pio\libdeps\d1_mini_pro\ESP8266 IoT Framework\build\preBuildCertificates.py", line 7, in <module>
    from asn1crypto.x509 import Certificate
  File "C:\Users\xxx\AppData\Local\Programs\Python\Python38-32\lib\site-packages\asn1crypto\x509.py", line 19, in <module>
    from encodings import idna  # noqa
  File "C:\Users\xxx\.platformio\python3\lib\encodings\idna.py", line 3, in <module>
    import stringprep, re, codecs
  File "C:\Users\xxx\.platformio\python3\lib\stringprep.py", line 8, in <module>
    from unicodedata import ucd_3_2_0 as unicodedata
ImportError: DLL load failed while importing unicodedata: %1 is not a valid Win32 application.
Found 33 compatible libraries

maakbaas, 2020-09-11:
Two sources for the errors you are seeing:

https://stackoverflow.com/questions/46509441/module-importlib-bootstrap-has-no-attribute-w-long

https://stackoverflow.com/questions/14629818/importerror-dll-load-failed-1-is-not-a-valid-win32-application

Both seem to relate to potentially mixing python versions in your path, can you check that?

gl1024, 2020-09-11:
I only have ...\Python38\ and ...\Phthon38\Scripts\ in PATH.  From my original post you can see that I didn't have this problem a couple days ago, so I'm not sure what changes.  I also re-started with a clean project file and I'm having the same errors.

gl1024, 2020-09-12:
I was able to resolve my problems.  The VS code platformio IDE updated to V2.0.0 and then 2.0.1 a couple days ago.  It appears that an embedded Python was added along with the update and that was conflicting with my installed version.  Even though I tried to disable the embedded Python I was still having the conflict, so I when ahead and uninstalled Python 3.8.5 and down graded to 3.7.7 which is the same as the embedded and that resolved the  error:
_w_long = _bootstrap_external._w_long
AttributeError: module 'importlib._bootstrap_external' has no attribute '_w_long'

Now, with that fixed, I was able to proceed.  I needed to update the path to openssl in preBuildCertificates.py as you specified above and then the file generation worked.  I also verified the operation of the build flag -DDOMAIN_LIST=maker.ifttt.com for my application.

Some additional instructions would be helpful.  I needed to install asn1crypto for preBuildCertificates.py to run.  Openssl also needs to be installed, which I happened to already have.  It would be much better if the path to openssl could be specified in platform.ini or if possible use PATH.

maakbaas, 2020-09-13:
Thanks for your feedback and the suggestions! I will close this issue for now.

Any chance you are willing to do a pull request with updates to the documentation in the area's where you struggled? Since you just went through this, you would be best placed to describe the solutions to potential issues, and prevent them for others in the future.

gl1024, 2020-09-14:
I looked through the docs again and you did cover the points that I mentioned above excluding the need to install asn1crypto which was obvious when it failed.  Had I re-reviewed them they should have been sufficient.  Thanks for your help.

maakbaas, 2020-09-15:
In the meantime I have received more reports of the same error, so I have opened an [issue](https://github.com/platformio/platformio-vscode-ide/issues/2014) in the PlatformIO IDE repo.

wdmtech, 2020-10-11:
Hey, I've also hit this issueI think this is related - I'm not trying to do anything with certificates, but creating a new project in PlatformIO, adding lib_deps = ESP8266 IoT Framework to platformio.ini, and then building results in the following:

> Executing task in folder esp8266-iot-framework: C:\Users\Will\.platformio\penv\Scripts\platformio.exe run <

Processing esp01_1m (platform: espressif8266; board: esp01_1m; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 (2.6.2) > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing ESP8266 IoT Framework
Library Manager: ESP8266 IoT Framework @ 1.3.2 has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing ESP Async WebServer
Library Manager: ESP Async WebServer @ 1.2.3 has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing ESPAsyncTCP
Library Manager: ESPAsyncTCP @ 1.2.2 has been installed!
Library Manager: Installing AsyncTCP
Library Manager: AsyncTCP @ 1.1.1 has been installed!
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Library Manager: Installing ArduinoJson
Library Manager: ArduinoJson @ 6.16.1 has been installed!
ModuleNotFoundError: No module named 'asn1crypto':
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Will\.platformio\platforms\espressif8266\builder\main.py", line 238:
    target_elf = env.BuildProgram()
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1064:
    lib_builders = env.GetLibBuilders()
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 993:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 61:
    obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 137:
    self.process_extra_options()
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 282:
    exports={"env": self.env, "pio_lib_builder": self},
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Will\Documents\PlatformIO\Projects\esp8266-iot-framework\.pio\libdeps\esp01_1m\ESP8266 IoT Framework\build\preBuild.py", line 9:
    from preBuildCertificates import preBuildCertificatesFun
  File "C:\Users\Will\Documents\PlatformIO\Projects\esp8266-iot-framework\.pio\libdeps\esp01_1m\ESP8266 IoT Framework\build\preBuildCertificates.py", line 7:
    from asn1crypto.x509 import Certificate
==================================================== [FAILED] Took 5.92 seconds ====================================================
The terminal process "C:\Users\Will\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

maakbaas, 2020-10-12:
What PlatformIO version are you using? 

Can you try: 

`pip install asn1crypto`?

wdmtech, 2020-10-12:
It worked! Thanks @maakbaas. I also had to uninstall Python on my host, then I ran `pip install asn1crypto` in the PlatformIO terminal for my project.

Other gotcha was adding WiFiManager as a dependency, I see now it wasn't needed (hope this helps someone)

maakbaas, 2020-10-12:
Great to hear, thanks for the feedback!

aaronse, 2020-10-26:
Encountered, didn't have to uninstall/reinstall Python in my env to fix.  Sharing my experience below.  HTH!

PROBLEM/CAUSE:

Build fails with the following.  See git log 2a19480 for related edits "Travis CI".

ModuleNotFoundError: No module named 'asn1crypto':
  File "C:\users\aaron\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\aaron\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\aaron\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\aaron\.platformio\platforms\espressif8266\builder\main.py", line 238:
    target_elf = env.BuildProgram()
  File "C:\Users\aaron\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)

...

============================================================ [FAILED] Took 1.00 seconds ============================================================ 
The terminal process "C:\Users\aaron\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

FIX:

Open "PlatformIO Core CLI" and enter:

pip install asn1crypto

ENV:

Visual Studio Code

Version: 1.50.1 (user setup)
Commit: d2e414d9e4239a252d1ab117bd7067f125afd80a
Date: 2020-10-13T15:06:15.712Z (1 wk ago)
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19041

maakbaas, 2020-10-26:
Thanks for your comment. Reopening this issue as a reminder to add documentation about installing asn1crypto

maakbaas, 2020-11-06:
Documentation added in 336668211ea74ec1ba770daabb6c6472d9db7b59

closed 2020-11-06 by maakbaas

vertoforce, 2020-12-07:
Had the same problem and no matter what I tried pio kept saying the module wasn't found.  Ended up having to do this

```
source ~/.platformio/penv/bin/activate
pip install asn1crypto
```

That worked.  Hope this helps someone!