maakbaas/esp8266-iot-framework #38: Enable OTA from a separate http server (over your https implementation)

Duckle29, 2020-09-24:
This IoT framework is pretty much exactly what I need for a few of my upcoming projects. It rolls all the different things I need into one coherent package, and I love it.

The only thing I'm missing (and the thing that lead me to your blog post on https) is a reliable (no fingerprints) way to implement OTA from a https server. 

I have a server set up behind some http_basic auth, severed over ssl, that complies with the built in OTA implementation: https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#http-server

I am currently behind on some responsibilities in life, so will have to prioritize those, but looking at your OTA class it seems to be fairly easy to implement a version compatible with the one in the core.

When I get some more time I'll try to implement this and do a PR if it works :)
It seems the general steps to get this to work would be:

1) Compute/add the same headers as the core class. As there's no official server implementation for this, supporting all headers would be best, but most likely not actually required. 
In any case, if the default headers can be populated I'd imagine something like below (with dynamic values changed to the appropriate function calls)
1) Send get request to server with/without auth
1) Read http code to determine if an update is available 
1) If available, stream binary file to littlefs
1) initate ota update with framework requestStart()

Dirty back-of-the napkin example:

```C

// ESP core has a more extensive typedef in the http client: https://github.com/esp8266/Arduino/blob/70e4457041eeb723033fc8011f3d724245d004ae/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.h#L65
// Codes used for this example have been put here
typedef enum {
    HTTP_CODE_OK = 200,
    HTTP_CODE_NOT_MODIFIED = 304,
    HTTP_CODE_UNAUTHORIZED = 401,
    HTTP_CODE_FORBIDDEN = 403,
    HTTP_CODE_NOT_FOUND = 404, 
} t_http_codes;

// if password is left empty, then user is treated as a base64 auth string
int update(String url, String user="", String pass="", String version="")
{
    fetch.begin(url)
    add_headers(version);

    if (user != "")
    {
        if (pass != "")
        {
            fetch.setAuthorization(user.c_str(), pass.c_str());
        }
        else
        {
            fetch.setAuthorization(user.c_str());
        }
    }

    int response = fetch.GET();

    switch(response)
    {
        case HTTP_CODE_OK:
            if (get_binary())
            {
                updater.requestStart('IoTF_otafirmware');
                while(updater.getStatus() != > 1)
                {
                    yield();
                    updater.loop();
                }
                return updater.getStatus();
            }
            break;

        case HTTP_CODE_NOT_MODIFIED:
            // No update needed (current version)
            break;
        
        case HTTP_CODE_UNAUTHORIZED:
            // No auth provided
            break;
        
        case HTTP_CODE_FORBIDDEN:
            // Provided auth was denied
            break;
        
        case HTTP_CODE_NOT_FOUND:
            // Give me the right URL you dumb human
            break;
        
        default:
            // Well shit
            break;
    }
}

// This should be replaced with something that actually gets the correct values
void add_headers(String version)
{
    const uint8_t HEADER_COUNT = 9;

    String http_update_headers [header_cnt][2] = {
        {"HTTP_USER_AGENT", "ESP8266-http-Update"},
        {"HTTP_X_ESP8266_STA_MAC", "00:00:00:00:00:01"},
        {"HTTP_X_ESP8266_AP_MAC", "00:00:00:00:00:10"},
        {"HTTP_X_ESP8266_FREE_SPACE", "10"},
        {"HTTP_X_ESP8266_SKETCH_SIZE", "15"},
        {"HTTP_X_ESP8266_SKETCH_MD5", "feedbeefeatbee5"},
        {"HTTP_X_ESP8266_CHIP_SIZE", "42"},
        {"HTTP_X_ESP8266_SDK_VERSION", "1.3.0"},
        {"HTTP_X_ESP8266_VERSION", version}
    };

    fetch.begin(url, true);

    for (i=0; i<HEADER_COUNT; i++)
    {
        fetch.addHeader(http_update_headers[i][0], http_update_headers[i][1]);
    }
}

bool get_binary()
{
    File f = LittleFS.open('IoTF_otafirmware', w);

    while(fetch.busy())
    {
        if (fetch.available())
        {
            f.write(fetch.read());
        }
    }
    f.close();
    fetch.clean();

    return true;
}
```

Okay, me from further ahead in time. Still a dirty "kinda like this" example, but not quite napkin grade :P

I think ultimately I'd implement this as a separate class, and it definitely needs more error checking, but that's the idea :)

maakbaas, 2020-09-25:
Thanks for your input, I think it is a great idea. I can see how the reliable HTTPS implementation is really an enabler to implement real OTA updates as well. 

It is something I have not yet done in the past, but might be interested in in the future. So I am happy to contribute. I will give it some thought what the best way would be to incorporate this into the framework,

maakbaas, 2020-10-18:
@Duckle29: After giving it some thought, I do not have any objections to incorporate this into the current updater class. What I would think is to add a method that sets the server details, such as the address and auth information, and maybe an interval which which the ESP should poll the update server?

The updater class already has a loop function, so this could then be leveraged to actually check for updates on the defined interval.

Does an architecture like that makes sense? I am curious to hear your thoughts. 

do you have any idea if or when you could potentially do a PR with this function? Otherwise I am willing to take a crack at it, but maybe it is slightly easier for you if you have already an update server setup.

Duckle29, 2020-10-18:
Hey, sounds cool!

I did some initial work on it, and I got it to work... sorta. It would download and flash the binary, but would be soft-bricked from there, needing USB flashing.

I also ran into some issues with the captive portal and wifi connection. It was a bit hit-and miss if it'd connect to wifi, with the button seemingly doing nothing.

The code is pushed here:
https://github.com/Duckle29/esp8266-iot-framework/blob/fetchupdater/src/fetchUpdater.cpp

and is quite messy right now. I'm currently quite busy at uni (currently procrastinating in fact :P) so it's on hold for a week or two at least. You're welcome to give it a go :) I can add some credentials for you on my update server if you want, and you can use that.

maakbaas, 2020-11-13:
I will close this issue for now, since I can't find the time to implement this on my side.

Of course, your pull request will be more than welcome if you want to pursue this further.

closed 2020-11-13 by maakbaas