maakbaas/esp8266-iot-framework #39: Using the web interface at runtime once WiFi configuration is complete

wdmtech, 2020-10-12:
What a wonderful project. 
This is what I've been looking for for months and I never knew it existed. Thank you for all your hard work.

As far as I understand, packages like WiFiManager allow a user to configure their device with their SSID and password, and then the interface/AP goes away. Is it possible to still run the AP (to change settings) even though the WiFi credentials have been added? I have checked the docs but maybe I missed something.

Here's what I'm trying to achieve with your framework:

The idea is that a user can connect to their device's AP via WiFi, it reboots (as normal), and then they can still access the AP in order to configure the toy. 

I've achieved this (in a pretty clunky way!) for now using my own setup, but I'd love to switch over to this project if I can achieve this! (nothing stopping me using it for other projects however 😉)

Thanks again @maakbaas

maakbaas, 2020-10-13:
Looks like a very cool project!

Yes, accessing the web interface after you have set the WiFi details is exactly how this framework is intended. 

The only thing is that you need to find out what the local IP address is for the ESP8266 (This is actually printed to the serial terminal upon startup). Then you can browse to this IP address, and it should serve the same web interface as when the device is AP mode. 

That is based on the assumption that the web interface is what you are after, and not necessarily the access point. If you specifically want the ESP to remain in AP mode as well, that is currently not supported out of the box.

maakbaas, 2020-10-13:
Also, a color picker option for the configuration manager is on my todo list, but feel free to make a pull request with this functionality if this is something you need as well :).

wdmtech, 2020-10-13:
Now I understand, that's ideal for my project! Already doing some hacking on it. Thanks for all the help!

closed 2020-10-13 by wdmtech