maakbaas/esp8266-iot-framework #42: How to store a String as char[32] in configManager

wdmtech, 2020-10-15:
e.g. https://github.com/maakbaas/esp8266-iot-framework/blob/eba7edc4367ffd54b62e9eb03dcb38483d8f2670/examples/configManager/configManagerExample.cpp#L43

But for `char[32]`

Tried my best to figure this out, but totally stuck (I don't know C++) I know this isn't really the right place to ask, but after 2 hours not sure where else to look - this is the closest I've gotten.

```c++
void loop()
{
    //software interrupts
    WiFiManager.loop();
    updater.loop();

    //task A
    if (taskA.previous==0 || (millis() - taskA.previous > taskA.rate ))
    {
        taskA.previous = millis();

        //do task
        // Serial.println(ESP.getFreeHeap());
                
        fetch.GET("http://api.thingspeak.com/channels/1417/field/2/last.txt");

        while (fetch.busy())
        {
            if (fetch.available())
            {
                String s;

                String color = fetch.readString();   
                Serial.println(color); // #FF0000

                char c[32];

                s.toCharArray(c, sizeof(c));

                // Error: expression must be a modifiable lvalue
                configManager.data.dummyColor = s;

                configManager.save();     
            }
        }
            
        fetch.clean();
    }
}
```

wdmtech, 2020-10-16:
Got it! 

```c++
String color = fetch.readString();   

strncpy(configManager.data.dummyColor, color.c_str(), 32);

configManager.save();   
```

closed 2020-10-16 by wdmtech

maakbaas, 2020-10-18:
Sorry for not responding sooner, I am glad that you figured it out :).