maakbaas/esp8266-iot-framework #43: Feature - Realtime display/polling of Config data

wdmtech, 2020-10-16:
Any plans for displaying data in the UI as it arrives?

Had a look at running a websocket server, but ran into issues (I think my device (NodeMCU) is running out of RAM)

In case you're interested, this is what I tried (see commented out lines)

```c++
// main.cpp

#include <Arduino.h>
#include "LittleFS.h"
// #include <WebSocketsServer.h>
#include "WiFiManager.h"
#include "webServer.h"
#include "updater.h"
#include "fetch.h"
#include "configManager.h"
#include "timeSync.h"

struct task
{
    unsigned long rate;
    unsigned long previous;
};

task taskA = {.rate = 5000, .previous = 0};
static String lastColor = "#ffffff";

WebSocketsServer webSocket = WebSocketsServer(81);

void webSocketEvent(uint8_t num, WStype_t type, uint8_t * payload, size_t length) { // When a WebSocket message is received
    Serial.printf("[%u] get Text: %s\n", num, payload);
}

// void startWebSocket() { // Start a WebSocket server
//   webSocket.begin();                          // start the websocket server
//   webSocket.onEvent(webSocketEvent);          // if there's an incomming websocket message, go to function 'webSocketEvent'
//   Serial.println("WebSocket server started.");
// }

void setup()
{
    Serial.begin(115200);

    LittleFS.begin();
    GUI.begin();
    configManager.begin();
    WiFiManager.begin(configManager.data.projectName);
    timeSync.begin();
    // startWebSocket();

    // webSocket.begin();
    // webSocket.onEvent(webSocketEvent);
}

void loop()
{
    //software interrupts
    WiFiManager.loop();
    updater.loop();
    
    // webSocket.loop();
    
    //task A
    if (taskA.previous == 0 || (millis() - taskA.previous > taskA.rate))
    {
        taskA.previous = millis();

        //do task
        // Serial.println(ESP.getFreeHeap());

        fetch.GET("http://api.thingspeak.com/channels/1417/field/2/last.txt");

        while (fetch.busy())
        {
            if (fetch.available())
            {
                String color = fetch.readString();

                // Update dummyColor only when color changes
                if (color != lastColor)
                {
                    lastColor = color;
                    Serial.println(color); // #FF0000
                    strncpy(configManager.data.dummyColor, color.c_str(), 32);
                    configManager.save();
                }
            }
        } 

        fetch.clean();
    }
}
```

```js
// index.js

// ...

import io from 'socket.io-client';

https://socket.io/docs/client-api/
const socket = io(`ws://192.168.1.13:81/`, ['arduino']);

socket.onopen = function () {
    socket.send('Message from Browser to ESP8266 yay its Working!! ' + new Date());
    socket.send('ping');

    setInterval(function () {
        connection.send('Time: ' + new Date());
    }, 20);

    socket.send('Time: ' + new Date());
};

// ...

```

wdmtech, 2020-10-16:
Looks like `ESPAsyncWebServer` has a socket implementation 🤔 didn't have much luck with the above, so will try that instead

maakbaas, 2020-10-18:
I have some experience in another project with the websocket functionality of `ESPAsyncWebServer` and streaming content to the GUI. I think this might be interesting to incorporate, but of course having some kind of generic dashboard in the front end will significantly increase the scope of this framework, which can be a good thing or a bad thing ;).

wdmtech, 2020-10-18:
I was just thinking for people creating IoT projects that read values from a sensor, it would be great to be able to view values in realtime/near realtime, and I saw there are socket server implementations, so thought I'd tfloat the idea - I'd sure be interested in seeing how you've done it

maakbaas, 2020-11-14:
I am close to finishing this. Stay tuned :).

closed 2020-11-17 by maakbaas

maakbaas, 2020-11-18:
The dashboard functionality over websocket is added in v1.6.0. I am curious what you think!

wdmtech, 2020-12-12:
Incredible!

clabnet, 2020-12-12:
Very good !