maakbaas/esp8266-iot-framework #45: Can't build (custom) web interface

patrickomatik, 2020-10-27:
I've followed the instructions closely. I'm building on a Mac fwiw (I had to fix some paths to work in the unixy way, no biggie), and a fresh repo builds fine. I think this is a really exciting project, with the potential to be a fantastic time saver.

The trouble comes when I try to `npm run build` -- I get errors regarding 

> [webpack-cli] Error: ENOENT: no such file or directory, open './dist/index.html.gz'

This seems to be a failure of `webpack.config.js` assuming this file exists; I'm no npm expert and I can't see how/where  index.html.gz might be generated.

Manually creating a file with a copy of the standard index.html doesn't help - it resolves the error only to fail later with:

> Add @babel/preset-react (https://git.io/JfeDR) to the 'presets' section of your Babel config to enable transformation.
> If you want to leave it as-is, add @babel/plugin-syntax-jsx (https://git.io/vb4yA) to the 'plugins' section to enable parsing.

... and I am at a loss as to how to resolve this (believe me, I've tried!) because I'm unfamiliar with npm and friends.

Building from command line or with -D options both give similar results. 

Any ideas, please?

maakbaas, 2020-10-27:
you did run the `npm ci` command to install all the correct modules?

Normally `npm run build` would create this `index.html.gz` file. When I comment the last two lines of `webpack.config.js` to prevent deletion of the build artefacts, I get a dist folder with the files `bundle.js`, `bundle.js.gz` and `index.html.gz`. The reason this file is created in a gzipped version is because of the `CompressionPlugin()` in the webpack config.

Do you get any files in your dist folder at all? And how does your command line output look preceding the error message?

patrickomatik, 2020-10-29:
Thanks for the speedy response, most appreciated.
I have run `npm ci`, and it gave an error I corrected using the suggested `--legacy-peer-deps`. 

Here's the output from `npm ci`

> ➜  ESP8266 IoT Framework git:(master) ✗ npm ci
npm ERR! code ERESOLVE
npm ERR! ERESOLVE unable to resolve dependency tree
npm ERR!
npm ERR! Found: compression-webpack-plugin@6.0.4
npm ERR! node_modules/compression-webpack-plugin
npm ERR!   dev compression-webpack-plugin@"^3.1.0" from the root project
npm ERR!
npm ERR! Could not resolve dependency:
npm ERR! dev compression-webpack-plugin@"^3.1.0" from the root project
npm ERR!
npm ERR! Fix the upstream dependency conflict, or retry
npm ERR! this command with --force, or --legacy-peer-deps
npm ERR! to accept an incorrect (and potentially broken) dependency resolution.
npm ERR!
npm ERR! See /Users/patrick/.npm/eresolve-report.txt for a full report.

The output from `npm run build` (after running `npm ci --legacy-peer-deps`) is:

> > esp@1.0.0 build
> webpack --mode=production
Doing webby stuff
Error: ENOENT: no such file or directory, open './dist/index.html.gz'
    at Object.openSync (node:fs:476:3)
    at Object.readFileSync (node:fs:377:35)
    at done (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/webpack.config.js:110:37)
    at AsyncSeriesHook.eval [as callAsync] (eval at create (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/tapable/lib/HookCodeFactory.js:33:10), <anonymous>:18:1)
    at AsyncSeriesHook.lazyCompileHook (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/tapable/lib/Hook.js:154:20)
    at onCompiled (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/webpack/lib/Compiler.js:271:21)
    at /Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/webpack/lib/Compiler.js:681:15
    at AsyncSeriesHook.eval [as callAsync] (eval at create (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/tapable/lib/HookCodeFactory.js:33:10), <anonymous>:6:1)
    at AsyncSeriesHook.lazyCompileHook (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/tapable/lib/Hook.js:154:20)
    at /Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/webpack/lib/Compiler.js:678:31
    at AsyncSeriesHook.eval [as callAsync] (eval at create (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/tapable/lib/HookCodeFactory.js:33:10), <anonymous>:6:1)
    at AsyncSeriesHook.lazyCompileHook (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/tapable/lib/Hook.js:154:20)
    at /Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/webpack/lib/Compilation.js:1423:35
    at AsyncSeriesHook.eval [as callAsync] (eval at create (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/tapable/lib/HookCodeFactory.js:33:10), <anonymous>:6:1)
    at AsyncSeriesHook.lazyCompileHook (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/tapable/lib/Hook.js:154:20)
    at /Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/webpack/lib/Compilation.js:1414:32
    at eval (eval at create (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/tapable/lib/HookCodeFactory.js:33:10), <anonymous>:11:1)
    at /Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/terser-webpack-plugin/dist/index.js:321:9
    at step (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/terser-webpack-plugin/dist/TaskRunner.js:87:9)
    at /Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/terser-webpack-plugin/dist/TaskRunner.js:111:15
    at tryCatcher (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/bluebird/js/release/util.js:16:23)
    at Promise._settlePromiseFromHandler (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/bluebird/js/release/promise.js:547:31)
    at Promise._settlePromise (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/bluebird/js/release/promise.js:604:18)
    at Promise._settlePromise0 (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/bluebird/js/release/promise.js:649:10)
    at Promise._settlePromises (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/bluebird/js/release/promise.js:729:18)
[... cut ...]
    at Promise._settlePromise0 (/Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/node_modules/bluebird/js/release/promise.js:649:10)
npm ERR! code 1
npm ERR! path /Users/patrick/Documents/PlatformIO/Projects/PROJECTNAME/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework
npm ERR! command failed
npm ERR! command sh -c webpack --mode=production

The `dist` directory is empty. `Doing webby stuff` is a console log line I added to webpack.config.js on line 103 to check the production html code stuff is really running.

Many thanks for any insights you can give!

maakbaas, 2020-10-29:
Can you remove the node_modules folder, and retry with `npm install` instead of `npm ci`?

patrickomatik, 2020-10-29:
I removed node_modules, and re-ran as above - legacy flag required again, same output as above.

I switched to a Ubuntu machine, and tried to compile one of the examples - it works, but there is an error from Sconscript:

> Missing Sconcript `.pio/libdeps/nodemcu2v/ESP8266 IoT Framework/build/preBuild.py`

And indeed, the `build` directory is missing, along with `docs` -- platformio doesn't seem to install them. This is with vscode 1.50.1 and platformio 2.1.3

So I think there's something wrong with the brew/npm environment on the Mac, and platformio behaves differently between mac and linux platforms resulting in a broken build under linux? I will experiment further with the linux environment to see if I can hack a build that correctly applies preBuild.py

patrickomatik, 2020-10-29:
Ha! I think I have made some progress. The root cause of the problem may lie in the use of spaces within the library name "ESP8266 IoT Framework".  Creating a symbolic link from `ESP8266 IoT Framework` to `ESP8266_IoT_Framework` and adding the build flag `DPROJECT_DIR=/home/patrick/Documents/Arduino/iot_test2/.pio/libdeps/nodemcuv2/ESP8266_IoT_Framework/build` to `platformio.ini` gets significantly further in the build process, failing with:

>  ERROR in Entry module not found: Error: Can't resolve './src' in '/home/patrick/Documents/Arduino/iot_test2/.pio/libdeps/nodemcuv2/ESP8266 IoT Framework'

which does of course exist, but needs presumably to be correctly escaped to handle the spaces. Is that a major re-factor?

maakbaas, 2020-10-30:
Sorry, for all the issues you are facing, and thanks for your patience. To make sure it is due to the space in the library name:

Can you try to:
1. Clone this repository to your computer 
2. Take one of the example .c files and add it to the src folder
3. Do a build

In this case you use the project as a main project, and not as a platformio library. This should remove the spaces in your path. There is also work ongoing to make this library compatible with the ESP32 after which a name change will be needed anyway. At this point I will also remove the spaces from the name if this indeed causes the issue on mac :).

patrickomatik, 2020-10-30:
Success - under Ubuntu:

1. Create empty VScode platformio project
2. cd lib
3.  git clone https://github.com/maakbaas/esp8266-iot-framework.git
4.  cd esp8266-iot-framework
5.  npm ci
6. cd .. (back to lib dir again) - add dependencies
7.  git clone https://github.com/me-no-dev/ESPAsyncWebServer.git
8.  git clone https://github.com/me-no-dev/ESPAsyncTCP.git
9.  git clone https://github.com/bblanchon/ArduinoJson.git
10. cp esp8266-iot-framework/examples/helloWorld.cpp ../../../src
11. Verify helloWorld build succeeds 
12. To customise web config, add iot_config.json to src directory and add
build_flags = -DCONFIG_PATH=src/iot_config.json -DREBUILD_HTML 
to platformio.ini and rebuild should proceed without error

PS. Given that .pio isn't under source control by default, and github libraries do come and go over time, I think this method is to be preferred over the use of lib_deps because the code can be rebuilt from just the local git repo if required?

Feel free to reuse any of above as you may wish, many thanks for the help.

maakbaas, 2020-10-30:
Glad that you got it working. Although in this way you lose some of the convenience benefits of PlatformIO of course ;).

PlatformIO is actually there to do the configuration management, but indeed with your method you have everything fully local and you can employ your own strategies for managing the libraries.

maakbaas, 2020-11-10:
See this discussion: https://github.com/platformio/platformio-core/issues/3730

maakbaas, 2020-11-12:
fixed in 9d3664885aee9a6eaf5ee07eec1db2f17d3083eb

closed 2020-11-12 by maakbaas