maakbaas/esp8266-iot-framework #46: Setting Up fixed IP Address

marcolino7, 2020-11-04:
Hi,
reading documentation is not clear how to setup a fixed IP address, from the code or from web interface. Is ti possible to do?

Thanks

maakbaas, 2020-11-06:
This is indeed not yet supported. Is it needed for your use case? You are definitely encouraged to add this in a pull request, please let me know if you are open to that.

marcolino7, 2020-11-06:
Hi @maakbaas 
thanks for reply. I'm stuck on another issue. I am not able to make MQTT working together with esp8266-iot-framework. I use PubSubClient.h and get compilation errore with async webserver adding this library. Do you have some example with MQTT?

Marco

maakbaas, 2020-11-12:
This will be included in the next release.

PR #52 (2020-11-12): Static IP address configuration
Adds static IP configuration to the web interface. Add an internal data block to EEPROM, and add a checksum mechanism to make future changes more robust.

Updating to this version will invalidate your existing configuration data. Hopefully this is a one time thing.

Closes #46.
closed 2020-11-12 by maakbaas