maakbaas/esp8266-iot-framework #48: Error Compiling Trying to include MQTT library

marcolino7, 2020-11-07:
Hi,
I am trying to develop my IOT node. I would like to use this powerful framework, and MQTT.
Just adding PubSubClient.h to projet, I got this compiling:

`Linking .pio\build\esp12e\firmware.elf
c:/users/administrator/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\lib9d4\libESP Async WebServer.a(WebRequest.cpp.o):(.text._ZN21AsyncWebServerRequest21requestAuthenticationEPKcb+0x10): undefined reference to `requestDigestAuthentication(char const*)'
c:/users/administrator/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\lib9d4\libESP Async WebServer.a(WebRequest.cpp.o): in 
function `AsyncWebServerRequest::requestAuthentication(char const*, bool)':
WebRequest.cpp:(.text._ZN21AsyncWebServerRequest21requestAuthenticationEPKcb+0xa8): undefined reference to `requestDigestAuthentication(char const*)'
c:/users/administrator/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\lib9d4\libESP Async WebServer.a(WebRequest.cpp.o):(.text._ZN21AsyncWebServerRequest12authenticateEPKcS1_S1_b+0x0): undefined reference to `checkDigestAuthentication(char const*, char const*, char const*, char const*, char const*, bool, char const*, char const*, char const*)'
c:/users/administrator/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\lib9d4\libESP Async WebServer.a(WebRequest.cpp.o):(.text._ZN21AsyncWebServerRequest12authenticateEPKcS1_S1_b+0x4): undefined reference to `checkBasicAuthentication(char const*, char const*, char const*)'
c:/users/administrator/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\lib9d4\libESP Async WebServer.a(WebRequest.cpp.o): in 
function `AsyncWebServerRequest::authenticate(char const*, char const*, char const*, bool)':
WebRequest.cpp:(.text._ZN21AsyncWebServerRequest12authenticateEPKcS1_S1_b+0x5a): undefined reference to `checkDigestAuthentication(char const*, char const*, char const*, char const*, char const*, bool, char const*, char const*, char const*)'
c:/users/administrator/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: WebRequest.cpp:(.text._ZN21AsyncWebServerRequest12authenticateEPKcS1_S1_b+0x72): undefined reference to `checkBasicAuthentication(char const*, char const*, char const*)'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp12e\firmware.elf] Error 1`

It is very frustrating, because I love this framewort and have the Web Portal to update firmware and WiFi connection very simple, but if is not possible to use MQTT this would be a problem for iot-framework.
I am not skilled to fix this issue, can anyone helpme on how to fix?

Thanks

maakbaas, 2020-11-07:
I did a quick test wit the empty helloWorld example, and added 

#include <PubSubClient.h>

This builds succesfully, so I don't think the issue is with this framework.

marcolino7, 2020-11-08:
Hi,
Thanks for answer. I created a fresh project and installed only framework and PubSubClient and now compile.
I got an Alert:

```
Compiling .pio\build\esp12e\libfb2\ESP Async WebServer\WebHandlers.cpp.o
In file included from .pio\libdeps\esp12e\ESP Async WebServer\src\SPIFFSEditor.cpp:1:0:
.pio\libdeps\esp12e\ESP Async WebServer\src\SPIFFSEditor.h:16:101: warning: 'SPIFFS' is deprecated (declared at C:\Users\Administrator\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:269): 
SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
     SPIFFSEditor(const String& username=String(), const String& password=String(), const fs::FS& fs=SPIFFS);
                                                                                                     ^
Compiling .pio\build\esp12e\libfb2\ESP Async WebServer\WebRequest.cpp.o

```
Regarding SPIFF, but as I read you use LittleFS. I got this Alert also without PubSubClient.

Marco

maakbaas, 2020-11-08:
Yes, indeed. This alert is caused by an internal file of Async Webserver, but the framework itself uses littleFS. Glad you got it working!

closed 2020-11-08 by maakbaas