maakbaas/esp8266-iot-framework #49: Is't ESP01 compatible ?

MohammedBENHADINE, 2020-11-09:
Greetings,
First, thanks for this great repo.
I'm trying to use ESP01 for some HTTPS requests , and i came across this platform , after wraping all , i got this issue with the HelloWorld example : 
`Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266 IoT Framework> 1.3.2  
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.2    
|   |   |   |-- <ESP8266WiFi> 1.0  
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0      
|   |   |-- <ArduinoJson> 6.17.1   
|   |-- <ArduinoJson> 6.17.1       
|   |-- <ESP8266WiFi> 1.0
|   |-- <EEPROM> 1.0
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <LittleFS(esp8266)> 0.1.0
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|-- <LittleFS(esp8266)> 0.1.0
Building in release mode
Linking .pio\build\esp01\firmware.elf
c:/users/mtrd/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp01\firmware.elf section `.irom0.text' will not fit in region `irom0_0_seg'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp01\firmware.elf] Error 1
=============================================================================== [FAILED] Took 5.01 seconds ===============================================================================
The terminal process "C:\Users\mtrd\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.`
It seems like a memory leak issue , is that an expected behaviour ?

Regards,
Mohammed

maakbaas, 2020-11-10:
how does your platformio.ini look? It seems as if the configuration for memory sections is not ok.

MohammedBENHADINE, 2020-11-10:
It looks as follow : 
`[env:esp01]
platform = espressif8266
framework = arduino
board = esp01
monitor_speed = 115200`

maakbaas, 2020-11-10:
I see the esp01 has 512kB of flash? There is also a version with 1MB of flash. If you have this one, you should use:

`board = esp01_1m`

instead. If I do a quick test, the 512kB is indeed not enough, and will not be compatible with this framework. But the version with 1MB might be enough.

closed 2020-11-10 by maakbaas

MohammedBENHADINE, 2020-11-10:
Thanks , it's working fine now.

clabnet, 2020-11-28:
Working fine on ESP-01 1Mb.