maakbaas/esp8266-iot-framework #50: Build fails with various "No such file or directory errors"

supermar1010, 2020-11-09:
Hello,

I tried using the framework, but failed pretty soon. I made a very small demo, it's just a plain platformio project with the library added.

My platformio.ini:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino

monitor_speed = 115200

lib_deps = ESP8266 IoT Framework
```

My main.cpp

```cpp
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

Here is the error message I'm getting upon executing build.

<details>
  <summary>platfromio build output</summary>
  
 > Executing task: platformio run --environment nodemcuv2 <

Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (2.6.2) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=0 to SConscript calls.
Missing SConscript '.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/build/preBuild.py'
File "/home/mario/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/piolib.py", line 280, in process_extra_options
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266 IoT Framework> 1.3.2
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <Hash> 1.0
|   |   |-- <ArduinoJson> 6.17.1
|   |-- <ArduinoJson> 6.17.1
Building in release mode
Compiling .pio/build/nodemcuv2/lib94a/ESP8266 IoT Framework/WiFiManager.cpp.o
.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/src/WiFiManager.cpp:5:25: fatal error: ESP8266WiFi.h: No such file or directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266WiFi.h
*
*********************************************************************

 #include <ESP8266WiFi.h>
                         ^
compilation terminated.
Compiling .pio/build/nodemcuv2/lib94a/ESP8266 IoT Framework/certStore.cpp.o
Compiling .pio/build/nodemcuv2/lib94a/ESP8266 IoT Framework/configManager.cpp.o
Compiling .pio/build/nodemcuv2/lib94a/ESP8266 IoT Framework/fetch.cpp.o
.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/src/configManager.cpp:1:20: fatal error: EEPROM.h: No such file or directory

****************************************************************
* Looking for EEPROM.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:EEPROM.h"
* Web  > https://platformio.org/lib/search?query=header:EEPROM.h
*
****************************************************************

 #include <EEPROM.h>
                    ^
compilation terminated.
Compiling .pio/build/nodemcuv2/lib94a/ESP8266 IoT Framework/generated/config.cpp.o
Compiling .pio/build/nodemcuv2/lib94a/ESP8266 IoT Framework/timeSync.cpp.o
Compiling .pio/build/nodemcuv2/lib94a/ESP8266 IoT Framework/updater.cpp.o
In file included from .pio/libdeps/nodemcuv2/ESP8266 IoT Framework/src/certStore.cpp:1:0:
.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/src/certStore.h:9:28: fatal error: BearSSLHelpers.h: No such file or directory

************************************************************************
* Looking for BearSSLHelpers.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:BearSSLHelpers.h"
* Web  > https://platformio.org/lib/search?query=header:BearSSLHelpers.h
*
************************************************************************

 #include <BearSSLHelpers.h>
                            ^
compilation terminated.
In file included from .pio/libdeps/nodemcuv2/ESP8266 IoT Framework/src/fetch.cpp:1:0:
.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/src/fetch.h:5:31: fatal error: ESP8266HTTPClient.h: No such file or directory

***************************************************************************
* Looking for ESP8266HTTPClient.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266HTTPClient.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266HTTPClient.h
*
***************************************************************************

 #include <ESP8266HTTPClient.h>
                               ^
compilation terminated.
.pio/libdeps/nodemcuv2/ESP8266 IoT Framework/src/updater.cpp:3:22: fatal error: LittleFS.h: No such file or directory

******************************************************************
* Looking for LittleFS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:LittleFS.h"
* Web  > https://platformio.org/lib/search?query=header:LittleFS.h
*
******************************************************************

 #include "LittleFS.h"
                      ^
compilation terminated.
Compiling .pio/build/nodemcuv2/lib94a/ESP8266 IoT Framework/webServer.cpp.o
Compiling .pio/build/nodemcuv2/src/main.cpp.o
Generating LD script .pio/build/nodemcuv2/ld/local.eagle.app.v6.common.ld
Compiling .pio/build/nodemcuv2/lib5b9/ESPAsyncTCP/AsyncPrinter.cpp.o
In file included from .pio/libdeps/nodemcuv2/ESP8266 IoT Framework/src/webServer.h:4:0,
                 from .pio/libdeps/nodemcuv2/ESP8266 IoT Framework/src/webServer.cpp:1:
.pio/libdeps/nodemcuv2/ESP Async WebServer/src/ESPAsyncWebServer.h:35:25: fatal error: ESP8266WiFi.h: No such file or directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266WiFi.h
*
*********************************************************************

 #include <ESP8266WiFi.h>
                         ^
compilation terminated.
*** [.pio/build/nodemcuv2/lib94a/ESP8266 IoT Framework/WiFiManager.cpp.o] Error 1
*** [.pio/build/nodemcuv2/lib94a/ESP8266 IoT Framework/configManager.cpp.o] Error 1
*** [.pio/build/nodemcuv2/lib94a/ESP8266 IoT Framework/certStore.cpp.o] Error 1
*** [.pio/build/nodemcuv2/lib94a/ESP8266 IoT Framework/fetch.cpp.o] Error 1
*** [.pio/build/nodemcuv2/lib94a/ESP8266 IoT Framework/updater.cpp.o] Error 1
Compiling .pio/build/nodemcuv2/lib5b9/ESPAsyncTCP/ESPAsyncTCP.cpp.o
*** [.pio/build/nodemcuv2/lib94a/ESP8266 IoT Framework/webServer.cpp.o] Error 1
================================================================================= [FAILED] Took 1.27 seconds =================================================================================
The terminal process "platformio 'run', '--environment', 'nodemcuv2'" terminated with exit code: 1.
</details>

Do you have any tip how to resolve this issue?

Btw iI'm new to platformio so I might have missed something there, though I wouldn't know what.

maakbaas, 2020-11-10:
Can you start from the helloWorld.c example provided?

Your issue is interesting, and I can reproduce it. If you add below `#include <Arduino.h>` the lines:

```c
#include "WiFiManager.h"
#include "fetch.h"
#include "configManager.h"
```
The build will succeed as well. It seems that these framework files have a dependency on `Arduino.h` which is not included explicitly. I will fix this issue later. Thanks for reporting!

maakbaas, 2020-11-10:
See also https://github.com/maakbaas/esp8266-iot-framework/issues/45#issuecomment-724492461 for the first warning in your error log.

supermar1010, 2020-11-10:
Thanks for your quick reply :) and the workaround

maakbaas, 2020-11-12:
fixed in 8c12a29434c9b951cd9099033ccee27be4f3c248

closed 2020-11-12 by maakbaas