maakbaas/esp8266-iot-framework #51: Starting Using Configuration Manager

marcolino7, 2020-11-12:
Ciao,
I am a new developer and I am not skilled on this kind of stuff. I have created my basic node using WiFi Manager, and now I would like to move to next level. I would like to use configuration manager and my config parameters and read them from my code.
I have not yet undestood how start with it, edit out of the box filed and add some new parameter.
Can you give me some suggestion on this side?

Thanks
Marco

maakbaas, 2020-11-12:
All these parts are described in the documentation. Let me know what you have tried so far and where you run into issues.

marcolino7, 2020-11-12:
Hi,
I use PlatformIO in VSCode, i installed esp8266-iot-framework via PlatformIO Library Manager. Then in the path .pio\libdeps\esp12e\ESP8266 IoT Framework\ i created a folder called build and copied inside all py script contained in scripts folder. The I created into the root folder of the projet a file called configuration-filename.json and added to platformio.ini the build_flags = -DCONFIG_PATH=configuration_filename.json parameter.
Compiling I got some warning and compilation failed:

```
PLATFORM: Espressif 8266 (2.6.2) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
ModuleNotFoundError: No module named 'asn1crypto':
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Administrator\.platformio\platforms\espressif8266\builder\main.py", line 238:
    target_elf = env.BuildProgram()
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 880:
    if _is_builtin(spec):
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 872:
    for lb in self.env.GetLibBuilders():
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 993:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 61:
    obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 137:
    self.process_extra_options()
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 282:
    exports={"env": self.env, "pio_lib_builder": self},
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Administrator\Documents\Privati\Elettronica\PlatformIO\NodoSalottoDX\.pio\libdeps\esp12e\ESP8266 IoT Framework\build\preBuild.py", line 9:
    from preBuildCertificates import preBuildCertificatesFun
  File "C:\Users\Administrator\Documents\Privati\Elettronica\PlatformIO\NodoSalottoDX\.pio\libdeps\esp12e\ESP8266 IoT Framework\build\preBuildCertificates.py", line 7:
    from asn1crypto.x509 import Certificate
```

Where I miss something?

THanks

maakbaas, 2020-11-12:
You need to install the asn1crypto Python module as described here:

https://github.com/maakbaas/esp8266-iot-framework/blob/master/docs/fetch.md#code-generation

(if you use 1.4.0 which I released today, you no longer need to copy the build folder manually. This folder is now renamed to scripts and will be included in the PlatformIO library directly.)

marcolino7, 2020-11-12:
Hi,
I installed asn1crypto but still same error. PlatformIO use Python installed on my PC or a versione installed with Platformio Itsefl?

Marco

maakbaas, 2020-11-12:
version installed from PlatformIO itself, so you need to open a pio terminal to execute the pip command.

marcolino7, 2020-11-12:
Hi,
done thanks, now got this:

`FileNotFoundError: [Errno 2] No such file or directory: '../gui/js/configuration.json':`

need I to create folder and create files?
Marco

maakbaas, 2020-11-12:
can you delete your .pio folder, and try again with a clean pull of the latest version 1.4.0 of the framework?

marcolino7, 2020-11-12:
@maakbaas 
thanks a lot for support, now it use my configutation file.
Last 2 questions:

1) I suppose i have to use this syntax to retrieve the values: `configManager.data.projectName`
2) is It possible to add a custom button and fire a function, for example if I want to restart board or toggle a relè from web interface?

Thanks

maakbaas, 2020-11-12:
I'm happy you got it working.

1. This is correct
2. This is not yet possible, but is one of the features that might get added in the future, potentially together with #43.

closed 2020-11-12 by maakbaas

marcolino7, 2020-11-12:
@maakbaas 
great news.
When I will finish to develop a couple of node for my HomeAutomation System, I will try to chec if I am able to add static IP config to your code, in case then I will create a pull request to add mu code.
Thanks for quick and usefull support

Marco

maakbaas, 2020-11-12:
Haha, I just added this 5 minutes ago in #52  ;). It will be part of the next release.

marcolino7, 2020-11-12:
and when it will be released?

maakbaas, 2020-11-12:
I am planning to work on some other things, so not sure yet. 

You can already try it out by changing your `platformio.ini` to:

`lib_deps = https://github.com/maakbaas/esp8266-iot-framework.git`

Let me know what you think!

marcolino7, 2020-11-12:
ah ok, doing this it will use the developmente version?
better do delete .pio folder and download all libraries again?

marcolino7, 2020-11-12:
Sorry I still have a problem with configuration.
I changed my file to:

```
[
    {
        "name": "projectName",
        "label": "MQTT Node Name",
        "type": "char",
        "length": 32,
        "value": "ESP8266 IoT Framework 123"
    },
    {
        "name": "mqttserverip",
        "label": "MQTT Server Address",
        "type": "char",
        "length": 20,
        "value": ""
    },
    {
        "name": "mqttserverport",
        "type": "uint16_t",
        "value": 1883
    }
]
```
But changes do not reflect on webpage
[image]

maakbaas, 2020-11-12:
Please follow the instructions:

https://github.com/maakbaas/esp8266-iot-framework/blob/master/docs/config-manager.md#code-generation

Look at the note that says 'Important'.

marcolino7, 2020-11-13:
Hi just have installed npm on my laptop for working purpose.
In ran npm ci and then npm run build and i got this error:

```
Insufficient number of arguments or no entry found.
Alternatively, run 'webpack(-cli) --help' for usage info.

Hash: 3633fb7c0144b7793e91
Version: webpack 4.42.1
Time: 178ms
Built at: 2020-11-13 12:59:49 AM

ERROR in Entry module not found: Error: Can't resolve './src' in 'C:\Users\Administrator\Documents\Privati\Elettronica\PlatformIO\NodoSalottoDX\.pio\libdeps\esp12e\ESP8266 IoT Framework'
npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! esp@1.0.0 build: `webpack --mode=production`
npm ERR! Exit status 2
npm ERR!
npm ERR! Failed at the esp@1.0.0 build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Administrator\AppData\Roaming\npm-cache\_logs\2020-11-12T23_59_49_223Z-debug.log
```

I run both command in library's root folder .pio\libdeps\esp12e\ESP8266 IoT Framework

maakbaas, 2020-11-13:
Sorry, I can't reproduce this on my side, not sure what is going on.

marcolino7, 2020-11-13:
Seems it look for './src' in .pio\libdeps\esp12e\ESP8266 IoT Framework, but ./src is not present, is present /src instead

marcolino7, 2020-11-13:
Here is the log produced by npm

```
0 info it worked if it ends with ok
1 verbose cli [
1 verbose cli   'C:\\Program Files\\nodejs\\node.exe',
1 verbose cli   'C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js',
1 verbose cli   'run',
1 verbose cli   'build'
1 verbose cli ]
2 info using npm@6.14.8
3 info using node@v14.15.0
4 verbose run-script [ 'prebuild', 'build', 'postbuild' ]
5 info lifecycle esp@1.0.0~prebuild: esp@1.0.0
6 info lifecycle esp@1.0.0~build: esp@1.0.0
7 verbose lifecycle esp@1.0.0~build: unsafe-perm in lifecycle true
8 verbose lifecycle esp@1.0.0~build: PATH: C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin;C:\Users\Administrator\Documents\Privati\Elettronica\PlatformIO\NodoSalottoDX\.pio\libdeps\esp12e\ESP8266 IoT Framework\node_modules\.bin;C:\Users\Administrator\.platformio\penv\Scripts;C:\Users\Administrator\.platformio\penv;C:\Users\Administrator\.platformio\python3;C:\Program Files (x86)\Lenovo\FusionEngine;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Common Files\lenovo\easyplussdk\bin;C:\Program Files (x86)\ATI Technologies\ATI.ACE\Core-Static;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Calibre2\;C:\Users\Administrator\.dnx\bin;C:\Program Files\Microsoft DNX\Dnvm\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\Client SDK\ODBC\130\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\140\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\140\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL Server\140\Tools\Binn\ManagementStudio\;D:\PerlSrtawberry\c\bin;D:\PerlSrtawberry\perl\site\bin;D:\PerlSrtawberry\perl\bin;C:\Program Files\Git\cmd;C:\Program Files (x86)\Microsoft SQL Server\110\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL Server\130\DTS\Binn\;C:\Program Files\TortoiseGit\bin;C:\Program Files (x86)\AMD\ATI.ACE\Core-Static;C:\Program Files (x86)\GitExtensions\;C:\Program Files\nodejs\;C:\ProgramData\chocolatey\bin;C:\Users\Administrator\AppData\Local\Microsoft\WindowsApps;;C:\Program Files\Microsoft VS Code\bin;C:\Users\Administrator\AppData\Local\Programs\Fiddler;C:\Users\Administrator\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\Administrator\AppData\Roaming\npm;C:\Program Files (x86)\Lenovo\FusionEngine;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Common Files\lenovo\easyplussdk\bin;C:\Program Files (x86)\ATI Technologies\ATI.ACE\Core-Static;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Calibre2\;C:\Users\Administrator\.dnx\bin;C:\Program Files\Microsoft DNX\Dnvm\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\Client SDK\ODBC\130\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\140\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\140\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL Server\140\Tools\Binn\ManagementStudio\;D:\PerlSrtawberry\c\bin;D:\PerlSrtawberry\perl\site\bin;D:\PerlSrtawberry\perl\bin;C:\Program Files\Git\cmd;C:\Program Files (x86)\Microsoft SQL Server\110\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL Server\130\DTS\Binn\;C:\Program Files\TortoiseGit\bin;C:\Program Files (x86)\AMD\ATI.ACE\Core-Static;C:\Program Files (x86)\GitExtensions\;C:\Program Files\nodejs\;C:\ProgramData\chocolatey\bin;C:\Users\Administrator\AppData\Local\Microsoft\WindowsApps;;C:\Program Files\Microsoft VS Code\bin;C:\Users\Administrator\AppData\Local\Programs\Fiddler;C:\Users\Administrator\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\Administrator\AppData\Roaming\npm
9 verbose lifecycle esp@1.0.0~build: CWD: C:\Users\Administrator\Documents\Privati\Elettronica\PlatformIO\NodoSalottoDX\.pio\libdeps\esp12e\ESP8266 IoT Framework
10 silly lifecycle esp@1.0.0~build: Args: [ '/d /s /c', 'webpack --mode=production' ]
11 silly lifecycle esp@1.0.0~build: Returned: code: 2  signal: null
12 info lifecycle esp@1.0.0~build: Failed to exec build script
13 verbose stack Error: esp@1.0.0 build: `webpack --mode=production`
13 verbose stack Exit status 2
13 verbose stack     at EventEmitter.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:315:20)
13 verbose stack     at ChildProcess.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\lib\spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:315:20)
13 verbose stack     at maybeClose (internal/child_process.js:1048:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
14 verbose pkgid esp@1.0.0
15 verbose cwd C:\Users\Administrator\Documents\Privati\Elettronica\PlatformIO\NodoSalottoDX\.pio\libdeps\esp12e\ESP8266 IoT Framework
16 verbose Windows_NT 10.0.18363
17 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "run" "build"
18 verbose node v14.15.0
19 verbose npm  v6.14.8
20 error code ELIFECYCLE
21 error errno 2
22 error esp@1.0.0 build: `webpack --mode=production`
22 error Exit status 2
23 error Failed at the esp@1.0.0 build script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 2, true ]
```

maakbaas, 2020-11-13:
I finally was able to reproduce the issue, and fixed it, thanks for reporting.

Please upgrade to the latest version v1.5.0 (which also contains the static IP settings), and make sure you have PlatformIO 5.0.3.

marcolino7, 2020-11-13:
@maakbaas 
thanks, I will update in next hours after work and I will let you know

Marco

marcolino7, 2020-11-13:
Hi @maakbaas 
get code compile and web interface is working great.
Regarding fixed IP how should work?

I can't see parameters in config.json

marco

marcolino7, 2020-11-13:
Hi,
all work great and I have my updated interface.
I also tried fixed IP, it work and at reboot ESP take fixed IP, But opening Wifi Configuration, DHCP il still checked:

[image]

and fixed ip is not shown if i remove the checkbox:
[image]

let me know If I have to open a new issue

Thanks

maakbaas, 2020-11-13:
This is intended behavior. I can see how it can be confusing, but these form fields are under the header 'update credentials', and are therefore only used to enter new details. They will not show existing details.

Glad you finally got it working. Thanks for your patience working through the issues.

marcolino7, 2020-11-13:
@maakbaas 
perhaps i  were not clear in my description. Fixet IP settings is not persistent. When I turn off ESP8266 board and restart, I take DHCP again. Is this intended behaviour?

Marco

maakbaas, 2020-11-13:
did you also enter the subnet and the gateway? Right now you need to enter all three for it to be persistent. I might change this in a future update if that makes more sense.

marcolino7, 2020-11-13:
yes I added all 3 values, when click on save ESP Board start with new fixed IP. If i power off and power on again, it roll back to DHCP and ignore Fixed IP.

Marco

maakbaas, 2020-11-13:
can you share the values you entered, and then show here the ESP serial output for the moment when you set the new IP, and then also the serial output when you reset the ESP?

marcolino7, 2020-11-13:
yes sure
here is values I insert:
[image]

when i click save, got this in terminal
```
New connection unsuccessful
Reconnection successful
192.168.1.43
```

And I got this when i Turn Off and On the node
```
Connected to stored WiFi details
192.168.1.236
```

maakbaas, 2020-11-13:
For some reason it does not succeed to connect when you enter the new details

```
New connection unsuccessful
Reconnection successful
```

In my case the same looks like this:

```
New connection successful
```

It only saves the new IP when the connection is successful. Can you try if it makes a difference if you `forget` the WiFi details and then enter the static IP from the captive portal?

marcolino7, 2020-11-13:
Yes I got Nwe connection unsucceful, but then after Reconnection successful, i got ESP on fixed IP until reboot

From captive prtal seems to work.
This is after setting uf wifi and network detail:
```
New connection successful
192.168.1.143
```
and this after rebbot of the node:
```
Using static IP
Connected to stored WiFi details
192.168.1.143
```

maakbaas, 2020-11-13:
maybe a stupid question, but when you entered the IP address when you were already connected, did you also enter the ssid and pass again (this would be needed)

marcolino7, 2020-11-13:
This is operation to reproduce issue:

- New node with captive portal
- Set up wifi information with DHCP
- Open portal of the node at the IP provided to DHCP
- Setup Fixed IP as above, without insert again wifi detail

maakbaas, 2020-11-13:
Yes, but then you are trying to connect to an empty ssid with an empty password. So that explains it. I can see if the interface can be made more clear.

marcolino7, 2020-11-13:
@maakbaas 
no problem for that, now that i know how it work, it's fine for me.

Thanks