maakbaas/esp8266-iot-framework #57: SSID "Generic ESP8266 Firmware" not showing

emilnorde, 2020-11-21:
Hi!

I followed your instructions steps 1-10 with "timeSyncExample.cpp" as example code. Uploading code to esp8266 works great and without any issues. 

On step 11 though, no WiFi with SSID of Generic ESP8266 Firmware is detected. 

Serial monitor show:

> Fatal exception (28): 
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000

over and over again. 

What am I doing wrong?

Best regards.

maakbaas, 2020-11-21:
I don't know ;).

What is your hardware version, what is the PlatformIO version, and what is the version of the framework?

I can not reproduce this in any way. So more details would be needed to help your case.

maakbaas, 2020-11-21:
As a side note: the docs should be updated, the actual firmware name you should be looking for is 

`ESP8266 IoT Framework`. 

But it sounds like this is not the issue here.

emilnorde, 2020-11-21:
Haha, yes. :) First of all, thanks for quick response. 

I use the VS Code extension and installed the esp8266-iot-framework via PlatformIO Library Manager 
I added `lib_deps = ESP8266 IoT Framework` to the platformio.ini file as instructed. 

Also executed the command `pip install asn1crypto` after getting error message of that missing (followed the issue in #51 )

When I build and upload this is what the terminal shows:
```
> Executing task in folder esp8266_test3: C:\Users\emiln\.platformio\penv\Scripts\pio.exe run --target upload <

Processing esp01_1m (platform: espressif8266; board: esp01_1m; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 (2.6.2) > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - tool-mklittlefs 1.203.200522 (2.3)
 - tool-mkspiffs 1.200.0 (2.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266 IoT Framework> 1.6.1
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson> 6.17.2
|   |-- <ArduinoJson> 6.17.2
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <EEPROM> 1.0
|   |-- <LittleFS(esp8266)> 0.1.0
|-- <LittleFS(esp8266)> 0.1.0
Building in release mode
Retrieving maximum program size .pio\build\esp01_1m\firmware.elf
Checking size .pio\build\esp01_1m\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  45.2% (used 37004 bytes from 81920 bytes)
Flash: [========  ]  77.4% (used 589688 bytes from 761840 bytes)
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM8
Uploading .pio\build\esp01_1m\firmware.bin
esptool.py v2.8
Serial port COM8
Connecting........_____
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 48:3f:da:70:b3:64
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 593840 bytes to 440846...

Writing at 0x00000000... (3 %)
Writing at 0x00004000... (7 %)
Writing at 0x00008000... (11 %)
Writing at 0x0000c000... (14 %)
Writing at 0x00010000... (18 %)
Writing at 0x00014000... (22 %)
Writing at 0x00018000... (25 %)
Writing at 0x0001c000... (29 %)
Writing at 0x00020000... (33 %)
Writing at 0x00024000... (37 %)
Writing at 0x00028000... (40 %)
Writing at 0x0002c000... (44 %)
Writing at 0x00030000... (48 %)
Writing at 0x00034000... (51 %)
Writing at 0x00038000... (55 %)
Writing at 0x0003c000... (59 %)
Writing at 0x00040000... (62 %)
Writing at 0x00044000... (66 %)
Writing at 0x00048000... (70 %)
Writing at 0x0004c000... (74 %)
Writing at 0x00050000... (77 %)
Writing at 0x00054000... (81 %)
Writing at 0x00058000... (85 %)
Writing at 0x0005c000... (88 %)
Writing at 0x00060000... (92 %)
Writing at 0x00064000... (96 %)
Writing at 0x00068000... (100 %)
Wrote 593840 bytes (440846 compressed) at 0x00000000 in 38.8 seconds (effective 122.5 kbit/s)...
Hash of data verified.

Leaving...
Soft resetting...
================================================================== [SUCCESS] Took 46.45 seconds ==================================================================

Terminal will be reused by tasks, press any key to close it.
```

And then, no WiFi sent from esp8266 to connect to :(

maakbaas, 2020-11-21:
Thats interesting.

- What is the exact hardware/development board you are using?
- Can you delete your .pio folder and build/flash again, and paste that log?
- Do you have a second board to try by any chance? 
- Have you tried one of the other example C files to see if that changes anything?

emilnorde, 2020-11-21:
I went back to the basics as I never used platformIO as IDE for esp8266. Tried to upload simple Blink program through it. 

```
/*
 * Blink
 * Turns on an LED on for one second,
 * then off for one second, repeatedly.
 */

#include "Arduino.h"

void setup()
{
  Serial.begin(115200);
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);
  // wait for a second
  delay(1000);
  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);
   // wait for a second
  delay(1000);
}
```
It uploads but the esp8266 doesn't blink.

Tried with the same code but uploading through Arduino IDE, and it works. So perhaps some wrong setting in PlatformIO? Can´t figure it out.

Hardware: 
----
ESP8266-01 connected through an Arduino UNO via RX/TX pins. (Uploaded programs without issues through Arduino IDE as described. )

https://www.amazon.com/HiLetgo-Wireless-Transceiver-Development-Compatible/dp/B010N1ROQS/ref=sr_1_9?dchild=1&keywords=esp-01&qid=1605975828&sr=8-9

Terminal after removing .pio
-----
```
> Executing task in folder esp8266_test3: C:\Users\emiln\.platformio\penv\Scripts\pio.exe run --target upload <

Processing esp01_1m (platform: espressif8266; board: esp01_1m; framework: arduino)
---------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 (2.6.2) > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - tool-mklittlefs 1.203.200522 (2.3) 
 - tool-mkspiffs 1.200.0 (2.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing ESP8266 IoT Framework
Library Manager: ESP8266 IoT Framework @ 1.6.1 has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing ESP Async WebServer
Library Manager: ESP Async WebServer @ 1.2.3 has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing ESPAsyncTCP
Library Manager: ESPAsyncTCP @ 1.2.2 has been installed!
Library Manager: Installing AsyncTCP
[... 229 lines cut ...]
Writing at 0x00050000... (77 %)
Writing at 0x00054000... (81 %)
Writing at 0x00058000... (85 %)
Writing at 0x0005c000... (88 %)
Writing at 0x00060000... (92 %)
Writing at 0x00064000... (96 %)
Writing at 0x00068000... (100 %)
Wrote 593840 bytes (440846 compressed) at 0x00000000 in 38.2 seconds 
(effective 124.4 kbit/s)...
Hash of data verified.

Leaving...
Soft resetting...
=================== [SUCCESS] Took 73.93 seconds ===================

Terminal will be reused by tasks, press any key to close it.
```
Other board
--------
I'm afraid i only have this one board available, but I've been able to connect to websites with it before. So it should work. 

Other example programs
---------
I've tried with `helloWorld.cpp` , `timeSyncExample.cpp` and `Minimal main.c`. 
No difference, still no WiFi to connect to.

maakbaas, 2020-11-21:
I will close this issue for now, since it is more of a generic setup/PlatformIO issue.

I found this thread, which might help you:
https://community.platformio.org/t/problem-upload-code-to-esp-01s/8411

you can try setting `board_build.flash_mode = dout` in your `platformio.ini`? Also you can try setting a lower `upload_speed` in that file, but other than that I have no immediate suggestions.

Good luck, and let me know if you manage to find the issue!

closed 2020-11-21 by maakbaas

alanjohnson, 2021-03-16:
I wanted to also mention, that IF YOU had already used the Tzapu WifiManager, with the same board.  This will use the same credentials. So, you need to make sure to clear it (can use the included WiFiManager.forget()).  otherwise you will never see the SSID appear, as the board is actually already connected to your network from the previous settings.