maakbaas/esp8266-iot-framework #58: DashBoard

FSCSoft, 2020-11-26:
[
[image]
](url)

the dashboard does not work on mobile devices

maakbaas, 2020-11-26:
How does your JSON file look?

Does the same page function on the desktop for you? Can you tell me what happens here, and why it does not work? It seems surprising that there would be a difference between mobile and desktop. 

Do you see in the ESP8266 terminal that a new Websocket is connected when you go to this page from your mobile? Actually, the fact that the `live` label is shown indicates that there is a successful Websocket connection...

FSCSoft, 2020-11-27:
Yes on desktop if it works perfectly I could only make it work in chrome dev in android for my thinking it can be the bintojson converter and if the connection to the websocket and api rest is fine because I can set the options

 

Sentek Ver.

El 25 nov 2020, a la(s) 11:33 p. m., Maakbaas <notifications@github.com> escribió:

> How does your JSON file look?
>
> Does the same page function on the desktop for you? Can you tell me what happens here, and why it does not work? It seems surprising that there would be a difference [...]

maakbaas, 2020-11-27:
Without providing more details in response to my earlier questions I am inclined to close this issue, because then it will be difficult to debug...

FSCSoft, 2020-11-27:
If friend will be difficult debugging I agree

[dashboard.zip](https://github.com/maakbaas/esp8266-iot-framework/files/5609433/dashboard.zip)

FSCSoft, 2020-11-27:
don't think about using arduinoJSON

maakbaas, 2020-11-27:
and your C code?

FSCSoft, 2020-11-28:
here it is
[WifiLavadora.zip](https://github.com/maakbaas/esp8266-iot-framework/files/5610054/WifiLavadora.zip)

closed 2020-11-28 by FSCSoft

maakbaas, 2020-11-28:
Thanks for sharing. I tried your code on 2 different mobile phones (1 android and 1 iPhone) and in both cases it works as expected. Hopefully for you now the solution is acceptable to use Chrome dev for Android. If more people run into the same issue we can investigate further!

FSCSoft, 2020-11-29:
Hello friend, it's me again and I think what the solution would be instead of converting to pass a string in websoket and maybe in api dash set of json format with only the name and value of each object { name: dumyFloat, value: 2.1},{...} etc. example or maybe to reduce characters {dumy: {value: 2.2}, input {value: true}, etc}. to pass the data in real values ​​and not go through the conversion and it is already an ArduinoJSON dependency of some libs of the project. sorry for my bad English