maakbaas/esp8266-iot-framework #59: Build errors

clabnet, 2020-11-27:
Hi, 
this project seems interestings.

Cloned this repo, build on Windows 10PRO with VSCODE PlatformIO Core 5.0.3,  I have this error on building.

Any suggestion ? Thanks

```
c:/users/claudio/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
c:/users/claudio/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
c:/users/claudio/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\libFrameworkArduino.a(core_esp8266_main.cpp.o): in function `loop_wrapper()':
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x21): undefined reference to `setup'
c:/users/claudio/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x2d): undefined reference 
to `loop'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\nodemcuv2\firmware.elf] Error 1
========================================================================================== [FAILED] Took 41.15 seconds ==========================================================================================The terminal process "C:\Users\Claudio\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.
```

clabnet, 2020-11-27:
Sorry, my mistake. I missing the main.cpp. 
Please close, this is not an issue. :o)

closed 2020-11-27 by clabnet

maakbaas, 2020-11-27:
I was just typing that, glad you figured it out :).

This and other steps are described in the installation guide:
https://github.com/maakbaas/esp8266-iot-framework/blob/master/docs/installation-guide.md