maakbaas/esp8266-iot-framework #60: Error building HTML

clabnet, 2020-11-28:
Hi,
very very good your framework. 
I would minimal change HTML code, I know npm, React and webpack. 

This is my platformio.ini:

```
[platformio]
default_envs = nodemcuv2
[common]
platform = https://github.com/platformio/platform-espressif8266.git
[env]
framework = arduino
build_flags = 
	-w ; supress all warnings
monitor_speed = 115200
upload_speed = 921600
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps = ESP Async WebServer, ArduinoJson
extra_scripts = scripts/preBuild.py
build_flags = -DREBUILD_HTML  forces a rebuild of the html.h, this is needed if you make modifications to the GUI, or the configuration or dashboard JSON files
#-DREBUILD_CERTS forces a rebuild of the root certificate store
#-DREBUILD_CONFIG forces a rebuild of the configuration manager cpp files based on the JSON
#-DREBUILD_DASHBOARD forces a rebuild of the dashboard cpp files based on the JSON
#-DDOMAIN_LIST=google.com,maakbaas.com comma separated list of domain names to limit the certificates included
#-DCONFIG_PATH=configuration.json defines a custom location for your JSON file
#-DDASHBOARD_PATH=dashboard.json defines a custom location for your JSON file
```
I have this warning, but I don't know how to solve.

```
...
Compiling .pio\build\nodemcuv2\FrameworkArduino\umm_malloc\umm_poison.c.o
Archiving .pio\build\nodemcuv2\libFrameworkArduino.a
*** [.pio\build\nodemcuv2\firmware.elf] Implicit dependency `C:\users\claudio\.platformio\platforms\espressif8266\builder\forces' not found, needed by target `.pio\build\nodemcuv2\firmware.elf'.
``` 

Thank's for sharing your work.

Best regards,
Claudio

maakbaas, 2020-11-28:
I am glad you like it! You need to remove the comment in `platformio.ini` ;).

`build_flags = -DREBUILD_HTML  forces a rebuild of the html.h, this is needed if you make modifications to the GUI, or the configuration or dashboard JSON files`

needs to change in:

`build_flags = -DREBUILD_HTML`

maakbaas, 2020-11-28:
By the way, If you have any changes in mind for the web interface that would benefit others, feel free to propose this, or share it in a pull request!

closed 2020-11-28 by maakbaas

clabnet, 2020-11-28:
Thank you Maakbaas,

also the build procedure running as swiss watch, and it is not simply working fine all together with python, modern web technologies, and embedded IOT code.

Eisbaeeer, 2021-02-15:
> Thank you Maakbaas,
> 
> also the build procedure running as swiss watch, and it is not simply working fine all together with python, modern web technologies, and embedded IOT code.

Im trying to add a button for external link like <a href="www.google.de">Open google</a>
It seems quiet easy, but I don´t firm with React. Im able to compile the HTML page after changing the Dashboard.json things, but don´t know how to add buttons or Links. The perfect solution will be adding buttons via variables. This will be cool.