maakbaas/esp8266-iot-framework #61: Strange dummy float bug

clabnet, 2020-11-28:
Hi Maakbaas,

I have applied your default config for testing.
Specifying a field as dummyFloat 1-2, I can correctly insert values as: 1, 2 NOT 5. 
But I would insert 1.3. The result is 1.29999.... 
or 1.2 the result is 1.20000.....  
[image]

maakbaas, 2020-11-28:
This is not a bug, but is to be expected. A 32-bit floating point datatype can not store exact representations of all numbers, and therefore it will be stored as the closest binary representation.

See this calculator to play around with this effect:
https://www.binaryconvert.com/convert_float.html

One way to resolve this is to add a new option into the JSON that specifies the number of decimals that should be shown in the interface, and to which the numbers should be rounded. I think that could be a nice addition, so for that reason I will leave this issue open. If anyone wants to create a pull request for this functionality that would be more than welcome.

clabnet, 2020-11-28:
Leaning a lot of things from you.

FSCSoft, 2020-11-29:
Hello friend, it's me again and I think what the solution would be instead of converting to pass a string in websoket and maybe in api dash set of json format with only the name and value of each object { name: dumyFloat, value: 2.1},{...} etc. example or maybe to reduce characters {dumy: {value: 2.2}, input {value: true}, etc}. to pass the data in real values ​​and not go through the conversion and it is already an ArduinoJSON dependency of some libs of the project. sorry for my bad English

maakbaas, 2020-11-29:
Thanks for your input. I can agree that this makes sense from a front-end and web point of view, but it does not make sense from the embedded side of things. Parsing JSON on the ESP8266 is much more expensive than converting JSON to binary in the front-end. In the second case, which is how it is implemented now, the conversion to binary is done by your computer or smartphone, which has more than enough resources to do this simple conversion from JavaScript object to binary format.

If we would choose to send JSON data over the Websocket, the ESP8266 would need to parse this JSON data and store it in a C structure. This will cost more memory and computational power on the ESP8266, where it is much more limited. With the current implementation, all that the ESP8266 needs to do is a single memory copy action.

maakbaas, 2021-02-17:
Added this functionality in 13988ca :)

closed 2021-02-17 by maakbaas