maakbaas/esp8266-iot-framework #62: Update certificates store

clabnet, 2020-11-28:
Hi Maakbaas,

I don't think if this is a bug of your code, but updating and reducing the  certificates store, (to reduce size on RAM) I have this error:

```
Start building certificate store
Only for domains: google.com
Skipped: AC Camerfirma, S.A.:AC Camerfirma SA CIF A82743287:http://www.chambersign.org
Skipped: AC Camerfirma, S.A.:AC Camerfirma S.A.:
Skipped: AC Camerfirma, S.A.:AC Camerfirma SA CIF A82743287:http://www.chambersign.org
Skipped: AC Camerfirma, S.A.:AC Camerfirma S.A.:
Skipped: Actalis:Actalis S.p.A./03358520967:
Skipped: Amazon Trust Services:Amazon:
Skipped: Amazon Trust Services:Amazon:
Skipped: Amazon Trust Services:Amazon:
Skipped: Amazon Trust Services:Amazon:
Skipped: Amazon Trust Services:Starfield Technologies, Inc.:
Skipped: Asseco Data Systems S.A. (previously Unizeto Certum):Unizeto Sp. z o.o.:
Skipped: Asseco Data Systems S.A. (previously Unizeto Certum):Unizeto Technologies S.A.:Certum Certification Authority
Skipped: Asseco Data Systems S.A. (previously Unizeto Certum):Unizeto Technologies S.A.:Certum Certification Authority
Skipped: Atos:Atos:
Skipped: Autoridad de Certificacion Firmaprofesional::
Skipped: Buypass:Buypass AS-983163327:
Skipped: Buypass:Buypass AS-983163327:
Skipped: certSIGN:certSIGN:certSIGN ROOT CA
Skipped: certSIGN:CERTSIGN SA:certSIGN ROOT CA G2
Skipped: China Financial Certification Authority (CFCA):China Financial Certification Authority:
Skipped: Chunghwa Telecom:Chunghwa Telecom Co., Ltd.:ePKI Root Certification Authority
Skipped: Consorci Administraci Oberta de Catalunya (Consorci AOC, CATCert):Agencia Catalana de Certificacio (NIF Q-0801176-I):Serveis Publics de Certificacio, Vegeu https://www.catcert.net/verarrel (c)03, Jerarquia Entitats de Certificacio Catalanes
Skipped: Cybertrust Japan / JCSI:Japan Certification Services, Inc.:
Skipped: D-TRUST:D-Trust GmbH:
Skipped: D-TRUST:D-Trust GmbH:
Skipped: D-TRUST:D-Trust GmbH:
Skipped: Deutsche Telekom Security GmbH:T-Systems Enterprise Services GmbH:T-Systems Trust Center
Skipped: Deutsche Telekom Security GmbH:T-Systems Enterprise Services GmbH:T-Systems Trust Center
Skipped: Dhimyotis / Certigna:Dhimyotis:
Skipped: Dhimyotis / Certigna:Dhimyotis:0002 48146308100036
Skipped: DigiCert:Baltimore:CyberTrust
Skipped: DigiCert:Cybertrust, Inc:
Skipped: DigiCert:DigiCert Inc:www.digicert.com
Skipped: DigiCert:DigiCert Inc:www.digicert.com
Skipped: DigiCert:DigiCert Inc:www.digicert.com
Skipped: DigiCert:DigiCert Inc:www.digicert.com
Skipped: DigiCert:DigiCert Inc:www.digicert.com
Skipped: DigiCert:DigiCert Inc:www.digicert.com
Skipped: DigiCert:DigiCert Inc:www.digicert.com
Skipped: DigiCert:DigiCert Inc:www.digicert.com
Skipped: DigiCert:GeoTrust Inc.:
Skipped: DigiCert:GeoTrust Inc.:
Skipped: DigiCert:GeoTrust Inc.:(c) 2007 GeoTrust Inc. - For authorized use only
Skipped: DigiCert:GeoTrust Inc.:(c) 2008 GeoTrust Inc. - For authorized use only
Skipped: DigiCert:GeoTrust Inc.:
Skipped: DigiCert:GeoTrust Inc.:
Skipped: DigiCert:Symantec Corporation:Symantec Trust Network
Skipped: DigiCert:Symantec Corporation:Symantec Trust Network
Skipped: DigiCert:thawte, Inc.:Certification Services Division, (c) 2006 thawte, Inc. - For authorized use only
Skipped: DigiCert:thawte, Inc.:(c) 2007 thawte, Inc. - For authorized use only
Skipped: DigiCert:thawte, Inc.:Certification Services Division, (c) 2008 thawte, Inc. - For authorized use only
Skipped: DigiCert:VeriSign, Inc.:VeriSign Trust Network, (c) 1999 VeriSign, Inc. - For authorized use only
Skipped: DigiCert:VeriSign, Inc.:VeriSign Trust Network, (c) 1999 VeriSign, Inc. - For authorized use only
Skipped: DigiCert:VeriSign, Inc.:VeriSign Trust Network, (c) 2007 VeriSign, Inc. - For authorized use only
Skipped: DigiCert:VeriSign, Inc.:VeriSign Trust Network, (c) 2006 VeriSign, Inc. - For authorized use only
Skipped: DigiCert:VeriSign, Inc.:VeriSign Trust Network, (c) 2008 VeriSign, Inc. - For authorized use only
Skipped: Disig, a.s.:Disig a.s.:
Skipped: E-Tugra:E-Tura EBG Biliim Teknolojileri ve Hizmetleri A..:E-Tugra Sertifikasyon Merkezi
Skipped: eMudhra Technologies Limited:eMudhra Inc:emSign PKI
Skipped: eMudhra Technologies Limited:eMudhra Technologies Limited:emSign PKI
Skipped: eMudhra Technologies Limited:eMudhra Inc:emSign PKI
Skipped: eMudhra Technologies Limited:eMudhra Technologies Limited:emSign PKI
Skipped: Entrust:AffirmTrust:
Skipped: Entrust:AffirmTrust:
Skipped: Entrust:AffirmTrust:
Skipped: Entrust:AffirmTrust:
Skipped: Entrust:Entrust, Inc.:www.entrust.net/CPS is incorporated by reference, (c) 2006 Entrust, Inc.
Skipped: Entrust:Entrust, Inc.:See www.entrust.net/legal-terms, (c) 2012 Entrust, Inc. - for authorized use only
Skipped: Entrust:Entrust, Inc.:See www.entrust.net/legal-terms, (c) 2009 Entrust, Inc. - for authorized use only
Skipped: Entrust:Entrust, Inc.:See www.entrust.net/legal-terms, (c) 2015 Entrust, Inc. - for authorized use only
Skipped: Entrust:Entrust.net:www.entrust.net/CPS_2048 incorp. by ref. (limits liab.), (c) 1999 Entrust.net Limited
Skipped: Global Digital Cybersecurity Authority Co., Ltd. (Formerly Guang Dong Certificate Authority (GDCA)):GUANG DONG CERTIFICATE AUTHORITY CO.,LTD.:
Skipped: GlobalSign nv-sa:GlobalSign:GlobalSign Root CA - R3
Skipped: GlobalSign nv-sa:GlobalSign:GlobalSign Root CA - R6
Skipped: GlobalSign nv-sa:GlobalSign:GlobalSign ECC Root CA - R5
Skipped: GlobalSign nv-sa:GlobalSign nv-sa:Root CA
Skipped: GoDaddy:The Go Daddy Group, Inc.:Go Daddy Class 2 Certification Authority
Skipped: GoDaddy:GoDaddy.com, Inc.:
Skipped: GoDaddy:Starfield Technologies, Inc.:Starfield Class 2 Certification Authority
Skipped: GoDaddy:Starfield Technologies, Inc.:
**Added: Google Trust Services LLC (GTS):GlobalSign:GlobalSign Root CA - R2
PermissionError: [WinError 5] Access is denied:
  File "C:\users\claudio\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 178:
    env.SConscript(item, exports="env")
  File "C:\Users\Claudio\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Claudio\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\IOT PlatformIO\esp8266-iot-framework\scripts\preBuild.py", line 49:
    preBuildCertificatesFun(domains)
  File "D:\IOT PlatformIO\esp8266-iot-framework\scripts\preBuildCertificates.py", line 122:
    ssl = Popen([openssl,'x509','-inform','PEM','-outform','DER','-out', certName + '.der'], shell = False, stdin = PIPE)**
  File "C:\Users\Claudio\.platformio\python3\lib\subprocess.py", line 800:
    restore_signals, start_new_session)
  File "C:\Users\Claudio\.platformio\python3\lib\subprocess.py", line 1207:
    startupinfo)
```

maakbaas, 2020-11-28:
Did you follow these steps (https://github.com/maakbaas/esp8266-iot-framework/blob/master/docs/fetch.md#code-generation) to set a correct OpenSSL location in the Python script? Can you check that the current Windows user has the correct permissions to execute this application?

clabnet, 2020-11-28:
Yes I have double checked the settings 
```
#path to openssl
openssl = "C:\\mypath\\of\\openssl"
```
and I tried with two different versions of openssl.exe executables, in different paths. 

The problem is the same, but I have note that doesn't always stop at the same point.

```
Skipped: GoDaddy:The Go Daddy Group, Inc.:Go Daddy Class 2 Certification Authority
Skipped: GoDaddy:GoDaddy.com, Inc.:
PermissionError: [WinError 5] Access is denied:
Skipped: GoDaddy:Starfield Technologies, Inc.:Starfield Class 2 Certification Authority
  File "C:\users\claudio\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 178:
Skipped: GoDaddy:Starfield Technologies, Inc.:
    env.SConscript(item, exports="env")
Added: Google Trust Services LLC (GTS):GlobalSign:GlobalSign Root CA - R2
  File "C:\Users\Claudio\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Claudio\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\IOT PlatformIO\esp8266-iot-framework\scripts\preBuild.py", line 49:
    preBuildCertificatesFun(domains)
  File "D:\IOT PlatformIO\esp8266-iot-framework\scripts\preBuildCertificates.py", line 122:
    ssl = Popen([openssl,'x509','-inform','PEM','-outform','DER','-out', certName + '.der'], shell = False, stdin = PIPE)
  File "C:\Users\Claudio\.platformio\python3\lib\subprocess.py", line 800:
    restore_signals, start_new_session)
  File "C:\Users\Claudio\.platformio\python3\lib\subprocess.py", line 1207:
    startupinfo)
```
A question :
try the code  to connect the internet during this process ? It can be the lowest of my internet band ?

Thanks.

clabnet, 2020-11-28:
I have tried removing the -DOMAINLIST flag. 
The procedure stop at first adding cert. 
```
Start building certificate store
For all domains
Added: AC Camerfirma, S.A.:AC Camerfirma SA CIF A82743287:http://www.chambersign.org
PermissionError: [WinError 5] Access is denied:
  File "C:\users\claudio\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 178:
    env.SConscript(item, exports="env")
  File "C:\Users\Claudio\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Claudio\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\IOT PlatformIO\esp8266-iot-framework\scripts\preBuild.py", line 49:
    preBuildCertificatesFun(domains)
  File "D:\IOT PlatformIO\esp8266-iot-framework\scripts\preBuildCertificates.py", line 122:
    ssl = Popen([openssl,'x509','-inform','PEM','-outform','DER','-out', certName + '.der'], shell = False, stdin = PIPE)
  File "C:\Users\Claudio\.platformio\python3\lib\subprocess.py", line 800:
    restore_signals, start_new_session)
  File "C:\Users\Claudio\.platformio\python3\lib\subprocess.py", line 1207:
    startupinfo)
```

maakbaas, 2020-11-28:
I can reproduce this on my side, if I take away the execute rights from the openssl.exe. 

Since this is a Python error that can only mean this one thing I am quite sure that this is the issue in your case as well. Please check that administrators have execute access for this binary. 

```
Added: Google Trust Services LLC (GTS):GlobalSign:GlobalSign Root CA - R2
PermissionError: [WinError 5] Access is denied:
  File "C:\users\johannes\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 178:
    env.SConscript(item, exports="env")
  File "C:\Users\Johannes\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Johannes\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\Johannes\Maker\Framework\scripts\preBuild.py", line 49:
    preBuildCertificatesFun(domains)
  File "D:\Johannes\Maker\Framework\scripts\preBuildCertificates.py", line 122:
    ssl = Popen([openssl,'x509','-inform','PEM','-outform','DER','-out', certName + '.der'], shell = False, stdin = PIPE)
  File "C:\Users\Johannes\.platformio\python3\lib\subprocess.py", line 800:
    restore_signals, start_new_session)
  File "C:\Users\Johannes\.platformio\python3\lib\subprocess.py", line 1207:
    startupinfo)
```

clabnet, 2020-11-28:
It is not a bug. I want to explain what happened.
Your code into prebuildCertificates.py say:
```
#path to openssl
    openssl = "C:/msys32/usr/bin/openssl" 
```
I changed with my path only without the executable name. 
I swapped the final openssl as part of the path and not as the executable name, 
also because in the comment it says "path" and not "fullname", and it is missing the .exe extension. 
But it work as the same. 

Sorry for wasting your time, but your code is very interesting and very well written. 
Thanks for your fantastic support.

maakbaas, 2020-11-28:
I'm glad you figured it out

closed 2020-11-28 by maakbaas