maakbaas/esp8266-iot-framework #63: preBuildCertificates.py error

viktak, 2020-12-01:
Hi,

I am now starting to look at your solution to TLS (which I find interesting), but I ran into an issue right at the beginning.

When I run the script I get the following errors:

```
2037.09.30 -> ca_000.der
unable to load certificate
2038.07.31 -> ca_000.der
9340:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
2037.09.30 -> ca_000.der
2636:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
2038.07.31 -> ca_000.der
16864:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
.
.
.
```

Also, the generated `certificates.h` file is "empty", as expected:
```
#ifndef CERT_H
#define CERT_H

#include <Arduino.h>

//2037.09.30 AC Camerfirma, S.A.:AC Camerfirma SA CIF A82743287:http://www.chambersign.org
//2038.07.31 AC Camerfirma, S.A.:AC Camerfirma S.A.:
//2037.09.30 AC Camerfirma, S.A.:AC Camerfirma SA CIF A82743287:http://www.chambersign.org
//2038.07.31 AC Camerfirma, S.A.:AC Camerfirma S.A.:
//2030.09.22 Actalis:Actalis S.p.A./03358520967:
//2038.01.17 Amazon Trust Services:Amazon:
//2040.05.26 Amazon Trust Services:Amazon:
//2040.05.26 Amazon Trust Services:Amazon:
//2040.05.26 Amazon Trust Services:Amazon:
//2037.12.31 Amazon Trust Services:Starfield Technologies, Inc.:
//2027.06.11 Asseco Data Systems S.A. (previously Unizeto Certum):Unizeto Sp. z o.o.:
//2029.12.31 Asseco Data Systems S.A. (previously Unizeto Certum):Unizeto Technologies S.A.:Certum Certification Authority
//2046.10.06 Asseco Data Systems S.A. (previously Unizeto Certum):Unizeto Technologies S.A.:Certum Certification Authority
//2030.12.31 Atos:Atos:
//2030.12.31 Autoridad de Certificacion Firmaprofesional::
//2040.10.26 Buypass:Buypass AS-983163327:
//2040.10.26 Buypass:Buypass AS-983163327:
//2031.07.04 certSIGN:certSIGN:certSIGN ROOT CA
//2042.02.06 certSIGN:CERTSIGN SA:certSIGN ROOT CA G2
//2029.12.31 China Financial Certification Authority (CFCA):China Financial Certification Authority:
//2034.12.20 Chunghwa Telecom:Chunghwa Telecom Co., Ltd.:ePKI Root Certification Authority
//2031.01.07 Consorci Administració Oberta de Catalunya (Consorci AOC, CATCert):Agencia Catalana de Certificacio (NIF Q-0801176-I):Serveis Publics de Certificacio, Vegeu https://www.catcert.net/verarrel (c)03, Jerarquia Entitats de Certificacio Catalanes
//2029.04.08 Cybertrust Japan / JCSI:Japan Certification Services, Inc.:
//2028.09.20 D-TRUST:D-Trust GmbH:
//2029.11.05 D-TRUST:D-Trust GmbH:
//2029.11.05 D-TRUST:D-Trust GmbH:
//2033.10.01 Deutsche Telekom Security GmbH:T-Systems Enterprise Services GmbH:T-Systems Trust Center
//2033.10.01 Deutsche Telekom Security GmbH:T-Systems Enterprise Services GmbH:T-Systems Trust Center
//2027.06.29 Dhimyotis / Certigna:Dhimyotis:
//2033.10.01 Dhimyotis / Certigna:Dhimyotis:0002 48146308100036
//2025.05.12 DigiCert:Baltimore:CyberTrust
//2021.12.15 DigiCert:Cybertrust, Inc:
//2031.11.10 DigiCert:DigiCert Inc:www.digicert.com
//2038.01.15 DigiCert:DigiCert Inc:www.digicert.com
//2038.01.15 DigiCert:DigiCert Inc:www.digicert.com
//2031.11.10 DigiCert:DigiCert Inc:www.digicert.com
//2038.01.15 DigiCert:DigiCert Inc:www.digicert.com
//2038.01.15 DigiCert:DigiCert Inc:www.digicert.com
//2031.11.10 DigiCert:DigiCert Inc:www.digicert.com
//2038.01.15 DigiCert:DigiCert Inc:www.digicert.com
[... 101 lines cut ...]
//2021.04.06 Telia Company (formerly TeliaSonera):Sonera:
//2032.10.18 Telia Company (formerly TeliaSonera):TeliaSonera:
//2029.12.31 TrustCor Systems:TrustCor Systems S. de R.L.:TrustCor Certificate Authority
//2029.12.31 TrustCor Systems:TrustCor Systems S. de R.L.:TrustCor Certificate Authority
//2034.12.31 TrustCor Systems:TrustCor Systems S. de R.L.:TrustCor Certificate Authority
//2024.01.21 Trustis:Trustis Limited:Trustis FPS Root CA
//2029.12.31 Web.com:Network Solutions L.L.C.:
//global variables for certificates using 0 bytes
const uint16_t numberOfCertificates PROGMEM = 0;

const uint16_t certSizes[] PROGMEM = {};

const uint8_t* const certificates[] PROGMEM = {};

const uint8_t* const indices[] PROGMEM = {};

#endif

```

My environment is Visual Studio Code with PlatformIO on a Windows Pro 64 box with OpenSSL extracted from [this file ](https://mirror.firedaemon.com/OpenSSL/openssl-1.1.1i-dev.zip)on a local folder.

Most likely it is an installation issue, but, since I haven't worked with openssl before, I don't know how to go on...

Thanks for any pointers!

maakbaas, 2020-12-01:
can you post the full console output when you run a PlatformIO build with `build_flags = -DREBUILD_CERTS` added to `platformio.ini`?

viktak, 2020-12-01:
Here we go:
```pio.exe run <

Warning! Ignore unknown configuration option `major_build_number` in section [env:esp12e]
Processing esp12e (platform: espressif8266; framework: arduino; board: esp12e)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
2037.09.30 -> ca_000.der
unable to load certificate
12948:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2038.07.31 -> ca_000.der
unable to load certificate
2576:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2037.09.30 -> ca_000.der
unable to load certificate
16720:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2038.07.31 -> ca_000.der
2030.09.22 -> ca_000.der
unable to load certificate
17604:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
14224:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2038.01.17 -> ca_000.der
unable to load certificate
2040.05.26 -> ca_000.der
18452:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
5144:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2040.05.26 -> ca_000.der
unable to load certificate
18584:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2040.05.26 -> ca_000.der
2037.12.31 -> ca_000.der
unable to load certificate
11576:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
17536:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2027.06.11 -> ca_000.der
unable to load certificate
4944:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2029.12.31 -> ca_000.der
unable to load certificate
7904:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2046.10.06 -> ca_000.der
2030.12.31 -> ca_000.der
unable to load certificate
18120:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
2030.12.31 -> ca_000.der
16732:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
16104:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2040.10.26 -> ca_000.der
unable to load certificate
10544:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2040.10.26 -> ca_000.der
2031.07.04 -> ca_000.der
unable to load certificate
18448:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
17284:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2042.02.06 -> ca_000.der
2029.12.31 -> ca_000.der
unable to load certificate
19332:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
16584:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2034.12.20 -> ca_000.der
unable to load certificate
2031.01.07 -> ca_000.der
14604:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
[... 403 lines cut ...]
|   |-- <SPI> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <EEPROM> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <LittleFS(esp8266)> 0.1.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\esp12e\src\main.cpp.o
Linking .pio\build\esp12e\firmware.elf
Retrieving maximum program size .pio\build\esp12e\firmware.elf
Checking size .pio\build\esp12e\firmware.elf
Building .pio\build\esp12e\firmware.bin
Creating BIN file ".pio\build\esp12e\firmware.bin" using "C:\Users\Viktor\.platformio\packages\framework-arduinoespressif8266\bootloaders\eboot\eboot.elf" and ".pio\build\esp12e\firmware.elf"
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  40.5% (used 33188 bytes from 81920 bytes)
Flash: [====      ]  37.7% (used 393892 bytes from 1044464 bytes)
============================================================================================ [SUCCESS] Took 9.50 seconds ============================================================================================
```

maakbaas, 2020-12-01:
It seems like you are running an old version of the framework, since the console output has changed a while ago. Can you try again with the latest version of the framework (1.6.1)?

If you use the framework as a PlatformIO library, the quickest way is to delete your `.pio` folder.

viktak, 2020-12-02:
@maakbaas Old version of what? I usually try to keep up with the latest of everything.

Anyway, I deleted the `.pio` folder, as instructed, but the result doesn't seem changed:

```
Warning! Ignore unknown configuration option `major_build_number` in section [env:esp12e]
Processing esp12e (platform: espressif8266; framework: arduino; board: esp12e)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
2037.09.30 -> ca_000.der
unable to load certificate
15752:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2038.07.31 -> ca_000.der
unable to load certificate
2037.09.30 -> ca_000.der
16056:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
14664:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2038.07.31 -> ca_000.der
unable to load certificate
2030.09.22 -> ca_000.der
16544:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
8488:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2038.01.17 -> ca_000.der
unable to load certificate
7760:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2040.05.26 -> ca_000.der
unable to load certificate
2040.05.26 -> ca_000.der
13036:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
16244:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2040.05.26 -> ca_000.der
unable to load certificate
16400:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2037.12.31 -> ca_000.der
unable to load certificate
2027.06.11 -> ca_000.der
16636:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
12732:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2029.12.31 -> ca_000.der
unable to load certificate
2046.10.06 -> ca_000.der
1416:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
6548:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2030.12.31 -> ca_000.der
2030.12.31 -> ca_000.der
unable to load certificate
14376:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
13872:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2040.10.26 -> ca_000.der
unable to load certificate
2040.10.26 -> ca_000.der
16116:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
16172:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2031.07.04 -> ca_000.der
2042.02.06 -> ca_000.der
unable to load certificate
5272:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
15992:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
2029.12.31 -> ca_000.der
unable to load certificate
2034.12.20 -> ca_000.der
15404:error:0909006C:PEM routines:get_name:no start line:..\crypto\pem\pem_lib.c:745:Expecting: TRUSTED CERTIFICATE
unable to load certificate
[... 565 lines cut ...]
Compiling .pio\build\esp12e\FrameworkArduino\sntp-lwip2.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\spiffs\spiffs_cache.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\spiffs\spiffs_check.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\spiffs\spiffs_gc.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\spiffs\spiffs_hydrogen.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\spiffs\spiffs_nucleus.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\spiffs_api.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\sqrt32.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\time.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\uart.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\umm_malloc\umm_info.c.o
Compiling .pio\build\esp12e\FrameworkArduino\umm_malloc\umm_integrity.c.o
Compiling .pio\build\esp12e\FrameworkArduino\umm_malloc\umm_local.c.o
Compiling .pio\build\esp12e\FrameworkArduino\umm_malloc\umm_malloc.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\umm_malloc\umm_poison.c.o
Archiving .pio\build\esp12e\libFrameworkArduino.a
Linking .pio\build\esp12e\firmware.elf
Retrieving maximum program size .pio\build\esp12e\firmware.elf
Checking size .pio\build\esp12e\firmware.elf
Building .pio\build\esp12e\firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  40.4% (used 33132 bytes from 81920 bytes)
Flash: [====      ]  37.7% (used 393540 bytes from 1044464 bytes)
Creating BIN file ".pio\build\esp12e\firmware.bin" using "C:\Users\Viktor\.platformio\packages\framework-arduinoespressif8266\bootloaders\eboot\eboot.elf" and ".pio\build\esp12e\firmware.elf"
`

maakbaas, 2020-12-02:
Old version of this project. 

How do you use this project? Did you clone this repo or are you using it as a PlatformIO library? what is the version number in library.json?

viktak, 2020-12-02:
I simply made a local copy of this single file and I "call" it from platformio.ini.
Does it have dependencies? Do I need to have the whole project locally?

maakbaas, 2020-12-02:
You do not need to use the whole project if you don't want to of course :). But I can not give support beyond the standard use case of using the entire framework.

As indicated, you are most definitely not using the latest version of the script which can be found here:
https://github.com/maakbaas/esp8266-iot-framework/blob/master/scripts/preBuildCertificates.py

The syntax `2042.05.30 -> ca_000.der` was removed on august 30 in commit 1a46b2fc9654c00f685fec7c185209dd68ca3761.

Read the Python script carefully, or use debugging. I assume it might have something to do with hardcoded expected relative paths, Also, keep in mind that you are nothing with the generated `html.h` header without the fetch class and its dependencies.

closed 2020-12-02 by maakbaas

viktak, 2020-12-02:
Cool, thanks!