maakbaas/esp8266-iot-framework #64: How to intercept the save config ?

clabnet, 2020-12-11:
Hi maakbaas, 

this is not a bug, it is a request info. 

On the normal running of your code, I would change a value on config. 
I will go on the config settings page and I change one value, then press Save. The new value is saved into EEPROM.
It's good. 

But how to get on the "loop" the config change event ? Because I would restart the esp to running with the new values.

Thanks for this code.

clabnet, 2020-12-11:
I have found this working solution. It's a manual choice but it is work.
Created a new page named Restart, with a single button Restart, added it to menu, and call the API restart from button click. 
Good, it is simple !

closed 2020-12-11 by clabnet

cychrn, 2021-03-16:
Hello, is it possible to share the way you create a new page, put a restart button and call the API restart from the button click?

i am in need of this feature badly. thank you very much!