maakbaas/esp8266-iot-framework #65: Cannot build the project

Shalke04, 2020-12-12:
Hi, Im trying to build and upload one of examples of this project, but without success. I made gif with the steps I did, maybe I did something wrong. Thanks for any advice.
[image]

maakbaas, 2020-12-13:
You can probably resolve this by opening the PlatformIO terminal using the small button [>] in the bottom of VS code, and then executing the command 

`pip install asn1crypto`

Let me know if this works. In one of the coming weeks I will update the scripts to do this automatically since it has caused more confusion before. Thanks for the clear visualisation!

Shalke04, 2020-12-13:
I already did what you suggested and here is the result. Some errors got fixed but some still remained.
[image]

maakbaas, 2020-12-13:
Can you paste as text the full output of the build command (not what is on the problems tab, but the terminal output).

Shalke04, 2020-12-13:
> Executing task in folder Test_Project_Framework: C:\Users\shalk\.platformio\penv\Scripts\pio.exe run <

Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2.6.2) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266 IoT Framework> 1.6.1  
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.2    
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson> 6.17.2
|   |-- <ArduinoJson> 6.17.2
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <EEPROM> 1.0
|   |-- <LittleFS(esp8266)> 0.1.0
|-- <LittleFS(esp8266)> 0.1.0
Building in release mode
Compiling .pio\build\esp12e\src\main.cpp.o
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\ESP8266WiFiAP.cpp.o
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
src\main.cpp: In function 'void loop()':
src\main.cpp:43:28: error: 'struct configData' has no member named 'dummyInt'
         configManager.data.dummyInt++;
                            ^
src\main.cpp:55:32: error: 'struct configData' has no member named 'dummyInt'
             configManager.data.dummyInt + 1,
                                ^
src\main.cpp:59:9: error: too many initializers for 'configData'
         };
         ^
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\ESP8266WiFiGratuitous.cpp.o
*** [.pio\build\esp12e\src\main.cpp.o] Error 1
==================================================================== [FAILED] Took 9.72 seconds ====================================================================
The terminal process "C:\Users\shalk\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

> Executing task in folder Test_Project_Framework: C:\Users\shalk\.platformio\penv\Scripts\pio.exe run <

Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2.6.2) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266 IoT Framework> 1.6.1
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson> 6.17.2
|   |-- <ArduinoJson> 6.17.2
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <EEPROM> 1.0
|   |-- <LittleFS(esp8266)> 0.1.0
|-- <LittleFS(esp8266)> 0.1.0
Building in release mode
Compiling .pio\build\esp12e\src\main.cpp.o
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\ESP8266WiFiMulti.cpp.o
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\ESP8266WiFiSTA-WPS.cpp.o
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\ESP8266WiFiSTA.cpp.o
src\main.cpp: In function 'void loop()':
src\main.cpp:43:28: error: 'struct configData' has no member named 'dummyInt'
         configManager.data.dummyInt++;
                            ^
src\main.cpp:55:32: error: 'struct configData' has no member named 'dummyInt'
             configManager.data.dummyInt + 1,
                                ^
src\main.cpp:59:9: error: too many initializers for 'configData'
         };
         ^
*** [.pio\build\esp12e\src\main.cpp.o] Error 1
==================================================================== [FAILED] Took 6.77 seconds ====================================================================
The terminal process "C:\Users\shalk\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

maakbaas, 2020-12-13:
Ah, this point should indeed be made more clear in the documentation. Thanks for reporting.

The thing is, since you are using the configManager example, you also need to use the corresponding JSON file for the configManager. So the best you can do is to copy the example [JSON file](https://github.com/maakbaas/esp8266-iot-framework/blob/master/examples/configManager/configuration.json) into your project folder, and then add `build_flags = -DCONFIG_PATH=configuration.json` into your `platformio.ini` as described [here](https://github.com/maakbaas/esp8266-iot-framework/blob/master/docs/installation-guide.md#build-options)

Shalke04, 2020-12-13:
[image]
Copied json file into project folder and added build_flags into platformio.ini. Still without success

> Executing task in folder Test_Project_Framework: C:\Users\shalk\.platformio\penv\Scripts\pio.exe run <

Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2.6.2) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266 IoT Framework> 1.6.1  
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.2    
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson> 6.17.2
|   |-- <ArduinoJson> 6.17.2
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <EEPROM> 1.0
|   |-- <LittleFS(esp8266)> 0.1.0
|-- <LittleFS(esp8266)> 0.1.0
Building in release mode
Compiling .pio\build\esp12e\src\main.cpp.o
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\ESP8266WiFi.cpp.o
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\ESP8266WiFiAP.cpp.o
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\ESP8266WiFiGratuitous.cpp.o
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\ESP8266WiFiMulti.cpp.o
src\main.cpp: In function 'void loop()':
src\main.cpp:43:28: error: 'struct configData' has no member named 'dummyInt'
         configManager.data.dummyInt++;
                            ^
src\main.cpp:55:32: error: 'struct configData' has no member named 'dummyInt'
             configManager.data.dummyInt + 1,
                                ^
src\main.cpp:59:9: error: too many initializers for 'configData'
         };
         ^
*** [.pio\build\esp12e\src\main.cpp.o] Error 1
==================================================================== [FAILED] Took 7.20 seconds ====================================================================
The terminal process "C:\Users\shalk\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

> Executing task in folder Test_Project_Framework: C:\Users\shalk\.platformio\penv\Scripts\pio.exe run <

Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2.6.2) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266 IoT Framework> 1.6.1
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson> 6.17.2
|   |-- <ArduinoJson> 6.17.2
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <EEPROM> 1.0
|   |-- <LittleFS(esp8266)> 0.1.0
|-- <LittleFS(esp8266)> 0.1.0
Building in release mode
Compiling .pio\build\esp12e\src\main.cpp.o
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\ESP8266WiFiSTA-WPS.cpp.o
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\ESP8266WiFiSTA.cpp.o
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\ESP8266WiFiScan.cpp.o
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\WiFiClientSecureAxTLS.cpp.o
src\main.cpp: In function 'void loop()':
src\main.cpp:43:28: error: 'struct configData' has no member named 'dummyInt'
         configManager.data.dummyInt++;
                            ^
src\main.cpp:55:32: error: 'struct configData' has no member named 'dummyInt'
             configManager.data.dummyInt + 1,
                                ^
src\main.cpp:59:9: error: too many initializers for 'configData'
         };
         ^
Compiling .pio\build\esp12e\libef6\ESP8266WiFi\WiFiClientSecureBearSSL.cpp.o
*** [.pio\build\esp12e\src\main.cpp.o] Error 1
==================================================================== [FAILED] Took 8.43 seconds ====================================================================
The terminal process "C:\Users\shalk\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Te

maakbaas, 2020-12-14:
Finally you also need to add `-DREBUILD_HTML` to the line with the build flags, otherwise the JSON file will not be parsed upon new build. Once you have set this up correctly you can make changes to the JSON file any time you want and these will be reflected in new builds. 

Before you can do that you need to install the prerequisites by following step 1 and step 2 from [this section](https://github.com/maakbaas/esp8266-iot-framework/blob/master/docs/getting-started.md#editing-the-web-interface) in the documentation.

In future versions I will make those steps more automated.

Shalke04, 2020-12-14:
Great, thanks. Now it finally compiled without errors. So I can learn how to modify the whole interface and start using it. It looks awesome

maakbaas, 2020-12-14:
Thanks, happy you got it working, and hopefully using the framework will be more enjoyable than the initial struggles :).

closed 2020-12-14 by maakbaas

Shalke04, 2020-12-14:
> ** Mon, 14 Dec 2020 00:50:17 -0800 - "maakbaas/esp8266-iot-framework" <reply@reply.github.com>, "maakbaas/esp8266-iot-framework" <esp8266-iot-framework@noreply.github.com> **
>
> Closed #65.

Hi, I added lines of code from Dashboard example into my existing project and got these erros. Any idea?

> Executing task in folder Test_Project_Framework: C:\Users\shalk\.platformio\penv\Scripts\pio.exe run &lt;

Error: Invalid &#39;C:\Users\shalk\Documents\PlatformIO\Projects\Test_Project_Framework\platformio.ini&#39; (project configuration file): &#39;While reading from &#39;C:\\Users\\shalk\\Documents\\PlatformIO\\Projects\\Test_Project_Framework\\platformio.ini&#39; [line 19]: option &#39;build_flags&#39; in section &#39;env:esp12e&#39; already exists&#39;
The terminal process &quot;C:\Users\shalk\.platformio\penv\Scripts\pio.exe &#39;run&#39;&quot; terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

> Executing task in folder Test_Project_Framework: C:\Users\shalk\.platformio\penv\Scripts\pio.exe run &lt;

Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2.6.2) &gt; Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -&gt; http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft

> esp@1.0.0 build C:\Users\shalk\Documents\PlatformIO\Projects\Test_Project_Framework\.pio\libdeps\esp12e\ESP8266 IoT Framework
> webpack --mode=production

Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
Hash: 1e2315a71af60e4f0f7e
Version: webpack 4.42.1
Time: 20297ms
Built at: 14. 12. 2020 20:53:11
          Asset      Size  Chunks                    Chunk Names
   ./index.html   451 KiB          [emitted]  [big]
./index.html.gz   118 KiB          [emitted]
      bundle.js   447 KiB       0  [emitted]  [big]  main
   bundle.js.gz   117 KiB          [emitted]
       main.css  3.52 KiB    0, 0  [emitted]         main, main
    main.css.gz  1.01 KiB          [emitted]
Entrypoint main [big] = main.css main.css bundle.js
 [1] ./node_modules/preact/compat/dist/compat.module.js + 2 modules 19.2 KiB {0} [built]
     |    3 modules
 [3] ./gui/js/configuration.json 969 bytes {0} [built]
 [8] ./gui/js/dashboard.json 10 bytes {0} [built]
[11] (webpack)/buildin/global.js 472 bytes {0} [built]
[28] ./node_modules/@emotion/is-prop-valid/dist/is-prop-valid.browser.esm.js + 1 modules 4.41 KiB {0} [built]
     |    2 modules
[70] ./gui/js/index.js + 343 modules 897 KiB {0} [built]
     | ./gui/js/index.js 6.51 KiB [built]
     | ./gui/js/comp/UiComponents.js 10 KiB [built]
     | ./gui/js/comp/WifiPage.js 7.92 KiB [built]
     | ./gui/js/functions/configHelpers.js 4.57 KiB [built]
     | ./gui/js/comp/ConfigPage.js 1.96 KiB [built]
     | ./gui/js/comp/DashboardPage.js 4.14 KiB [built]
     | ./gui/js/comp/FilePage.js 393 bytes [built]
     | ./gui/js/comp/FirmwarePage.js 7.62 KiB [built]
     | ./gui/js/comp/DashboardItems.js 7.3 KiB [built]
     | ./gui/js/comp/FileListing.js 8.48 KiB [built]
     | ./gui/js/comp/ControlItem.js 5.52 KiB [built]
     | ./gui/js/comp/DisplayItem.js 2.08 KiB [built]
     |     + 332 hidden modules
    + 403 hidden modules

WARNING in asset size limit: The following asset(s) exceed the recommended size limit (244 KiB).
This can impact web performance.
Assets: 
  bundle.js (447 KiB)
  ./index.html (451 KiB)

[... 425 lines cut ...]
  File &quot;C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py&quot;, line 880:
    if _is_builtin(spec):
  File &quot;C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py&quot;, line 872:
    for lb in self.env.GetLibBuilders():
  File &quot;C:\Users\shalk\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py&quot;, line 219:
    return self.method(*nargs, **kwargs)
  File &quot;C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py&quot;, line 993:
    lb = LibBuilderFactory.new(env, lib_dir)
  File &quot;C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py&quot;, line 61:
    obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)
  File &quot;C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py&quot;, line 137:
    self.process_extra_options()
  File &quot;C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py&quot;, line 282:
    exports={&quot;env&quot;: self.env, &quot;pio_lib_builder&quot;: self},
  File &quot;C:\Users\shalk\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py&quot;, line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File &quot;C:\Users\shalk\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py&quot;, line 287:
    exec(compile(scriptdata, scriptname, &#39;exec&#39;), call_stack[-1].globals)
  File &quot;C:\Users\shalk\Documents\PlatformIO\Projects\Test_Project_Framework\.pio\libdeps\esp12e\ESP8266 IoT Framework\scripts\preBuild.py&quot;, line 45:
    preBuildConfigFun()
[... cut ...] e\ESP8266 IoT Framework\scripts\preBuildConfig.py&quot;, line 40:
    cpp.write(&quot;\t\&quot;&quot; + item[&#39;value&#39;] + &quot;\&quot;&quot;)
==================================================================== [FAILED] Took 22.37 seconds ====================================================================The terminal process &quot;C:\Users\shalk\.platformio\penv\Scripts\pio.exe &#39;run&#39;&quot; terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

clabnet, 2020-12-14:
See log please

On 14/12/2020 20:59, Shalke04 wrote:
> option &#39;build_flags&#39; in section &#39;env:esp12e&#39; already 
> exists&#39;

Shalke04, 2020-12-14:
> ** Mon, 14 Dec 2020 12:00:47 -0800 - "maakbaas/esp8266-iot-framework" <reply@reply.github.com>, "maakbaas/esp8266-iot-framework" <esp8266-iot-framework@noreply.github.com> **

Its working already. I realized that I need to copy also dashboard json file into project and add file path. You did really great job with this project

Shalke04, 2020-12-15:
Hi, is there any easy way of disabling not needed features, such as File manager, Firmware update?

clabnet, 2020-12-15:
Hi,

the easy way is to comment menu entries on gui>js>index.js
and regenerate html activating  REBUILD_HTML build flag.

Ciao

Shalke04, 2020-12-15:
Can be added X,Y axis labels on the graph?  
  

> ** Tue, 15 Dec 2020 10:19:51 -0800 - "maakbaas/esp8266-iot-framework"
<reply@reply.github.com> **  
  

>

>

>

> Hi,  
  
the easy way is to comment menu entries on gui>js>index.js  
and regenerate html activating  REBUILD_HTML build flag.  
  
Ciao  

> Hi, is there any easy way of disabling not needed features, such as  
> File manager, Firmware update?  
>  
>  
> > ** Mon, 14 Dec 2020 12:00:47 -0800 -  
> "maakbaas/esp8266-iot-framework" <reply@reply.github.com>,  
> "maakbaas/esp8266-iot-framework"  
> <esp8266-iot-framework@noreply.github.com> **  
> >  
> > See log please  
> >  
> > > option 'build_flags' in section 'env:esp12e'  
> already  
> > > exists'  
> >  
> >  
> >  
> https://github.com/maakbaas/esp8266-iot-
framework/issues/65#issuecomment-744676112  
>  
>  
> <https://github.com/maakbaas/esp8266-iot-
framework/issues/65#issuecomment-745468610>,  
auth/ABOJLGQZSWJSUM3TRNPZIF3SU6RAVANCNFSM4UYRKFGA>.  

GitHub](https://github.com/maakbaas/esp8266-iot-
framework/issues/65#issuecomment-745473865), or
auth/AOKZYZ2VWXKGX3CLXUTJ5MLSU6SEPANCNFSM4UYRKFGA).

maakbaas, 2020-12-16:
I am not planning to work on that, but feel free to add the functionality yourself. It can be considered as a pul request.

Shalke04, 2020-12-17:
> ** Wed, 16 Dec 2020 01:43:40 -0800 - "maakbaas/esp8266-iot-framework" <reply@reply.github.com>, "maakbaas/esp8266-iot-framework" <esp8266-iot-framework@noreply.github.com> **

Hi, I reinstal Python to the lastest version and would like to know what else I need to install in order to make it work. Dashboard example doesnt show any data, it just says  There are no items defined in the JSON file, I can compile the project but there are many warnings such as  [webpack-cli] Error: Cannot find module ‘html-webpack-plugin’ and a lot of npm ERR!. Any ideas?