maakbaas/esp8266-iot-framework #67: preBuildCertificates.py SyntaxError: invalid syntax

JEDtoo, 2020-12-14:
I don't do python or any other language named after a snake. I've hacked some .py a few times, though.
I can't even tell by inspection if it's python 2 or python 3, so I tried both.

jed@servpi1:~ $ python --version
Python 2.7.13
jed@servpi1:~ $ python preBuildCertificates.py
  File "preBuildCertificates.py", line 94
    print('Added: ' + re.sub(f'[^{re.escape(string.printable)}]', '', names[i]), flush=True)       
                                                               ^
SyntaxError: invalid syntax
(   actually, arrow is under last character (') of printable)}]'    )

How about python3?
jed@servpi1:~ $ python3 --version
Python 3.5.3
jed@servpi1:~ $ python3 preBuildCertificates.py
  File "preBuildCertificates.py", line 94
    print('Added: ' + re.sub(f'[^{re.escape(string.printable)}]', '', names[i]), flush=True)       
                                                               ^
SyntaxError: invalid syntax

Exactly the same error.

All the parentheses, brackets, and quotes match. This is beyond my poor ability in this language.
If only it were Fortran...

Just FYI:
jed@servpi1:~ $ uname -a
Linux servpi1 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
jed@servpi1:~ $ cat /proc/device-tree/model
Raspberry Pi 3 Model B Plus Rev 1.3

maakbaas, 2020-12-14:
I also dislike Python ;). Welcome to the team.

Have you tried running the scripts through PlatformIO during the build process? This uses the internal Python version of PlatformIO 

But I agree it is strange that you get syntax errors using both Python versions. I am not able to look at this in more detail in the coming days, sorry about that. But I can have a look later.

JEDtoo, 2020-12-15:
No, I haven't done Visual Studio (isn't that where PlatformIO is?) in perhaps 10 years. I've been using the Arduino IDE platform for this IoT project. All I'm trying to do here is get the current weather conditions from https://api.weather.gov and wanted to include the certificates with this certificates.h. I help pay for the government's sensors - I might as well use them instead of deploy my own.

A "look later" is fine and will be appreciated. No lives depend upon this.

Happy Holidays, too!

clabnet, 2020-12-16:
@JEDtoo This code run fine on PlatformIO, used as addons on Visual Studio Code, and not Visual Studio :o).
Using on Arduino IDE is very limited because maakbas have write scripts for code generation not exactly simple. 
There is a need for a serious platform like PlatformIO.

maakbaas, 2021-02-17:
I will close this issue, since it seems to be related to using the files outside of the framework. Feel free to open a new one if you can better define the issue and we can have another look.

closed 2021-02-17 by maakbaas