maakbaas/esp8266-iot-framework #68: Identifier is undefined

Shalke04, 2020-12-16:
After whole day of using the code without problems Im facing issues with Platformio: IntelliSense Index Rebuild running again and again and again. And getting errors with Identifiers are undefined.
[image]

> Executing task in folder Test_Project_Framework: C:\Users\shalk\.platformio\penv\Scripts\pio.exe run <

Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2.6.2) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft

> esp@1.0.0 build C:\Users\shalk\Documents\PlatformIO\Projects\Test_Project_Framework\.pio\libdeps\esp12e\ESP8266 IoT Framework
> webpack --mode=production

Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
Hash: 110bdeef2c8bae2456ea
Version: webpack 4.42.1
Time: 10839ms
Built at: 16. 12. 2020 9:16:40
          Asset      Size  Chunks                    Chunk Names
   ./index.html   452 KiB          [emitted]  [big]  
./index.html.gz   118 KiB          [emitted]
      bundle.js   448 KiB       0  [emitted]  [big]  main
   bundle.js.gz   117 KiB          [emitted]
       main.css  3.52 KiB    0, 0  [emitted]         main, main
    main.css.gz  1.01 KiB          [emitted]
Entrypoint main [big] = main.css main.css bundle.js
 [1] ./node_modules/preact/compat/dist/compat.module.js + 2 modules 19.2 KiB {0} [built]
     |    3 modules
 [3] ./gui/js/configuration.json 927 bytes {0} [built]
 [8] ./gui/js/dashboard.json 2.1 KiB {0} [built]
[11] (webpack)/buildin/global.js 472 bytes {0} [built]
[28] ./node_modules/@emotion/is-prop-valid/dist/is-prop-valid.browser.esm.js + 1 modules 4.41 KiB {0} [built]
     |    2 modules
[70] ./gui/js/index.js + 343 modules 897 KiB {0} [built]
     | ./gui/js/index.js 6.51 KiB [built]
     | ./gui/js/comp/UiComponents.js 10 KiB [built]
     | ./gui/js/comp/WifiPage.js 7.92 KiB [built]
     | ./gui/js/functions/configHelpers.js 4.57 KiB [built]
     | ./gui/js/comp/ConfigPage.js 1.96 KiB [built]
     | ./gui/js/comp/DashboardPage.js 4.14 KiB [built]
     | ./gui/js/comp/FilePage.js 393 bytes [built]
     | ./gui/js/comp/FirmwarePage.js 7.62 KiB [built]
     | ./gui/js/comp/DashboardItems.js 7.3 KiB [built]
     | ./gui/js/comp/FileListing.js 8.48 KiB [built]
     | ./gui/js/comp/ControlItem.js 5.52 KiB [built]
     | ./gui/js/comp/DisplayItem.js 2.08 KiB [built]
     |     + 332 hidden modules
    + 403 hidden modules

WARNING in asset size limit: The following asset(s) exceed the recommended size limit (244 KiB).
This can impact web performance.
Assets:
  bundle.js (448 KiB)
  ./index.html (452 KiB)

WARNING in entrypoint size limit: The following entrypoint(s) combined asset size exceeds the recommended limit (244 KiB). This can impact web performance.
Entrypoints:
  main (452 KiB)
      main.css
      bundle.js

WARNING in webpack performance recommendations: 
You can limit the size of your bundles by using import() or require.ensure to lazy load some parts of your application.
For more info visit https://webpack.js.org/guides/code-splitting/
Child html-webpack-plugin for "index.html":
     1 asset
    Entrypoint undefined = ./index.html
    [0] ./node_modules/html-webpack-plugin/lib/loader.js!./gui/index.html 277 bytes {0} [built]
Child mini-css-extract-plugin ../ESP8266 IoT Framework\node_modules\css-loader\dist\cjs.js!../ESP8266 IoT Framework\node_modules\react-vis\dist\style.css:
    Entrypoint mini-css-extract-plugin = *
       2 modules
KeyError: 'value':
  File "C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\shalk\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\shalk\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\shalk\.platformio\platforms\espressif8266\builder\main.py", line 238:
    target_elf = env.BuildProgram()
  File "C:\Users\shalk\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\shalk\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\shalk\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 880:
    if _is_builtin(spec):
  File "C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 872:
    for lb in self.env.GetLibBuilders():
  File "C:\Users\shalk\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 993:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 61:
    obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)
  File "C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 137:
    self.process_extra_options()
  File "C:\Users\shalk\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 282:
    exports={"env": self.env, "pio_lib_builder": self},
  File "C:\Users\shalk\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\shalk\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\shalk\Documents\PlatformIO\Projects\Test_Project_Framework\.pio\libdeps\esp12e\ESP8266 IoT Framework\scripts\preBuild.py", line 45:
    preBuildConfigFun()
  File "C:\Users\shalk\Documents\PlatformIO\Projects\Test_Project_Framework\.pio\libdeps\esp12e\ESP8266 IoT Framework\scripts\preBuildConfig.py", line 43:
    cpp.write("\t" + str(item['value']).lower())
==================================================================== [FAILED] Took 24.60 seconds ====================================================================The terminal process "C:\Users\shalk\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

clabnet, 2020-12-16:
> Webp net-gifmaker (3)

There are not problem of this code.

Google exist fortunately.

maakbaas, 2020-12-16:
Thanks for reporting, however I agree with clabnet, this is related to your local toolchain or PlatformIO, and not to this repository :).

closed 2020-12-16 by maakbaas

maxgerhardt, 2020-12-17:
For completeness sake since this was also discussed in https://community.platformio.org/t/cannot-build-the-project-keyerror-value/17834/

> KeyError: 'value':
 File "C:\Users\shalk\Documents\PlatformIO\Projects\Test_Project_Framework.pio\libdeps\esp12e\ESP8266 IoT Framework\scripts\preBuildConfig.py", line 43:
>cpp.write("\t" + str(item['value']).lower())

https://github.com/maakbaas/esp8266-iot-framework/blob/4ef80a1c1ee1e99573cc14480b01ad87445ed013/scripts/preBuildConfig.py#L14-L47

The extra script of the library in this repository is throwing that error (see stacktrace), but it seems to do a malformed .json file of the user, if it can't read the `value` attribute of the object in the JSON file.

FernandoGarcia, 2021-07-14:
Hi @maakbaas !

I'm getting this error for every single example that have tested.

`  File "/home/fernandogarcia/Downloads/ESP8266-Iot-framework-test-main/remote-esp-web/.pio/libdeps/esp12e/ESP8266 IoT Framework/scripts/preBuildConfig.py", line 61:
    cpp.write('\t"' + item["value"] + '"')`

I think it doesn't look like an user error since the content of `"/../gui/js/configuration.json"` is created automatically for your script.

Also I think there's no error in path since the file is opened correctly.

I have checked it using this piece of code before `preBuildConfigFun()` function.

```
import sys
filename = inspect.getframeinfo(inspect.currentframe()).filename
dir_path = os.path.dirname(os.path.abspath(filename))
filename = "config"
h = open(dir_path + "/../src/generated/" + filename + ".h", "w", encoding="utf8")
cpp = open(dir_path + "/../src/generated/" + filename + ".cpp", "w", encoding="utf8")
configFile = open(dir_path + "/../gui/js/configuration.json")

if configFile:
    print("Opened!")

    for line in configFile:
        print(line)
sys.exit()
```
Here you can see the output of this code.

[image]

Could you give some suggestion to fix this problem please?

Best regards.

maakbaas, 2021-07-14:
Are you sure you did not modify the code?

This line does not exist in the standard script:
`cpp.write('\t"' + item["value"] + '"')`

FernandoGarcia, 2021-07-14:
Max has [quoted](https://github.com/maakbaas/esp8266-iot-framework/issues/68#issuecomment-747444708) the code above. [Line 40](https://github.com/maakbaas/esp8266-iot-framework/blob/4ef80a1c1ee1e99573cc14480b01ad87445ed013/scripts/preBuildConfig.py#L40).

FernandoGarcia, 2021-07-14:
Full compiling output:

<details>
  <summary>Click to expand</summary>
> esp@1.0.0 build
> webpack --mode=production

Hash: 30c32e37ec7c4bc7986f
Version: webpack 4.42.1
Time: 9014ms
Built at: 14/07/2021 17:04:50
          Asset      Size  Chunks                    Chunk Names
   ./index.html   453 KiB          [emitted]  [big]  
./index.html.gz   122 KiB          [emitted]         
      bundle.js   450 KiB       0  [emitted]  [big]  main
   bundle.js.gz   120 KiB          [emitted]         
       main.css  3.52 KiB    0, 0  [emitted]         main, main
    main.css.gz  1.01 KiB          [emitted]         
Entrypoint main [big] = main.css main.css bundle.js
 [0] ./node_modules/preact/compat/dist/compat.module.js + 2 modules 19.2 KiB {0} [built]
     |    3 modules
 [2] ./gui/js/configuration.json 1.63 KiB {0} [built]
 [4] ./gui/js/lang/en.json 2.25 KiB {0} [built]
 [7] ./gui/js/lang sync ^\.\/.*\.json$ 216 bytes {0} [built]
[10] ./gui/js/dashboard.json 10 bytes {0} [built]
[12] (webpack)/buildin/global.js 472 bytes {0} [built]
[29] ./node_modules/@emotion/is-prop-valid/dist/is-prop-valid.browser.esm.js + 1 modules 4.41 KiB {0} [built]
     |    2 modules
[46] ./gui/js/lang/de.json 2.38 KiB {0} [optional] [built]
[47] ./gui/js/lang/nl.json 2.34 KiB {0} [optional] [built]
[48] ./gui/js/lang/zh.json 2.12 KiB {0} [optional] [built]
[74] ./gui/js/index.js + 335 modules 1000 KiB {0} [built]
     | ./gui/js/index.js 7.28 KiB [built]
     | ./gui/js/comp/UiComponents.js 10.7 KiB [built]
     | ./gui/js/comp/WifiPage.js 8.22 KiB [built]
     | ./gui/js/functions/configHelpers.js 5.25 KiB [built]
     | ./gui/js/comp/ConfigPage.js 2.22 KiB [built]
     | ./gui/js/comp/DashboardPage.js 4.45 KiB [built]
     | ./gui/js/comp/FilePage.js 785 bytes [built]
     | ./gui/js/comp/FirmwarePage.js 7.83 KiB [built]
     | ./gui/js/comp/DashboardItems.js 9.06 KiB [built]
     | ./gui/js/comp/FileListing.js 8.68 KiB [built]
     | ./gui/js/comp/ControlItem.js 7.27 KiB [built]
     | ./gui/js/comp/DisplayItem.js 2.41 KiB [built]
     |     + 324 hidden modules
    + 129 hidden modules

WARNING in asset size limit: The following asset(s) exceed the recommended size limit (244 KiB).
This can impact web performance.
Assets: 
  bundle.js (450 KiB)
  ./index.html (453 KiB)

WARNING in entrypoint size limit: The following entrypoint(s) combined asset size exceeds the recommended limit (244 KiB). This can impact web performance.
Entrypoints:
  main (453 KiB)
      main.css
      bundle.js

WARNING in webpack performance recommendations: 
You can limit the size of your bundles by using import() or require.ensure to lazy load some parts of your application.
For more info visit https://webpack.js.org/guides/code-splitting/
Child html-webpack-plugin for "index.html":
     1 asset
    Entrypoint undefined = ./index.html
    [0] ./node_modules/html-webpack-plugin/lib/loader.js!./gui/index.html 277 bytes {0} [built]
Child mini-css-extract-plugin ../ESP8266 IoT Framework/node_modules/css-loader/dist/cjs.js!../ESP8266 IoT Framework/node_modules/react-vis/dist/style.css:
    Entrypoint mini-css-extract-plugin = *
       2 modules
KeyError: 'value':
[... 18 lines cut ...]
  File "/home/fernandogarcia/.platformio/penv/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 1062:
    project.install_dependencies()
  File "/home/fernandogarcia/.platformio/penv/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 880:
    if _is_builtin(spec):
  File "/home/fernandogarcia/.platformio/penv/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 872:
    for lb in self.env.GetLibBuilders():
  File "/home/fernandogarcia/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "/home/fernandogarcia/.platformio/penv/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 993:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "/home/fernandogarcia/.platformio/penv/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 61:
    obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)
  File "/home/fernandogarcia/.platformio/penv/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 137:
    self.process_extra_options()
  File "/home/fernandogarcia/.platformio/penv/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 282:
    exports={"env": self.env, "pio_lib_builder": self},
  File "/home/fernandogarcia/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/fernandogarcia/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/fernandogarcia/Downloads/ESP8266-Iot-framework-test-main/remote-esp-web/.pio/libdeps/esp12e/ESP8266 IoT Framework/scripts/preBuild.py", line 58:
    preBuildConfigFun()
  File "/home/fernandogarcia/Downloads/ESP8266-Iot-framework-test-main/remote-esp-web/.pio/libdeps/esp12e/ESP8266 IoT Framework/scripts/preBuildConfig.py", line 47:
    cpp.write('\t"' + item["value"] + '"')
</details>

FernandoGarcia, 2021-07-14:
If do you want try reproduce this error you can use this files.

[IOT_framework.zip](https://github.com/maakbaas/esp8266-iot-framework/files/6819194/IOT_framework.zip)

maakbaas, 2021-07-15:
You used the example JSON file from the dashboard for the configuration manager. For the configuration manager, the field `value` in the JSON file is mandatory, that is what you see in the error message. Fix this by changing your `platformio.ini` to:

`build_flags = -DDASHBOARD_PATH=webconfig.json -DREBUILD_HTML`

FernandoGarcia, 2021-07-15:
Hi!

Thanks for your help.

I think there's a gap in the documentation in this point.

I did a [PR](https://github.com/maakbaas/esp8266-iot-framework/pull/114) to try leave it more clear.

Best regards.