maakbaas/esp8266-iot-framework #70: Cannot install it correctly on macOS Big Sur

varosi, 2020-12-16:
After installing the library via PlatfromIO on Visual Studio Code and when I try to build the project I get:
```
ModuleNotFoundError: No module named 'asn1crypto':
  File "/Users/home/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/home/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
```
I have already installed asn1crypto globally, but it's strange that this is a some other environment. My Python is 3.9, not 3.8. 
Is it possible this library to auto-install asn1crypto where is needed during installation process in Visual Studio Code?

varosi, 2020-12-16:
The same happen on Windows also.

maxgerhardt, 2020-12-16:
Just open a [PIO terminal](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and do a `pip3 install asn1crypto`.

varosi, 2020-12-16:
And this is the result:
`Requirement already satisfied: asn1crypto in /usr/local/lib/python3.9/site-packages (1.4.0)`

This still doesn't help.

maxgerhardt, 2020-12-16:
> Requirement already satisfied: asn1crypto in /usr/local/lib/python3.9/site-packages (1.4.0)
>  File "/Users/home/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 170:

You're still using your global 3.9 installation and not the isolated 3.8 installation of Python. 

Try it this way in a normal terminal:
```
cd /Users/home/.platformio/penv/Scripts
./activate
./pip3 install asn1crypto
```

varosi, 2020-12-16:
There is no "Scripts" folder there.

maxgerhardt, 2020-12-16:
How did you install PlatformIO? Via the CLI first or just with the VSCode plugin?

varosi, 2020-12-16:
VSCode plugin

varosi, 2020-12-16:
There is /Users/home/.platformio/penv/ but without "Scripts". In "bin" folder there is "activate", but it's not executable.

maxgerhardt, 2020-12-16:
Is a `pip3` executable file in this folder with which you can do `./pip3 install asn1crypto`?

varosi, 2020-12-16:
Yes, there is, but it still give "Requirement already satisfied: asn1crypto in /usr/local/lib/python3.9/site-packages (1.4.0)"

maxgerhardt, 2020-12-16:
Okay please post the output of `pio system info` when executed in a PIO terminal.

varosi, 2020-12-16:
It worked! At least continued with the compilation.

Thanks a lot!
It could be easier I guess if the installation process of the library include that dependency automatically.

closed 2020-12-16 by varosi

maxgerhardt, 2020-12-16:
I've proposed that in PR #71  :)

maakbaas, 2020-12-17:
@maxgerhardt Thanks for supporting, and for the pull request. Indeed automatic installation of this dependency (and others) was already on my list for some time. I appreciate it!