maakbaas/esp8266-iot-framework #72: npm ERROR!

Shalke04, 2020-12-18:
Hi, Im trying to build Dashboard example and Im getting these errors although it compiles fine. But with these errors dashboard doesnt work. Any advice?

[webpack-cli] Error: Cannot find module 'html-webpack-plugin'
Require stack:
- C:\Users\shalk\Documents\PlatformIO\Projects\CleanProject2\.pio\libdeps\esp12e\ESP8266 IoT Framework\webpack.config.js
- C:\Users\shalk\AppData\Roaming\npm\node_modules\webpack-cli\lib\groups\resolveConfig.js
- C:\Users\shalk\AppData\Roaming\npm\node_modules\webpack-cli\lib\webpack-cli.js
- C:\Users\shalk\AppData\Roaming\npm\node_modules\webpack-cli\lib\bootstrap.js
- C:\Users\shalk\AppData\Roaming\npm\node_modules\webpack-cli\bin\cli.js
- C:\Users\shalk\AppData\Roaming\npm\node_modules\webpack\bin\webpack.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:880:15)
    at Function.Module._load (internal/modules/cjs/loader.js:725:27)
    at Module.require (internal/modules/cjs/loader.js:952:19)
    at require (C:\Users\shalk\AppData\Roaming\npm\node_modules\webpack-cli\node_modules\v8-compile-cache\v8-compile-cache.js:159:20)
    at Object.<anonymous> (C:\Users\shalk\Documents\PlatformIO\Projects\CleanProject2\.pio\libdeps\esp12e\ESP8266 IoT Framework\webpack.config.js:1:89)
    at Module._compile (C:\Users\shalk\AppData\Roaming\npm\node_modules\webpack-cli\node_modules\v8-compile-cache\v8-compile-cache.js:192:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Module.require (internal/modules/cjs/loader.js:952:19) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    'C:\\Users\\shalk\\Documents\\PlatformIO\\Projects\\CleanProject2\\.pio\\libdeps\\esp12e\\ESP8266 IoT Framework\\webpack.config.js',
    'C:\\Users\\shalk\\AppData\\Roaming\\npm\\node_modules\\webpack-cli\\lib\\groups\\resolveConfig.js',
    'C:\\Users\\shalk\\AppData\\Roaming\\npm\\node_modules\\webpack-cli\\lib\\webpack-cli.js',
    'C:\\Users\\shalk\\AppData\\Roaming\\npm\\node_modules\\webpack-cli\\lib\\bootstrap.js',
    'C:\\Users\\shalk\\AppData\\Roaming\\npm\\node_modules\\webpack-cli\\bin\\cli.js',
    'C:\\Users\\shalk\\AppData\\Roaming\\npm\\node_modules\\webpack\\bin\\webpack.js'
  ]
}
npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! esp@1.0.0 build: `webpack --mode=production`
npm ERR! Exit status 2
npm ERR! 
npm ERR! Failed at the esp@1.0.0 build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\shalk\AppData\Roaming\npm-cache\_logs\2020-12-18T11_08_26_277Z-debug.log
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266 IoT Framework> 1.6.1
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson> 6.17.2
|   |-- <ArduinoJson> 6.17.2
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <EEPROM> 1.0
|   |-- <LittleFS(esp8266)> 0.1.0
|-- <ESP Async WebServer> 1.2.3
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 6.17.2
|-- <ArduinoJson> 6.17.2
|-- <LittleFS(esp8266)> 0.1.0
Building in release mode
npm ERR! missing script: build

[... 272 lines cut ...]
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <EEPROM> 1.0
|   |-- <LittleFS(esp8266)> 0.1.0
|-- <ESP Async WebServer> 1.2.3
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 6.17.2
|-- <ArduinoJson> 6.17.2
|-- <LittleFS(esp8266)> 0.1.0
Building in release mode
npm ERR! missing script: build

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\shalk\AppData\Roaming\npm-cache\_logs\2020-12-18T11_18_24_985Z-debug.log
Retrieving maximum program size .pio\build\esp12e\firmware.elf
Checking size .pio\build\esp12e\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  45.9% (used 37640 bytes from 81920 bytes)
Flash: [======    ]  57.2% (used 597060 bytes from 1044464 bytes)
==================================================================== [SUCCESS] Took 7.18 seconds ====================================================================
Terminal will be reused by tasks, press any key to close it.

clabnet, 2020-12-18:
This is not an error of this code. You must install packages with npm or yarn. 
This code is not simple for novices.  It need experiences of web platform and embedded platform. 
Google can help you. 

⁣---------
Claudio Barca

Ottieni BlueMail per Android ​

maakbaas, 2020-12-18:
Hi @Shalke04,

Sorry for the slow communication from my side. I am actually moving between houses, so I am much less available than normal. 

I want to refer back to an earlier comment I made to you:
https://github.com/maakbaas/esp8266-iot-framework/issues/65#issuecomment-744219176

If you get those NPM errors it means that the packages are not installed correctly. Can you check those steps again?

Shalke04, 2020-12-18:
> ** Fri, 18 Dec 2020 09:32:53 -0800 - "maakbaas/esp8266-iot-framework" <reply@reply.github.com>, "maakbaas/esp8266-iot-framework" <esp8266-iot-framework@noreply.github.com> **

Thats OK, I wish you good luck with moving. I have googled a lot to solve this issue, installed many things and unistall again. Really no idea what I have done not correctly. Already spend 2 days figuring out problem, but without success.

Shalke04, 2020-12-18:
> ** Fri, 18 Dec 2020 11:02:33 -0800 - "maakbaas/esp8266-iot-framework" <reply@reply.github.com>, "maakbaas/esp8266-iot-framework" <esp8266-iot-framework@noreply.github.com> **
> Can you comment on the steps I proposed? And can you show that your node_modules are installed correctly as a consequence?

Most of npm ERRORS are solved.  The remaining ones are there:
Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db

and this
npm ERR! missing script: build

Otherwise no problems.

wdmtech, 2021-01-03:
@Shalke04 I suspect you're not running `npm run build` in the same folder as `package.json`

maakbaas, 2021-02-17:
This issue will be closed since it is not completely clear if there is a remaining issue. Please open a new one if this is the case.

closed 2021-02-17 by maakbaas