maakbaas/esp8266-iot-framework #74: Adding new API Methods?

chriscla, 2021-01-21:
Hi, I have more of a question than an issue -- apologies if this is not the right forum.

What's the best way to add additional web API methods?  It looks like the webserver server object is private, so I don't see an easy way of adding additional methods.  Perhaps making it protected so that "bindall()" could be overridden?  Or is there a method that I'm missing? 

thanks for working on the framework!

maakbaas, 2021-02-17:
Hi chriscla,

Thanks for your question, and sorry for the delay in getting back to you. It is a good point, right now the only way to add API methods would be to modify the web server class, which is not ideal if you want to keep the framework files untouched.

I have made the server object public in a151e898244e83ccb2529aba58871aac1bc2bce8, so that you can define your own `server.on` calls in other parts of the application. I hope that helps.

closed 2021-02-17 by maakbaas

aaronse, 2022-02-28:
> Hi chriscla,
> 
> Thanks for your question, and sorry for the delay in getting back to you. It is a good point, right now the only way to add API methods would be to modify the web server class, which is not ideal if you want to keep the framework files untouched.
> 
> I have made the server object public in [a151e89](https://github.com/maakbaas/esp8266-iot-framework/commit/a151e898244e83ccb2529aba58871aac1bc2bce8), so that you can define your own `server.on` calls in other parts of the application. I hope that helps.

Thanks for this edit.  Am using public WebServer::server field to register additional apis enabling other non esp devices to send partial config/state changes. 

Already considered extending the framework to handle if partial config edits?  Curious if this is common scenario for folks.