maakbaas/esp8266-iot-framework #79: DNS not saved?

arjena, 2021-02-08:
I found that after a power cycle the set DNS (with static IP) is not saved in EEPROM. In WiFiManager there is no mention of dns in `storeToEEPROM`, but adding `configManager.internal.dns = dns.v4();` does not solve my issue.
I added a line to `configManager::begin` to print the contents of the EEPROM to the console and that indeed shows the content of dns as '0'. 
How to solve this?

maakbaas, 2021-02-17:
The line you added was indeed missing, thanks for reporting. I added this line.

Furthermore in the current implementation the DNS would only be stored if you also configured a static IP. Right now I have changed this so that you can also enter a DNS without entering a static IP. I hope this solves your issue.

closed 2021-02-17 by maakbaas

arjena, 2021-02-17:
I only used the DNS in combination with a fixed IP, otherwise I would not need a specific DNS...
I did a workaround by adding the Cloudflare dns (1.1.1.1) in the cpp file as the first dns. Since WiFi.config can take two dns's, this works, also when you set a dns in the UI. But that is a workaround, I'll try your new code later. 
Thanks!