maakbaas/esp8266-iot-framework #80: Modify Webpage or add a button with Link

Eisbaeeer, 2021-02-14:
Great Framework! 
Is it possible to create a button as weblink in Dashboard?
E.g.  <a href="https://www.google.de" target="new">Open Goolge</a>
I need such solution to get the user a Link.
Any example is welcome.

Eisbaeeer, 2021-02-15:
Ok, I checked, that I can provide e.g. an index.html to the /download/index.html by uploading a file via filemanager. This file will be presented in the users browser by using the "webServer.cpp. Is it possible to interact values between my code in loop and the webfile? Normaly I do that with `server.on("/config", getConfig);` or something else. But this will be done in the library and I will not change the whole lib. Is there a posibility to exchange values between the webinterface and my program? Don´t found something in den doc´s.
Greets

maakbaas, 2021-02-17:
Hi Eisbaeeer,

With the framework it is definitely possible to add your own links or other elements to the web interface. But the only way in which this is supported is by modifying the web interface files directly. I do not plan to add other methods to add elements like a link. Can you explain why that would be a problem for you?

closed 2021-02-17 by maakbaas