maakbaas/esp8266-iot-framework #85: NameError: name 'env' is not defined

roglio, 2021-02-21:
Once added "lib_deps = ESP8266 IoT Framework" to the platformio.ini a working nodemcuv2 project fails to build.

NameError: name 'env' is not defined:
  File "C:\Users\lzaccaria\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\lzaccaria\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\lzaccaria\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\lzaccaria\.platformio\platforms\espressif8266\builder\main.py", line 238:
    target_elf = env.BuildProgram()
  File "C:\Users\lzaccaria\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\lzaccaria\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\lzaccaria\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\lzaccaria\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\lzaccaria\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\lzaccaria\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1064:
    lib_builders = env.GetLibBuilders()
  File "C:\Users\lzaccaria\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\lzaccaria\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 993:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "C:\Users\lzaccaria\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 61:
    obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)
  File "C:\Users\lzaccaria\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 137:
    self.process_extra_options()
  File "C:\Users\lzaccaria\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 282:
    exports={"env": self.env, "pio_lib_builder": self},
  File "C:\Users\lzaccaria\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\lzaccaria\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\lzaccaria\Documents\platform.io\streetwash\.pio\libdeps\nodemcuv2\ESP8266 IoT Framework\scripts\preBuild.py", line 10:
    from preBuildCertificates import preBuildCertificatesFun
  File "C:\Users\lzaccaria\Documents\platform.io\streetwash\.pio\libdeps\nodemcuv2\ESP8266 IoT Framework\scripts\preBuildCertificates.py", line 10:
    env.Execute(

maakbaas, 2021-02-22:
It might be related to the changes made in #71.

Can you open a PlatformIO terminal, execute the command `pip install asn1crypto` and then try again? If this resolves it I will look into a fix.

roglio, 2021-02-22:
`pip3 install asn1crypto` installed the missing dependency but the error is still there.

roglio, 2021-02-22:
PS: removing `lib_deps = ESP8266 IoT Framework` does not fix the issue. The project seems "corrupted".

maakbaas, 2021-02-22:
Can you reproduce the issue in a new project?

smartynov, 2021-02-22:
I'm having the same issue. Created an empty project from scratch (copying a "hello world" example). Using Atom 1.54.0 and PlatformIO Core, version 5.1.0 on MacOS 10.15.7, python 3.9.2.

smartynov, 2021-02-22:
What helped me:

1. start Atom + platformio, open platformio terminal
2. type `pio system info` and locate "Python Executable" path in output
3. type `/Users/myusername/.platformio/penv/bin/python -m pip install asn1crypto` (starting with the path from system info)

(solution seen in issue #70)

roglio, 2021-02-22:
@maakbaas 
> Can you reproduce the issue in a new project?
Yes: each time I start a new project. I can build it before adding the lib_deps.

>  What version of the framework are you using, and what version of PlatformIO and the ESP8266 Arduino libraries?
is a fresh install then I suppose the latest of both (in windows but also under linux I get the same behavior)

@smartynov 
> type /Users/myusername/.platformio/penv/bin/python -m pip install asn1crypto (starting with the path from system info)
this worked for me! under linux works either: `~/.platformio/penv/bin/python -m pip install asn1crypto`

roglio, 2021-02-22:
It works on windows too but the path it slightly different: `c:\users\myusername\.platformio\penv\scripts\python.exe -m pip install asn1crypto`

maakbaas, 2021-02-23:
I have reworked the python script to resolve the initial env related error, so hopefully the manual pip commands are no longer needed :).

closed 2021-02-23 by maakbaas

roglio, 2021-02-24:
Still does not work :-(

Maybe there is a delay in the dependency distribution?

maakbaas, 2021-02-24:
Sorry, I did not yet update the PlatformIO package.

You can test the latest changes by using 

`lib_deps = https://github.com/maakbaas/esp8266-iot-framework`

in your `platformio.ini`.

roglio, 2021-02-24:
Thank you! It works :-)

commit 883ea9ed (2021-02-23): Move pip command (Fixes #85)