maakbaas/esp8266-iot-framework #87: WiFiManager flow

jbanaszczyk, 2021-02-26:
My suggestion for WiFiManager:

# Auto reconnect to defined SSID

If WiFi is started as Captive Portal it should be ready to auto reconnect to defined SID.

State machine could be something like that:

- begin
  - if can connect to SSID: just connect and leave in WIFI_STA
  - otherwise:
    - if SSID is empty: start WIFI_AP
    - if SSID is not empty: start WIFI_STA_AP, serve Captive Portal from AP, try to auto reconnect to defined SSID
      - if SSID become available (isConnected) - turn off access point: WIFI_STA
      - if connection with SSID is lost (! isConnected): switch back to WIFI_STA_AP, station will will try to reconnect, AP serves Captive Portal
      - if SSID was empty, ESP was started as WIFI_AP, and user fills SSID in WiFi Configuration:
        - switch to WIFI_STA_AP, station will try to connect
          - if it connects: turn off Captive Portal WIFI_STA

# Use projectName as hostname and APname

# Lost communication if static config is similar to DHCP

Try to reproduce:
- start ESP, connect to SSID
- write down values of localIp, gatewayIp, and SubnetMask
- force network settings with that values. It turns off dhcp deamon: `wifi_station_dhcpc_status() == DHCP_STOPPED`
- probably you would lost communication with WiFi router
  - probably you will not be able to connect with router until reouter will forget the connection (just restart router) 
- fix: `config(localIp, gatewayIp, SubnetMask)`, than turn of WiFi `WIFi.mode(WIFI_OFF);` than reconnect: `WIFi.mode(WIFI_STA);`

# Think about access point password 

in `WifiManager::startApMode()`: something like `WiFi.softAP(portalName, WiFi.softAPPSK());`

# -----------

Code supporting flow you will find here: https://github.com/jbanaszczyk/esp8266-iot-framework
It is not read for pull request because I'm not able to make proper `WiFiPage.js` (DHCP checkbox is not working properly)

Many thanks for your effort

maakbaas, 2021-03-06:
Hi Jacek, 

Thanks for your detailed suggestions. Let me respond to each one of them:

**Auto reconnect**
This sounds like a nice addition, but it is important that these changes will be tested thoroughly. The added complexity should not come at the cost of less stability. Are you encountering situations where the framework sometimes starts in AP mode, even though the WiFi network is present? Or is this feature mainly intended for cases where the configured WiFi network is not always available?

**use projectName**
Personally I don't think this is a good idea to add into the framework. It is already possible to set the AP name to the projectName with the current functions, in fact this is done in the hello world example. However, integrating this into the WiFiManager increases the interdepencies between the different parts of the framework. Also, I would like the framework to still work if the json files are completely empty.

**lost communication**
I am not sure if I follow this issue. Are you saying that you lose connection when you first connect using DHCP and then enter a static IP configuration with the same IP that was handed out by DHCP? I can try to reproduce this, but I believe during previous testing I have already done this in my home network without issues. If this is a potential issue, maybe it is different from router to router?

**AP Password**
Do you have a use case for this? Personally I would think that if the AP is mainly used to configure the WiFi settings there is no immediate need to have the AP page password protected. I don't see a problem in adding it as an optional configuration of course.

maakbaas, 2021-04-07:
I will close this issue for now since it is one month since my response. Feel free to reopen in the future.

closed 2021-04-07 by maakbaas