maakbaas/esp8266-iot-framework #95: Issue trying to deploy webserver with simple dashboard from example

vgreff, 2021-03-09:
Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (2.6.3) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft

> esp@1.0.0 build
> webpack --mode=production

Error: ENOENT: no such file or directory, open './dist/index.html.gz'
    at Object.openSync (node:fs:506:3)
    at Object.readFileSync (node:fs:402:35)
    at done (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\.pio\libdeps\nodemcuv2\ESP8266 IoT Framework\webpack.config.js:105:37)
    at AsyncSeriesHook.eval [as callAsync] (eval at create (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\webpack\node_modules\tapable\lib\HookCodeFactory.js:33:10), <anonymous>:18:1)
    at AsyncSeriesHook.lazyCompileHook (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\webpack\node_modules\tapable\lib\Hook.js:154:20)
    at onCompiled (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\webpack\lib\Compiler.js:271:21)
    at C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\webpack\lib\Compiler.js:681:15
    at AsyncSeriesHook.eval [as callAsync] (eval at create (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\webpack\node_modules\tapable\lib\HookCodeFactory.js:33:10), <anonymous>:6:1) 
    at AsyncSeriesHook.lazyCompileHook (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\webpack\node_modules\tapable\lib\Hook.js:154:20)
    at C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\webpack\lib\Compiler.js:678:31
    at AsyncSeriesHook.eval [as callAsync] (eval at create (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\webpack\node_modules\tapable\lib\HookCodeFactory.js:33:10), <anonymous>:6:1) 
    at AsyncSeriesHook.lazyCompileHook (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\webpack\node_modules\tapable\lib\Hook.js:154:20)
    at C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\webpack\lib\Compilation.js:1423:35
    at AsyncSeriesHook.eval [as callAsync] (eval at create (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\webpack\node_modules\tapable\lib\HookCodeFactory.js:33:10), <anonymous>:6:1) 
    at AsyncSeriesHook.lazyCompileHook (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\webpack\node_modules\tapable\lib\Hook.js:154:20)
    at C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\webpack\lib\Compilation.js:1414:32
    at eval (eval at create (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\webpack\node_modules\tapable\lib\HookCodeFactory.js:33:10), <anonymous>:11:1)
    at C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\terser-webpack-plugin\dist\index.js:321:9
    at step (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\terser-webpack-plugin\dist\TaskRunner.js:87:9)
    at C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\terser-webpack-plugin\dist\TaskRunner.js:111:15
    at tryCatcher (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\util.js:16:23)
    at Promise._settlePromiseFromHandler (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\promise.js:547:31)
    at Promise._settlePromise (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\promise.js:604:18)
    at Promise._settlePromise0 (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\promise.js:649:10)
    at Promise._settlePromises (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\promise.js:729:18)
    at Promise._fulfill (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\promise.js:673:18)
    at Promise._resolveCallback (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\promise.js:466:57)
    at Promise._settlePromiseFromHandler (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\promise.js:559:17)
    at Promise._settlePromise (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\promise.js:604:18)
    at Promise._settlePromise0 (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\promise.js:649:10)
    at Promise._settlePromises (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\promise.js:729:18)
    at Promise._fulfill (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\promise.js:673:18)
    at Promise._resolveCallback (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\promise.js:466:57)
    at Promise._settlePromiseFromHandler (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\promise.js:559:17)
    at Promise._settlePromise (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\promise.js:604:18)
    at Promise._settlePromise0 (C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\node_modules\bluebird\js\release\promise.js:649:10)
npm ERR! code 1
npm ERR! path C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\.pio\libdeps\nodemcuv2\ESP8266 IoT Framework
npm ERR! command failed
npm ERR! command C:\WINDOWS\system32\cmd.exe /d /s /c webpack --mode=production

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\vince\AppData\Local\npm-cache\_logs\2021-03-09T15_45_37_809Z-debug.log
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266 IoT Framework> 1.7.1
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson> 6.17.3
|   |-- <ArduinoJson> 6.17.3
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <EEPROM> 1.0
|   |-- <LittleFS(esp8266)> 0.1.0
|-- <LittleFS(esp8266)> 0.1.0
Building in release mode
Compiling .pio\build\nodemcuv2\lib243\ESP8266 IoT Framework\webServer.cpp.o
.pio\libdeps\nodemcuv2\ESP8266 IoT Framework\src\webServer.cpp: In static member function 'static void webServer::serveProgmem(AsyncWebServerRequest*)':
.pio\libdeps\nodemcuv2\ESP8266 IoT Framework\src\webServer.cpp:162:89: error: 'html' was not declared in this scope
     AsyncWebServerResponse *response = request->beginResponse_P(200, PSTR("text/html"), html, html_len);
                                                                                         ^
.pio\libdeps\nodemcuv2\ESP8266 IoT Framework\src\webServer.cpp:162:95: error: 'html_len' was not declared in this scope
     AsyncWebServerResponse *response = request->beginResponse_P(200, PSTR("text/html"), html, html_len);
                                                                                               ^
*** [.pio\build\nodemcuv2\lib243\ESP8266 IoT Framework\webServer.cpp.o] Error 1
================================================================================================================================ [FAILED] Took 10.79 seconds ================================================================================================================================
The terminal process "C:\Users\vince\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

maakbaas, 2021-03-14:
Seems like you still have an issue with running webpack, and potentially with NPM indeed.

```
npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\vince\AppData\Local\npm-cache\_logs\2021-03-09T15_45_37_809Z-debug.log
```

What does this log file say?

vgreff, 2021-03-14:
> 0 verbose cli [
0 verbose cli   'C:\\Program Files\\nodejs\\node.exe',
0 verbose cli   'C:\\Users\\vince\\AppData\\Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js',
0 verbose cli   'run',
0 verbose cli   'build'
0 verbose cli ]
1 info using npm@7.6.1
2 info using node@v15.11.0
3 timing config:load:defaults Completed in 2ms
4 timing config:load:file:C:\Users\vince\AppData\Roaming\npm\node_modules\npm\npmrc Completed in 3ms
5 timing config:load:builtin Completed in 3ms
6 timing config:load:cli Completed in 2ms
7 timing config:load:env Completed in 0ms
8 timing config:load:file:C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\.pio\libdeps\nodemcuv2\ESP8266 IoT Framework\.npmrc Completed in 0ms
9 timing config:load:project Completed in 1ms
10 timing config:load:file:C:\Users\vince\.npmrc Completed in 1ms
11 timing config:load:user Completed in 1ms
12 timing config:load:file:C:\Users\vince\AppData\Roaming\npm\etc\npmrc Completed in 0ms
13 timing config:load:global Completed in 0ms
14 timing config:load:cafile Completed in 0ms
15 timing config:load:validate Completed in 0ms
16 timing config:load:setUserAgent Completed in 1ms
17 timing config:load:setEnvs Completed in 2ms
18 timing config:load Completed in 12ms
19 verbose npm-session d0b526ab11f7b133
20 timing npm:load Completed in 20ms
21 timing command:run Completed in 5209ms
22 verbose stack Error: command failed
22 verbose stack     at ChildProcess.<anonymous> (C:\Users\vince\AppData\Roaming\npm\node_modules\npm\node_modules\@npmcli\promise-spawn\index.js:64:27)
22 verbose stack     at ChildProcess.emit (node:events:378:20)
22 verbose stack     at maybeClose (node:internal/child_process:1067:16)
22 verbose stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
23 verbose pkgid esp@1.0.0
24 verbose cwd C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\.pio\libdeps\nodemcuv2\ESP8266 IoT Framework\scripts
25 verbose Windows_NT 10.0.18363
26 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\vince\\AppData\\Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js" "run" "build"
27 verbose node v15.11.0
28 verbose npm  v7.6.1
29 error code 1
30 error path C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\.pio\libdeps\nodemcuv2\ESP8266 IoT Framework
31 error command failed
32 error command C:\WINDOWS\system32\cmd.exe /d /s /c webpack --mode=production
33 verbose exit 1

closed 2021-03-14 by vgreff

vgreff, 2021-03-14:
> 
> 
> Seems like you still have an issue with running webpack, and potentially with NPM indeed.
> 
> ```
> npm ERR! A complete log of this run can be found in:
> npm ERR!     C:\Users\vince\AppData\Local\npm-cache\_logs\2021-03-09T15_45_37_809Z-debug.log
> ```
> 
> What does this log file say?

here is the content of the file C:\Users\vince\AppData\Local\npm-cache\_logs\2021-03-09T15_45_37_809Z-debug.log
I looked it it before, but I cannot tell what caused the error at line 22. Any idea?
 I'm clueless , clearly something failed but what?
I can't seem to get this webpack working, I'd like to know what I'm doing wrong or missing here.

0 verbose cli [
0 verbose cli   'C:\\Program Files\\nodejs\\node.exe',
0 verbose cli   'C:\\Users\\vince\\AppData\\Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js',
0 verbose cli   'run',
0 verbose cli   'build'
0 verbose cli ]
1 info using npm@7.6.1
2 info using node@v15.11.0
3 timing config:load:defaults Completed in 2ms
4 timing config:load:file:C:\Users\vince\AppData\Roaming\npm\node_modules\npm\npmrc Completed in 3ms
5 timing config:load:builtin Completed in 3ms
6 timing config:load:cli Completed in 2ms
7 timing config:load:env Completed in 0ms
8 timing config:load:file:C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\.pio\libdeps\nodemcuv2\ESP8266 IoT Framework\.npmrc Completed in 0ms
9 timing config:load:project Completed in 1ms
10 timing config:load:file:C:\Users\vince\.npmrc Completed in 1ms
11 timing config:load:user Completed in 1ms
12 timing config:load:file:C:\Users\vince\AppData\Roaming\npm\etc\npmrc Completed in 0ms
13 timing config:load:global Completed in 0ms
14 timing config:load:cafile Completed in 0ms
15 timing config:load:validate Completed in 0ms
16 timing config:load:setUserAgent Completed in 1ms
17 timing config:load:setEnvs Completed in 2ms
18 timing config:load Completed in 12ms
19 verbose npm-session d0b526ab11f7b133
20 timing npm:load Completed in 20ms
21 timing command:run Completed in 5209ms
22 verbose stack Error: command failed
22 verbose stack     at ChildProcess.<anonymous> (C:\Users\vince\AppData\Roaming\npm\node_modules\npm\node_modules\@npmcli\promise-spawn\index.js:64:27)
22 verbose stack     at ChildProcess.emit (node:events:378:20)
22 verbose stack     at maybeClose (node:internal/child_process:1067:16)
22 verbose stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
23 verbose pkgid esp@1.0.0
24 verbose cwd C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\.pio\libdeps\nodemcuv2\ESP8266 IoT Framework\scripts
25 verbose Windows_NT 10.0.18363
26 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\vince\\AppData\\Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js" "run" "build"
27 verbose node v15.11.0
28 verbose npm  v7.6.1
29 error code 1
30 error path C:\Users\vince\Documents\PlatformIO\Projects\WebKb-iot-fw\.pio\libdeps\nodemcuv2\ESP8266 IoT Framework
31 error command failed
32 error command C:\WINDOWS\system32\cmd.exe /d /s /c webpack --mode=production
33 verbose exit 1