maakbaas/esp8266-iot-framework #100: Feature request - ConfigManager

Eisbaeeer, 2021-03-30:
I´m loving your work. This framework is great.
Is it possible to get a "drop-down" option in configManager.
I´l provide the time-zones for time-sync in configManager. In this case it will be perfect if the user is able to select his timezone by drop-down list.
A drop-down list is really missing ;-)
Greets Lars

Eisbaeeer, 2021-03-30:
It seems, that aaronse realized something like that:
https://github.com/aaronse/esp8266-iot-framework

[image]

maakbaas, 2021-04-07:
That is right, he worked on some of that functionality, and proposed it in #44, but the pull request was mixed with other things. I agree that this is still a nice addition.

Eisbaeeer, 2021-04-21:
Yes, it will be perfect eg. timezone selection and many other drop-down options.

aaronse, 2021-04-21:
Will try to take another look at this over the weekend.  I need to sync and pull latest edits from master for current ESP32 based project am working on.

maakbaas, 2021-06-15:
Given that there was not much action on this item, I have added a drop down menu option in commit c7d6e45
which will be part of the next release.

For `control` items, `"control": "select",` will turn them into a drop down menu, with the options defined as  `"options": [1, 2, 3]`.

Eisbaeeer, 2021-06-17:
Hey, many thanks for that!
I have additional question.
In html it is possible to "label" each entry separately. This would be perfect, if it will be possible to label each or is there an other way to da that?
This is what I mean:
<select>
  <option value="value_of_volvo">label_of_Volvo</option>
</select>

Because I must Label like this:
label: TZ_Europe_Berlin
value: "CET-1CEST,M3.5.0,M10.5.0/3"

You will find that in TZ.h as well.
Greets Lars

maakbaas, 2021-06-17:
I would propose to add this as a field optionLabels in the JSON file. 

The implementation should be quite straightforward. If you or somebody else will add this in a pull request I'll happily merge it to the master.

Eisbaeeer, 2021-06-17:
I will try this. Hope I get it running.

Eisbaeeer, 2021-06-18:
I think Im not able to implement that. I added the optionLables and tried to change the ".pio\libdeps\nodemcuv2\ESP8266 IoT Framework\gui\js\comp\ControlItem.js"  but get it not running.
`   {
        "name": "Time_Zone",
        "label": "Time Zone",
        "type": "char",
        "length": 32,
        "value": "CET-1CEST,M3.5.0,M10.5.0/3",
        "control": "select",
        "options": [
            "<-01>1",
            "<-04>4",
            "<-08>8",
            "CET-1CEST,M3.5.0,M10.5.0/3"
        ],
        "optionLabels": [
            "TZ_Etc_GMTp1",
            "TZ_Etc_GMTp4",
            "TZ_Etc_GMTp8",
            "TZ_Europe_Berlin"
            ]
    },`

Eisbaeeer, 2021-06-29:
Can you give me some hints how to implement that?

maakbaas, 2021-06-29:
Is your code somewhere on GitHub to see what you tried?

Eisbaeeer, 2021-06-29:
No, because it´s not working. Im not firm with react and don´t know where to change the functions for "optionLabels". Tried to change the HTML part in ControllItem.js but this don´t worked. I got the "optionLabels" in source code of web-page in same struct as the "options" but the web-page always displayed the option istead of "optionLables". Im only edited the ControllItems.js.

benklop, 2021-08-12:
I've implemented optionLabels with https://github.com/maakbaas/esp8266-iot-framework/pull/120, but as this is my first attempt at React, it might need additional work.

perkinsinnovation, 2021-12-14:
Hi Maakbaas,

I'm trying to implement your drop down option, but it doesn't seem to be working. I'm running 1.9.0

I added the entry below to my configuration.json file but it shows up as a standard text field. There must be something I'm missing.

Any help would be greatly appreciated.

Thanks,
Shawn

[image]

{

        "name": "mode",

        "label": "Mode",

        "type": "char",

        "length": 20,

        "inputControl": "select",

        "value": "Aurora",

        "options": [

            "Aurora",

            "Blue",

            "Blue Wave",

            "Custom Color",

            "Custom Wave",

            "Sunset",

            "Ocean",

            "Rainbow",

            "White",

            "White Wave"

        ]

    }

maakbaas, 2021-12-14:
> Hi Maakbaas,
>
> I'm trying to implement your drop down option, but it doesn't seem to be working. I'm running 1.9.0
>
> I added the entry below to my configuration.json file but it shows up as a standard [...]

Should be `"control": "select"` instead of `"inputControl": "select"`

perkinsinnovation, 2021-12-14:
> Should be `"control": "select"` instead of `"inputControl": "select"`

I changed it, and downloaded to the target. Now my configuration page is blank. It sort of locks up, but I can refresh the browser and get to the other menus, such as file manager, and firmware update.

I'm using your ConfigManagerExample.cpp  with the configure.json. and platformio.ini below.

I've tried creating a new project in Platform IO, with a fresh checkout, and have the same result.

Any ideas?

One additional note. I've not been able to get the -REBUILD_HTML flag to work with my system. Perhaps a path problem?

I use "npm ci" and "npm run build" from the command line in place of using this flag.

[

    {

        "name": "projectName",

        "label": "Project Name",

        "type": "char",

        "length": 32,

        "value": "ESP8266 IoT Framework"

    },

    {

        "name": "dummyInt",

        "label": "Dummy Int",

        "type": "uint16_t",

        "value": 1000

    },

    {

        "name": "dummyBool",

        "label": "Dummy Bool",

        "type": "bool",

        "value": true

    },    

    {

        "name": "dummyFloat",

        "label": "Dummy Float",

        "type": "float",

        "min": 1,

        "max": 10,

        "value": 1.2345

        

    },

    {

        "name": "mode",

        "label": "Mode",

        "type": "char",

        "length": 20,

        "control": "select",

        "value": "Aurora",

        "options": [

            "Aurora",

            "Blue",

            "Blue Wave",

            "Custom Color",

            "Custom Wave",

            "Sunset",

            "Ocean",

            "Rainbow",

            "White",

            "White Wave"

        ]

    }

]

platformio.ini

[env:nodemcuv2]
platform = espressif8266@3.2.0
board = d1_mini
framework = arduino
monitor_speed = 115200
lib_deps = 
	ESP Async WebServer
	ArduinoJson
	;chunkysteveo/OctoPrintAPI@^1.1.5
	;mathertel/OneButton@^2.0.3
	;fastled/FastLED@^3.4.0
	;joaolopesf/RemoteDebug@^2.1.2

src_dir = /home/shawn/esp8266-iot-framework
extra_scripts = scripts/preBuild.py
build_flags = -DREBUILD_DASHBOARD -DREBUILD_CONFIG

perkinsinnovation, 2021-12-14:
Hi Maakbaas,

Today I reinstalled everything from scratch on a clean Windows system. (I was using Ubuntu) I installed 1.9.0 from your .zip file in case there was a possible problem with git.

All of my npm packages are installed properly, and my html is getting rebuilt correctly by your build system.

Everything works fine until I add "control": "select" to configure.json. It compiles fine without errors, but when I load the GUI from the target, the Configuration page hangs, and nothing is displayed. 

When "control": "select" is removed, the page loads fine from the target, but I'm not getting my pull down field that I'd like.

perkinsinnovation, 2021-12-15:
I figured out my problem. If you leave out the optionlables field, the Web browser will throw the following error:Uncaught TypeError: c.optionLabels is undefined

This is only seen using the inspector.

Once I added that field, works like a champ!

Thanks for your help, and thanks for all of your hard work on this project.

Eisbaeeer, 2021-12-15:
> I figured out my problem. If you leave out the optionlables field, the Web browser will throw the following error:Uncaught TypeError: c.optionLabels is undefined
> 
> This is only seen using the inspector.
> 
> Once I added that field, works like a champ!
> 

Can you paste the part of configuration.json with the correct optionlabel, please? 
This will help me, too.
Thanks!

perkinsinnovation, 2021-12-15:
Hi Eisbaeeer,

This is what got it working for me. I copied from an earlier post in this thread...I had to put it into a text editor to clean it up a bit. It's not the greatest example due to the crazy characters, but it will get you going!

"name": "Time_Zone", 
"label": "Time Zone", 
"type": "char", 
"length": 32, 
"value": "CET-1CEST,M3.5.0,M10.5.0/3", 
"control": "select", 
"options": [ "<-01>1", "<-04>4", 
"<-08>8", "CET-1CEST,M3.5.0,M10.5.0/3" ], 
"optionLabels": [ "TZ_Etc_GMTp1", "TZ_Etc_GMTp4", "TZ_Etc_GMTp8", "TZ_Europe_Berlin" ]

Eisbaeeer, 2021-12-15:
Thank you.  He, he. You use it for the same case as I will use it.

maakbaas, 2021-12-15:
Thanks for the feedback. If I understand correctly, optionLabels are now mandatory? That should not be the case. I will re-open this issue, and make sure that there is no error when no optionLabels are present.

perkinsinnovation, 2021-12-16:
> Thanks for the feedback.
> 
> That should not be the case. I will re-open this issue, and make sure that there is no error when no optionLabels are present.

Yes, this seems to be the case. No worries about this bug from my end, because I plan to use the optionLabels for my project.

Eisbaeeer, 2021-12-20:
Can anyone try the following json, please? It doesn´t work and I get a blank page:

`{ 
        "name": "Time_Zone", 
        "label": "Time Zone", 
        "type": "char", 
        "length": 40, 
        "value": "CET-1CEST,M3.5.0,M10.5.0/3",
        "control": "select",  
        "options": [
            "CET-1CEST,M3.5.0,M10.5.0/31",
            "22"
        ],
        "optionLabels": [
            "EU-Berlin",
            "Label 22"
        ]        
    },`

Eisbaeeer, 2021-12-26:
I can confirm the reported issue from @perkinsinnovation. You have to define always both. "options" AND "optionLabels" in all select definitions. Im relaxed, too, because it is no "showstopper" if you know about that.
Thanks for reopening!

maakbaas, 2021-12-27:
Fixed it

commit c7d6e45c (2021-06-15): Add dropdown (fixes #100)
commit a198a569 (2021-12-27): Make OptionLabels optional (fixes #100)
closed 2021-12-27 by maakbaas