maakbaas/esp8266-iot-framework #102: Broken path

kindredally, 2021-04-13:
I am building on macbook pro using VS + PlatformIO plugin.  I have tried both adding libraries and cloning from github.  The library built, but the github clone did not for the dashboard example.  Hello world worked.  This is on d1_mini.  A file not found came up when building.  The dst path starts "../gui/....." which causes a file not found error.  So I hard coded in the full path.  It seems there is a relative path in the project which doesn't work correctly.

shutil.py
line 261

 if not follow_symlinks and _islink(src):
        os.symlink(os.readlink(src), dst)
    else:
        with open(src, 'rb') as fsrc, open('/Users/robin/Documents/PlatformIO/Projects/esp8266-iot-framework/gui/'+dst, 'wb') as fdst:
            # macOS
            if _HAS_FCOPYFILE:
                try:

kindredally, 2021-04-13:
this is my final fix:

I had to modify preBuild.py (removing the ../):
elif isinstance(item, tuple) and item[0] == "CONFIG_PATH":
        copyfile(env.get("PROJECT_DIR") + '/' + item[1], 'gui/js/configuration.json')
    elif isinstance(item, tuple) and item[0] == "DASHBOARD_PATH":
        copyfile(env.get("PROJECT_DIR") + '/' + item[1], 'gui/js/dashboard.json')
    elif isinstance(item, tuple) and item[0] == "DOMAIN_LIST":
        domains = item[1]
    elif isinstance(item, tuple) and item[0].lower() == "openssl":
        openssl = item[1]

and I added the empty subdirectory 'include'

maakbaas, 2021-04-29:
I tried reproducing this both in a Windows and a Linux environment, but I struggle to see the issue. 

I would not expect this behavior on mac be different from Linux? The Python script is executed from within its own folder, hence the three dots are needed? I can't replace this with an absolute path, since the absolute path will be different when using the framework as a library...

closed 2021-04-29 by maakbaas

maakbaas, 2021-04-29:
Funny, just after typing this I finally got the same error you had. It seems this happened after clearing the .pio folder, maybe there is a changed behavior in a recent PlatformIO version.

Thanks for reporting, and I have commited your fix in 7428ffa290fc3853b13920e9aa6b8cb1639087e0