maakbaas/esp8266-iot-framework #104: Web Page Reload re-triggers

kindredally, 2021-04-14:
Hi.  First off, I want to say this project is a really cool idea!

I have implemented basically a single button press to activate a device open/close depending on the initial state of the device.

I use the little checkbox form on the dashboard.  The problem is if the web disconnects or the page is reloaded, before the checkbox can be cleared, the page re-sends the checkbox ...triggering the device a second time.

To avoid this, the web page needs to remember the event has been sent and/or reset itself so that a page refresh doesn't re-do the same change.

It would be nice to have a new variable type which acts as a one time button,  and password field which obfuscates what is typed. And a way to use your own button to send a string in instead of the upload icon/button.

JoergTiedemann, 2021-04-17:
Investors the same issue

maakbaas, 2021-04-29:
I have changed the checkbox event to trigger in case of `onClick` instead of `onChange` in commit b2bf375e6b97f95b07a525e86ddc60a83ab476fc. Hopefully this resolves unwanted triggers on page refresh.

Can you check with the latest SVN version by using:

`lib_deps = https://github.com/maakbaas/esp8266-iot-framework`

to see if your issue has been resolved? If not, can you provide more details on the use case, sample code, and what browser you are using?

maakbaas, 2021-05-05:
I will close this issue, feel free to re-open if it is not resolved.

closed 2021-05-05 by maakbaas