maakbaas/esp8266-iot-framework #105: Certificates

kindredally, 2021-04-29:
Hi.  The system is generating certificates, but is there a way to also include your own ones?  I want to access Amazon's mqtt server which requires three custom keys to authenticate.  The pubsub library won't work because it is not async...although it works on its own.  Using the async mqtt library seems like a good idea except the certificates don't seem compatible.

maakbaas, 2021-04-29:
do you have an example of what you would like to do, and why the root certificates included do not suffice?

kindredally, 2021-04-29:
It is because MQTT brokers often require it.  The AWS broker does support
user id + password , only authentication.  I don't have the code in front
of me, but if you search on async esp8266 mqtt aws or look in the Bear
header file it mentions mqtt and adding the certificate but I can't see how
the framework could coexist.

maakbaas, 2021-04-30:
Right now, no MQTT related functionality is implemented. However, this framework does not prevent you from using any code, default functions or libraries, so whatever you aim to do should be possible with your own code on top of or next to the framework.

You are always welcome to propose new added functionality to add MQTT for instance.

Unless you can show a specific example where this framework causes an issue I will close this thread for now.

closed 2021-04-30 by maakbaas