maakbaas/esp8266-iot-framework #107: fetch.POST Example

pallabseal, 2021-05-09:
Can you provide POST example using fetch class.

Eisbaeeer, 2021-06-14:
> Can you provide POST example using fetch class.

Hi. You can do that like this example. It is also described in documentation:

// define url and create body
String url = F("www.example.com");
String body = F("Place here the POST-string");

// call fetch function
fetch.begin(url);
fetch.POST(body);

majo551, 2021-10-19:
@Eisbaeeer  have you figured out the post example? For me personally it is not obvious anwer: 

`POST
int POST(String url, String body);
int POST(String body);
This method starts a POST request to the URL specified in url with the payload specified in body. The URL must include a http:// or https:// prefix. The method returns the HTTP response code. If you first called the begin method, the variant without the url argument should be used.`

I am lost from the documentation example. :-(

Eisbaeeer, 2021-10-19:
@majo551 Im using POST and get functions from the library. Did you tried the example?

majo551, 2021-10-20:
Can anyone direct me to the example where POST is described in details ? What is meant by body for POST function? Is this just the parameters or the post link or the whole POST structure?  The documentation shows only : 
[image]

Example shows only GET method:
[image]

Have I missed some part with POST explained?
Thanks

maakbaas, 2021-11-06:
If I follow the GET example but replace the `fetch.GET` line with your:

`
fetch.POST("https://www.postman-echo.com/post","id=test");
`

The code works perfectly fine for me, and I get the response:

`{"args":{},"data":"id=test","files":{},"form":{},"headers":{"x-forwarded-proto":"https","x-forwarded-port":"443","host":"www.postman-echo.com","x-amzn-trace-id":"Root=1-61865fc6-381c39734b63eef41a3dec28","content-length":"7","user-agent":"ESP8266HTTPClient","accept-encoding":"identity;q=1,chunked;q=0.1,*;q=0","content-type":"application/json"},"json":null,"url":"https://www.postman-echo.com/post"}`

I suspect your reset might be caused because you try to read the entire response string at once, rather than streaming it.

closed 2021-11-06 by maakbaas

vvilho, 2022-10-25:
[image]
I'm getting the same kind of response as @majo551. Is there anything else I need to set up differently to get the real response from the POST request.