maakbaas/esp8266-iot-framework #109: CertStoreBase Error

Eisbaeeer, 2021-06-12:
I updated to the framework 1.71
Now I got the error:
Compiling .pio\build\nodemcuv2\lib979\ESP8266WiFi\ESP8266WiFiAP.cpp.o
In file included from .pio\libdeps\nodemcuv2\esp8266-iot-framework-1.71\src\fetch.cpp:3:
C:\Users\Lars Weimar\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:116:23: error: 'CertStoreBase' has not been declared
  116 |     void setCertStore(CertStoreBase *certStore) {
      |                       ^~~~~~~~~~~~~

Tried to update the ssl lib on platformio. But issue not solved. Any hints?
Thank you!
Greet

maakbaas, 2021-06-13:
This is because the Arduino ESP8266 library has been updated to 3.0.0 which introduce many breaking changes.

For now, use in your `platformio.ini` the line:

`platform = espressif8266@^2` 

to force the previous version. I will look into this.

maakbaas, 2021-06-13:
Will be fixed in the next release:

https://github.com/maakbaas/esp8266-iot-framework/commit/8a752f8443a8bb770ce8165442981d1362a64422

closed 2021-06-13 by maakbaas