maakbaas/esp8266-iot-framework #110: WiFiClientSecureBearSSL.h:116:23: error: 'CertStoreBase' has not been declared

Oxoproline, 2021-06-28:
Hi, using latest version 1.8.1 I am getting this issue on build:

`Compiling .pio/build/esp12e/lib21d/ESPAsyncTCP/SyncClient.cpp.o
In file included from .pio/libdeps/esp12e/ESP8266 IoT Framework/src/fetch.cpp:3:
/Users/fmokry001/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:116:23: error: 'CertStoreBase' has not been declared
  116 |     void setCertStore(CertStoreBase *certStore) {
      |                       ^~~~~~~~~~~~~
/Users/fmokry001/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:163:5: error: 'CertStoreBase' does not name a type; did you mean 'CertStore'?
  163 |     CertStoreBase *_certStore;
      |     ^~~~~~~~~~~~~
      |     CertStore
/Users/fmokry001/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h: In member function 'void BearSSL::WiFiClientSecureCtx::setCertStore(int*)':
/Users/fmokry001/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:117:7: error: '_certStore' was not declared in this scope; did you mean 'certStore'?
  117 |       _certStore = certStore;
      |       ^~~~~~~~~~
      |       certStore
/Users/fmokry001/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h: At global scope:
/Users/fmokry001/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:302:23: error: 'CertStoreBase' has not been declared
  302 |     void setCertStore(CertStoreBase *certStore) { _ctx->setCertStore(certStore); }
      |                       ^~~~~~~~~~~~~
.pio/libdeps/esp12e/ESP8266 IoT Framework/src/fetch.cpp: In member function 'void HTTPRequest::begin(String, bool)':
.pio/libdeps/esp12e/ESP8266 IoT Framework/src/fetch.cpp:20:35: error: cannot convert 'BearSSL::CertStore*' to 'int*'
   20 |         httpsClient->setCertStore(&certStore);
      |                                   ^~~~~~~~~~
      |                                   |
      |                                   BearSSL::CertStore*
In file included from .pio/libdeps/esp12e/ESP8266 IoT Framework/src/fetch.cpp:3:
/Users/fmokry001/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:302:38: note:   initializing argument 1 of 'void BearSSL::WiFiClientSecure::setCertStore(int*)'
  302 |     void setCertStore(CertStoreBase *certStore) { _ctx->setCertStore(certStore); }
      |                       ~~~~~~~~~~~~~~~^~~~~~~~~
Compiling .pio/build/esp12e/lib21d/ESPAsyncTCP/tcp_axtls.c.o
*** [.pio/build/esp12e/lib438/ESP8266 IoT Framework/fetch.cpp.o] Error 1
Compiling .pio/build/esp12e/lib53b/Hash/Hash.cpp.o
src/main.cpp: In function 'bool CheckShouldRun(long unsigned int, int)':
src/main.cpp:200:36: warning: comparison of integer expressions of different signedness: 'long unsigned int' and 'int' [-Wsign-compare]
  200 |   return millis() - previousMillis >= interval;
      |          ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
.pio/libdeps/esp12e/ESP8266 IoT Framework/src/dashboard.cpp: In member function 'void dashboard::send()':
.pio/libdeps/esp12e/ESP8266 IoT Framework/src/dashboard.cpp:26:11: warning: 'void* memcpy(void*, const void*, size_t)' forming offset [4, 7] is out of the bounds [0, 4] of object 'now' with type 'long unsigned int' [-Warray-bounds]
   26 |     memcpy(buffer, reinterpret_cast<uint8_t *>(&now), 8);
      |     ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp12e/ESP8266 IoT Framework/src/dashboard.cpp:25:19: note: 'now' declared here
   25 |     unsigned long now = millis();
      |                   ^~~
=============================================================================================== [FAILED] Took 6.96 seconds ===============================================================================================
The terminal process "platformio 'run', '--environment', 'esp12e'" terminated with exit code: 1.`

Downgrade to 1.8.0 solve the issue.

Eisbaeeer, 2021-06-29:
Have a look here: https://github.com/maakbaas/esp8266-iot-framework/issues/109

maakbaas, 2021-06-29:
I can't see how there would be a difference between 1.8.1 and 1.8.0 with regards to this. Are you using the 3.0.0 Arduino libraries?How does your platformio.ini look like? If 1.8.0 works, I think 1.8.1 should work now as well. Maybe something else changed in your configuration during the downgrade.

closed 2021-06-29 by maakbaas

Oxoproline, 2021-06-29:
Thank you guys for reply. I indeed yesterday upgraded libraries to 3.0.0 and that probably caused the issue. Can explain what happened when I downgraded the iot framework lib., but after that it builded. Probably smth else happened I am not aware of because it make sense that even on 1.8.0 it should not build.