maakbaas/esp8266-iot-framework #112: is there a way to add button component for dashboard

zenonasz, 2021-07-13:
how can we add a button on dashboard or a switch (instead of the traditional boolean checkbox)?

can we somehow try to modify the input to show only the button? 

how can we access the dashboard input button?

aaronse, 2021-07-13:
Afaik No, would need to extend and create pull request to merge to main branch.

Previously created a switch and some other UI controls, bit there was understandable push back about the pull request having too many edits.  I didn't get around to teasing apart into smaller pull requests.
https://github.com/maakbaas/esp8266-iot-framework/pull/44

zenonasz, 2021-07-13:
> Afaik No, would need to extend and create pull request to merge to main branch.
> 
> Previously created a switch and some other UI controls, but there was understandable push back about the pull request having too many edits. I didn't get around to teasing apart into smaller pull requests. Example of how to add such edits are buried in Pull Request
> #44 I intend to come back to this after wrapping up some other projects if no one gets this done first. HTH

Ok got it, thanks.

closed 2021-07-13 by zenonasz