maakbaas/esp8266-iot-framework #118: Examples are missing the required build_flags

JonasHess, 2021-08-02:
I find it hard to follow along, what build flags are required to actually get started with the provided examples. It would really help if the platformio.ini file was included with the examples.

maakbaas, 2021-08-02:
I agree it is a bit hidden, but stated in the docs here:
https://github.com/maakbaas/esp8266-iot-framework/blob/master/docs/installation-guide.md:

I will see if I can make it more clear in the future.

**Quoted from the docs:**

When using the examples make sure to use the suitable build flags in your platformio.ini.

When using the configManager example the build flags should be:

```ini
build_flags = -DREBUILD_HTML -DCONFIG_PATH=path_where_you_copied_configuration.json
```

When using the dashboard example the build flags should be:

```ini
build_flags = -DREBUILD_HTML -DDASHBOARD_PATH=path_where_you_copied_dashboard.json
```

closed 2021-08-02 by maakbaas

djanknecht, 2023-10-30:
@maakbaas really DREBUILD_HTML, not DREBUILD_DASHBOARD? 
I cannot get it to work - the Dashboard still says: "There are no items defined in the JSON file"