maakbaas/esp8266-iot-framework #121: Question - redirect the browser to an URL inner loop

Eisbaeeer, 2021-08-12:
Hi.
New question. I should redirect the client browser to a external URL after processing internaly things in loop.
I use the fetch process to get an TOKEN from a external page. Now the received token should integrated in a URL and the client browser should redirected to that URL.
The external URL is: 'https://www.facebook.com/device?user_code=<TOKEN>'

With "normal" http server I would send that to the client via:
`client.print("<HEAD>");
client.print("<meta http-equiv=\"refresh\" content=\"0;url=https://www.facebook.com/device?user_code=<TOKEN>\">");
client.print("</head>");`
 
My target: After getting the TOKEN from facebook, the user should redirect to the URL above.
Greets Lars

Eisbaeeer, 2021-08-31:
Again. Is it possible to realize that on a other way. How can I provide a dynamic link, filled from loop() ?
This should be a clickable link as example:
[image]

Greets Lars

maakbaas, 2021-11-06:
Sorry for the huge delay in responding. Right now I am too busy to work on new features myself (As you probably figured out from this delay ;)). Proposed features will only be handled in the form of pull requests.

closed 2021-11-06 by maakbaas

Eisbaeeer, 2021-12-06:
> Proposed features will only be handled in the form of pull requests.
Wow, OK. This means you only will do bugfixes in future or is this project dead on now?

maakbaas, 2021-12-06:
The project is not dead, I will keep doing bugfixes, and keep integrating pull requests for new features. 

But right now my priorities do not allow me to spend time on the development of new features myself. Of course in the future that might change again.

Eisbaeeer, 2021-12-06:
@maakbaas Good to know! Thank you verry much for your statement. This will be fine because the project is awesome 👌

aaronse, 2021-12-06:
> The project is not dead, I will keep doing bugfixes, and keep integrating pull requests for new features.
> 
> But right now my priorities do not allow me to spend time on the development of new features myself. Of course in the future that might change again.

Totally understand.  Thank you for creating this project.  I appreciate the time and effort you and other contributors have invested.  Look forward to seeing/helping this project continue to evolve.