maakbaas/esp8266-iot-framework #127: No Webserver?

t10r, 2021-11-30:
Hi,

even the building was successful and the configuration.json and dashboard.json should work, I get this screen on the dashboard.

[image]

The developer of the software tried it also with my setup and get the same result.

My setup:

OS: Linux Mint 20.2
DevTool: Vscodium

If I can provide any more information please ask.

Thanks
Tim

maakbaas, 2021-11-30:
Did you use the -DREBUILD_HTML flag?

t10r, 2021-11-30:
Yes, that is my platformio.ino

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:nodemcuv2]
platform = espressif8266@3.0.0
board = nodemcuv2
framework = arduino
board_build.f_cpu = 160000000L
upload_speed = 460800
monitor_speed = 115200
build_type = debug
monitor_filters = esp8266_exception_decoder
lib_deps = 
	ArduinoJson
	ESP8266 IoT Framework
	me-no-dev/ESP Async WebServer @ ^1.2.3
	knolleary/PubSubClient @ ^2.8
build_flags = -DCONFIG_PATH=configuration.json -DDASHBOARD_PATH=dashboard.json -DREBUILD_HTML
extra_scripts = scripts/preBuild.py
```

maakbaas, 2021-11-30:
And can you share your build output log?

t10r, 2021-12-01:
Sure

```
 so@soso  ~/workspace/Ferraris_MQTT_Energy_Counter_Meter_TCRT5000/source/Ferraris_MQTT_Energy_Counter_Meter_TCRT5000   master ● ? ⍟1  pio run  
Processing nodemcuv2 (platform: espressif8266@3.0.0; board: nodemcuv2; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (3.0.0) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.30000.210519 (3.0.0) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa 2.100200.0 (10.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Running npm run build...
npm <command>

Usage:

npm install        install all the dependencies in your project
npm install <foo>  add the <foo> dependency to your project
npm test           run this project's tests
npm run <foo>      run the script named <foo>
npm <command> -h   quick help on <command>
npm -l             display usage info for all commands
npm help <term>    search for help on <term>
npm help npm       more involved overview

All commands:

    access, adduser, audit, bin, bugs, cache, ci, completion,
    config, dedupe, deprecate, diff, dist-tag, docs, doctor,
    edit, exec, explain, explore, find-dupes, fund, get, help,
    hook, init, install, install-ci-test, install-test, link,
    ll, login, logout, ls, org, outdated, owner, pack, ping,
    pkg, prefix, profile, prune, publish, rebuild, repo,
    restart, root, run-script, search, set, set-script,
    shrinkwrap, star, stars, start, stop, team, test, token,
    uninstall, unpublish, unstar, update, version, view, whoami

Specify configs in the ini-formatted file:
    /home/so/.npmrc
or on the command line via: npm <command> --key=value

More configuration info: npm help config
Configuration fields: npm help 7 config

npm@8.1.4 /usr/local/lib/node_modules/npm
Found 41 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 6.18.5
|-- <ESP8266 IoT Framework> 1.9.0
|   |-- <ArduinoJson> 6.18.5
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <EEPROM> 1.0
|   |-- <LittleFS(esp8266)> 0.1.0
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson> 6.18.5
|-- <ESP Async WebServer> 1.2.3
[... 215 lines cut ...]
Compiling .pio/build/nodemcuv2/FrameworkArduino/spiffs/spiffs_check.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/spiffs/spiffs_gc.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/spiffs/spiffs_hydrogen.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/spiffs/spiffs_nucleus.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/spiffs_api.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/sqrt32.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/stdlib_noniso.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/time.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/uart.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/umm_malloc/umm_info.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/umm_malloc/umm_integrity.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/umm_malloc/umm_local.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/umm_malloc/umm_malloc.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/umm_malloc/umm_poison.c.o
Archiving .pio/build/nodemcuv2/libFrameworkArduino.a
Indexing .pio/build/nodemcuv2/libFrameworkArduino.a
Linking .pio/build/nodemcuv2/firmware.elf
Retrieving maximum program size .pio/build/nodemcuv2/firmware.elf
Checking size .pio/build/nodemcuv2/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [======    ]  55.3% (used 45320 bytes from 81920 bytes)
Flash: [========= ]  89.3% (used 932264 bytes from 1044464 bytes)
Building .pio/build/nodemcuv2/firmware.bin
Creating BIN file ".pio/build/nodemcuv2/firmware.bin" using "/home/so/.platformio/packages/framework-arduinoespressif8266@3.30000.210519/bootloaders/eboot/eboot.elf" and ".pio/build/nodemcuv2/firmware.elf"
```

maakbaas, 2021-12-02:
Something is wrong with your toolchain. In the log the help for `npm` is shown, meaning it does not actually call the npm command. It should show like this:

'''
Running npm run build...

> esp@1.0.0 build
> webpack --mode=production
'''

closed 2021-12-02 by maakbaas

t10r, 2021-12-02:
Yes, that could be. But want should be wrong? Which npm Version is required to build your code sucessfull?