maakbaas/esp8266-iot-framework #131: Cant connect to own webserver via WAN & LAN

kkuez, 2021-12-11:
I have got a tomcat running a java rest https application. It is reachable via WAN and locally in LAN. It works on Android and curl, but when I make a request I cant connect.
Requests to other addresses like google.com or github.com work fine over https.

```
  const char *host = "192.168.178.119:8080";
  fetch.begin(host);
  fetch.addHeader("111", "111"); 
  fetch.addHeader("222", "222");
  int httpCode = fetch.GET(host);
  if (httpCode == HTTP_CODE_OK)
  {
    while (fetch.busy())
    {
      if (fetch.available())
      {
        Serial.write(fetch.read());
      }
    }
  }
  else
  {
    Serial.printf("HTTP-Status: %d\n", httpCode);
  }

  fetch.clean();
```

`Starting HTTPS Framework...  
fpm close 3
mode : sta(8c:aa:b5:78:87:3b)
add if0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0) 
state: 3 -> 5 (10)
add 0
aid 1
cnt

connected with Auenland, channel 1
dhcp client start...
ip:192.168.178.50,mask:255.255.255.0,gw:192.168.178.1
Connected to stored WiFi details
192.168.178.50
[HTTP-Client] failed connect to Host , Port: 0
HTTP-Status: -1
Looping...
pm open,type:2 0`

maakbaas, 2021-12-13:
You can use the contact form on maakbaas.com to share your server so I can have a look!

maakbaas, 2021-12-14:
I sent you a response by e-mail.

kkuez, 2021-12-15:
Great, thank you!
So, for documentation purposes Ill write what went wrong here.

If you intent to connect your esp to your own webserver, make sure its certificate has not expired x)
Good luck and thanks again maakbaas!