maakbaas/esp8266-iot-framework #134: Feature request - Bargraph

Eisbaeeer, 2021-12-18:
I know that only be "Pull requests" will be the thing now, and I will try it by myself, but I´m not realy good in react.
The project is realy great and I love it so much!
Perhaps anyone can implement a bargraph in react for configuration/dashboard for integer values?
This will be great for e.g. brightness control.
Greets Lars

aaronse, 2021-12-18:
Hello Lars ( @Eisbaeeer ), fwiw I implemented slider and some other changes in an abandoned pull request https://github.com/maakbaas/esp8266-iot-framework/pull/44 that may help you.

Eisbaeeer, 2021-12-26:
> Hello Lars ( @Eisbaeeer ), fwiw I implemented slider and some other changes in an abandoned pull request #44 that may help you.

Hi aaronse
Thank you for the hint to your pull-request. Did you merged all other features in your repo backwards?
I need additional feature (redirect to a code-created URI). This will be awesome.
Actually my react knowledge is limited. I have to investigate more time to learn the react functionality.
Greets Lars

maakbaas, 2021-12-29:
Added this, you can now use `"control": "slider"` in either your `configuration.json` or `dashboard.json`.

closed 2021-12-29 by maakbaas